betaflight/betaflight #9859: RX Firmware update through S.Bus

awhite32, 2020-05-29:
I recently purchased a drone which came with an FrSky XM+ s.bus receiver pre-installed.  This is a micro-drone so the wires are soldered and the parts are shrink wrapped and tough to reach.  Well, the controller had the non-US firmware, so I had to update it via S.Bus, which meant de-soldering and connecting to my controller to perform the update.

But you talk S.Bus, and to the PC via USB.  Why can't you allow the firmware download to the RX via the Betaflight app, using the S.Bus connection, so we don't need to desolder. 

This will also help with future updates to RX firmware.  I am not sure how complex the FrSky protocol is, but the controller is running OpenTX, and programs the controller via onboard S.Bus connection, I would assume the code for this programming is available in the opentx source code.

etracer65, 2020-05-29:
FrSky receivers don't support firmware updates through the SBUS connection as it's unidirectional (output from receiver only).

awhite32, 2020-05-29:
I just performed the update using my Taranis, and their web page lists firmware, and how to load it:
https://www.frsky-rc.com/xm-plus-mini-sbus-non-telemetry-full-range/

awhite32, 2020-05-29:
https://www.frsky-rc.com/wp-content/uploads/Downloads/Firmware/ACCST-2.1.x/FW-XM-XMP-v2.1.0.zip

etracer65, 2020-05-29:
Sorry, you're correct. On the XM+ it's possible to use the SBUS. Other FrSky receivers require the S.port connection for firmware upgrades. So you possibly could do it with serialpassthrough at least on F7 flight controllers. For F4 there's the problem that there is a hardware unidirectional inverter that will block the signal from going from the flight controller to the receiver.

etracer65, 2020-05-29:
It's the `serialpassthrough` CLI command. But like I said, it might work for this. Depends on what the FrSky firmware update process needs. You'd also need to know what baud rate the firmware updater needs to communicate with the receiver. Don't know whether anybody has ever gotten this to work or not, but theoretically it's possible. And to be clear, this command doesn't only exist for F7. The problem with F4's is that you have the unidirectional hardware inverter so communication from the flight controller to the receiver is impossible. F7's don't need the hardware inverter so the connection is direct to the receiver.

mikeller, 2020-05-30:
@awhite32: Updating of FrSky RX firmware throught the flight controller is possible, as @etracer65 says by using `serialpassthrough`. However, the way that firmware updates are done for this hardware complicates this considerably: The RX hardware only waits for a couple of milliseconds after being powered for the firmware update to start, and after that starts normal operation with no way to get it back into firmware update mode except for a power cycle. This is reflected in the firmware update process through OpenTX, where power to the RX is controlled by OpenTX, and the RX is powered up _after_ OpenTX has started sending firmware update requests. In the case of the RX being connected to a flight controller, booting Betaflight and invoking `serialpassthrough` takes more time than the wait time of the RX for a firmware update, so the only way to update the firmware while connected to a flight controller is if power to the RX can be controlled independently of power to the flight controller. Even in this case, the constraints about bidirectional transmission that @etracer65 mentioned exist - and even on F7, in most cases the SBus pin of the RX will be connected to an RX pin on the flight controller, which is not supported for RX/TX use in Betaflight.
A workaround for the bidirectional transmission issue is to configure the pin in question for SOFTSERIAL temporarily for the duration of the firmware update.
But even then, the flight controller has to be powered and set up for `serialpassthrough`, and the FrSky firmware update tool started on the host PC and the firmware update initialised _before_ power is applied to the RX for this to work.

#3364 has a lengthy discussion on this topic.

RipperDrone, 2020-06-01:
[image]
Well, this is a nice twist, didn't know but makes perfect sense. Could theoretically be facilitated with a script like re-assigning connected UART pin prior and post firmware update process. 

For the power cycle, no workaround I could think of other than temporarily de-soldering the 5V or GND pin and manually reconnect for flashing OR putting a switch or solder bridge inline the RX power supply. Future FCs could offer a button for 5V to RX if it was a frequent enough use case for people to buy an FC for that very reason... :-)

mikeller, 2020-06-03:
> For the power cycle, no workaround I could think of other than temporarily de-soldering the 5V or GND pin and manually reconnect for flashing OR putting a switch or solder bridge inline the RX power supply. 

For some flight controllers that have 5V outputs that are not powered when on USB, this can be done by connecting the RX to one of these, and then powering the flight controller over USB, setting up serial passthrough, starting the firmware update utility, and only then plugging in a battery to power up the RX (*props off*). Or use an RX that can be supplied directly from the battery.

closed 2020-07-11 by stale[bot]