betaflight/betaflight #9860: BF 4.2 RC2: Crossfire telemetry, Sensor FM (Flightmode) shows !ERR*

Scavanger, 2020-05-29:
The Crossfire telemetry sensor FM (Flightmode) shows "!Err*" in Betaflight 4.2 RC2, all other sensors seems to work.

haslinghuis, 2020-05-30:
With 4.2RC2 and OpenTX 2.3.7 and LUA 1.5.0 after deleting sensors and discovering new sensors I get the following sensors:

`
1RSS
2RSS
RQly
RSNR
RFMD
TPWR
TRSS
TQly
TSNR
`

closed 2020-05-30 by Scavanger

mikeller, 2020-05-31:
This is weird - flight mode should be sent just as well as other frames like GPS, as all of this is done in `processCrsf()` - the frame format will probably have to be checked against the current CRSF spec to make sure it is still correct:

https://github.com/betaflight/betaflight/blob/bf9cc615c5c50c47dff4fc51331b79eceaca5972/src/main/telemetry/crsf.c#L409-L442