betaflight/betaflight #9861: BF4.2.0-RC3: OSD Profile Selection not working

sawsing, 2020-05-30:
The same set up using CLI or Adjustment Tab in configurator for OSD profile selection works in BF4.1.7 but not in 4.2.0rc3

# aux
aux 0 0 0 1300 2100 0 0
aux 1 1 2 1700 2100 0 0
aux 2 2 2 1300 1700 0 0
aux 3 13 3 1300 1700 0 0
aux 4 15 3 1700 2100 0 0
aux 5 35 1 1900 2100 0 0
aux 6 36 3 1700 2100 0 0

# adjrange
adjrange 0 0 1 900 2100 29 1 0 0

I am using DJI FPV remote controller with sbus_fast.
My FC is Mateksys F722SE

etracer65, 2020-05-30:
I just tested and OSD profile selection works exactly as it should. Betaflight has no control over the OSD presented by the DJI goggles. Contact DJI for support.

mikeller, 2020-05-31:
@sawsing: Are you using only the dji TX and air unit, or are you using the dji goggles as well?

sawsing, 2020-05-31:
> @sawsing [1]: Are you using only the dji TX and air unit, or are you
> Links:

I am using DJI goggles, DJI remote controller and Caddx Vista air unit.
OSD profile selection is done by a 3-way switch on the DJI remote 
controller.
I had sbus_baud_fast = ON
Thanks for your attention.

mikeller, 2020-05-31:
@sawsing:

> I am using DJI goggles, DJI remote controller and Caddx Vista air unit.

Ok, in this case Betaflight does not have any involvement in what is shown in the OSD - you will have to contact dji and / or Caddx to get this fixed.

closed 2020-05-31 by mikeller