betaflight/betaflight #9862: Hex File

4noxx, 2020-05-30:
Hello, where can i get a hex file? I need it for my Betaflight F4 Board with activated uart1. 

Problem is, my usb port is broken. Now i need a activated uart1 to connect via FTDI. Hope anyone can help

haslinghuis, 2020-05-30:
Look at the [https://github.com/betaflight/betaflight/wiki/Unified-Targets](url) on how to flash a hex on your board with a proper config.

Download the STM32F405.hex file at [https://github.com/Betaflight/Betaflight/releases](url)

You will need to select the proper unified target config for the Betaflight F4 FC.
Download the config file for your board at [https://github.com/betaflight/unified-targets/blob/master/configs/default/FPVM-BETAFLIGHTF4.config](url)

etracer65, 2020-05-30:
Directly from the Configurator.

[image]

haslinghuis, 2020-05-30:
Yeah the Broken USB Rescue Maintenance Release has enabled MSP on UART1 by default. Thanks @etracer65. Did not know this existed. `git log | grep` was not giving the results I was looking for.

Normal MSP can only use UART1 or UART3 and depends on the board config. If you should need UART3 you will have to change the code and build the firmware yourself [https://github.com/betaflight/betaflight/pull/2144](url)

closed 2020-07-11 by stale[bot]