betaflight/betaflight #9882: Support for new custom designed board

pratik206, 2020-06-03:
hello 
I am editing STM32_unified target for my new STM32F7xx based board. The STM32_unified only support  STM32F405, STM32F411, STM32F722, STM32F745 for these four MCU? If i want to add different  STM32F7xx how can i do that ?

mikeller, 2020-06-03:
@pratik206: If your MCU si compatible with one of these then it should work out of the box. If not then there is likely a lot more work needed than just updating the target definition. What MCU are you using?

pratik206, 2020-06-03:
hello mikeller 

Thanks for your reply. I am using STM32F777VITX. I made the changes in STM32_unified target file according to my hardaware connection. I modified if defined (STM32F745)  and build as make TARGET=STM32F745 and its done build successfully. but when i uploaded the hex file and checked in betaflight configurator it is not detecting USB VCP port. Could you please suggest how it will work for my STM32F777VITX board ?  

target.h 

#pragma once
#if defined(STM32F745)
#define TARGET_BOARD_IDENTIFIER "S745"
#define USBD_PRODUCT_STRING     "Betaflight STM32F745"
//#define USE_UNIFIED_TARGET

#define LED0_PIN                PB3
#define LED1_PIN                PB4

// *************** Gyro & ACC **********************
#define USE_SPI
#define USE_SPI_DEVICE_1

#define SPI1_SCK_PIN            PA5
#define SPI1_MISO_PIN           PA6
#define SPI1_MOSI_PIN           PA7
#define USE_EXTI
#define USE_GYRO_EXTI
#define GYRO_1_EXTI_PIN         PC5

#define USE_MPU_DATA_READY_SIGNAL

// ICM 20689
#define GYRO_1_CS_PIN           PA4
#define GYRO_1_SPI_INSTANCE     SPI1

#define USE_ACC
#define USE_ACC_SPI_ICM20689

#define USE_GYRO
#define USE_GYRO_SPI_ICM20689
#define GYRO_1_ALIGN            CW90_DEG

// Serial ports
#define USE_VCP
//#define USE_USB_DETECT
//#define USB_DETECT_PIN          PA8
......

etracer65, 2020-06-03:
Betaflight doesn't have any support from STM32F7x7 at this point.

mikeller, 2020-06-03:
@pratik206:

> I made the changes in STM32_unified target file according to my hardaware connection.

As I said before there is **a lot** more to getting a new MCU type supported than just modifying the target for a supported MCU type.

One thing that needs investigating is clock frequencies - if the clock frequency that is driving the USB port is off then the device will not be recognised on USB. In this case, if nothing else is wrong and the firmware actually boots, the LEDs will be blinking during / after startup.

pratik206, 2020-06-08:
hello mikeller 

could you please suggest me where to make USB port ON. The device is going to DFU mode and using betaflight configurator i am uploading the firmware. but after uploading firmware successfully the port is not detecting USB VCP.

mikeller, 2020-06-08:
@pratik206: https://github.com/betaflight/betaflight/issues/9882#issuecomment-638436955 gives you some pointers on what to look for.

closed 2020-07-18 by stale[bot]