betaflight/betaflight #9883: Angle Mode missing from Modes tab

nikstojkov, 2020-06-03:
Angle mode has not apperaed in the modes tab, unless im being an idiot? which is very possible. It had appeared when I first flashed RC4, but seems to be pot luck if it shows up, and haven't managed to get it to come back yet. Not a huge issue, as I fly acro 99% of the time, but it has its uses.

**To Reproduce**
Flashed Betaflight 4.2 RC4 to Aikon F7

**Expected behavior**
Angle mode should be there...

 ``` 
# version
# Betaflight / STM32F7X2 (S7X2) 4.2.0 Jun  3 2020 / 11:52:14 (9718914fd) MSP API: 1.43
# config: manufacturer_id: AIKO, board_name: AIKONF7, version: 34f72565, date: 2019-10-03T17:08:47Z

# start the command batch
batch start

board_name AIKONF7
manufacturer_id AIKO

# name: Skitzo

# feature
feature -RX_PARALLEL_PWM
feature -ANTI_GRAVITY
feature RX_SERIAL
feature TELEMETRY
feature OSD
feature ESC_SENSOR

# beeper
beeper -RX_LOST
beeper -RX_LOST_LANDING
beeper -DISARMING
beeper -ARMING
beeper -ARMING_GPS_FIX
beeper -BAT_CRIT_LOW
beeper -BAT_LOW
beeper -GPS_STATUS
beeper -RX_SET
beeper -ACC_CALIBRATION
beeper -ACC_CALIBRATION_FAIL
beeper -READY_BEEP
beeper -DISARM_REPEAT
beeper -SYSTEM_INIT
beeper -ON_USB
beeper -BLACKBOX_ERASE
beeper -CRASH_FLIP
beeper -CAM_CONNECTION_OPEN
beeper -CAM_CONNECTION_CLOSE
beeper -RC_SMOOTHING_INIT_FAIL

# serial
serial 0 64 115200 57600 0 115200
serial 2 8192 115200 57600 0 115200
serial 3 1024 115200 115200 0 115200

# aux
aux 0 0 0 1700 2100 0 0
aux 1 13 5 1700 2100 0 0
aux 2 35 2 1900 2100 0 0
aux 3 49 2 1300 1700 0 0

# vtxtable
vtxtable bands 5
vtxtable channels 8
vtxtable band 1 BOSCAM_A A CUSTOM  5865 5845 5825 5805 5785 5765 5745 5725
vtxtable band 2 BOSCAM_B B CUSTOM  5733 5752 5771 5790 5809 5828 5847 5866
vtxtable band 3 BOSCAM_E E CUSTOM  5705 5685 5665 5645 5885 5905 5925 5945
vtxtable band 4 FATSHARK F CUSTOM  5740 5760 5780 5800 5820 5840 5860 5880
vtxtable band 5 RACEBAND R CUSTOM  5658 5695 5732 5769 5806 5843 5880 5917
vtxtable powerlevels 3
vtxtable powervalues 25 200 400
vtxtable powerlabels 25 200 500

# master
set dyn_notch_width_percent = 0
set dyn_notch_q = 250
set dyn_notch_min_hz = 90
set dyn_notch_max_hz = 350
set acc_calibration = 35,30,16,1
set mag_hardware = NONE
set baro_hardware = NONE
set rssi_scale = 150
set rc_smoothing_auto_smoothness = 20
set serialrx_provider = CRSF
set dshot_idle_value = 700
set dshot_bidir = ON
set motor_pwm_protocol = DSHOT600
set vbat_max_cell_voltage = 440
set vbat_min_cell_voltage = 350
set vbat_warning_cell_voltage = 360
set current_meter = ESC
set battery_meter = ESC
set vbat_scale = 112
set beeper_dshot_beacon_tone = 4
set yaw_motors_reversed = ON
set small_angle = 180
set deadband = 6
set yaw_deadband = 6
set osd_vbat_pos = 2530
set osd_link_quality_pos = 2081
set osd_tim_2_pos = 2551
set osd_flymode_pos = 2073
set osd_throttle_pos = 2520
set osd_current_pos = 481
set osd_mah_drawn_pos = 235
set osd_avg_cell_voltage_pos = 2498
set debug_mode = GYRO_RAW
set vtx_band = 5
set vtx_channel = 4
set vtx_power = 2
set vtx_freq = 5769
set displayport_max7456_wht = 3
set gyro_rpm_notch_harmonics = 2
set gyro_rpm_notch_q = 800
set name = Skitzo

profile 0

# profile 0
set dyn_lpf_dterm_curve_expo = 7
set anti_gravity_gain = 1000
set feedforward_transition = 40
set iterm_relax = OFF
set iterm_relax_cutoff = 10
set yaw_lowpass_hz = 70
set throttle_boost_cutoff = 10
set acro_trainer_angle_limit = 21
set i_pitch = 30
set d_pitch = 15
set f_pitch = 48
set i_roll = 30
set d_roll = 15
set f_roll = 45
set i_yaw = 30
set f_yaw = 45
set d_min_roll = 0
set d_min_pitch = 0
set ff_interpolate_sp = AVERAGED_3
set ff_spike_limit = 55
set ff_smooth_factor = 40
set ff_boost = 0

rateprofile 0

# rateprofile 0
set rates_type = ACTUAL
set roll_rc_rate = 20
set pitch_rc_rate = 20
set yaw_rc_rate = 20
set roll_expo = 54
set pitch_expo = 54
set yaw_expo = 54
set roll_srate = 67
set pitch_srate = 67
set yaw_srate = 67

# end the command batch
batch end
 ``` 

[image]

wocoltd, 2020-06-03:
I had this problem too - accelerometer was enabled but ANGLE and HORIZON modes missing from modes tab. (Board FURY4OSD).

Fixed by rolling back to RC3.

mikeller, 2020-06-03:
Fixed in #9884, test firmware available!

nikstojkov, 2020-06-04:
As above, fixed by rolling back to RC3 for the AikonF7.

closed 2020-06-06 by mikeller

capposteve, 2020-06-08:
i didnt have this issue until using 10.7.0 rc2, rc1 didnt have the issue for me

haslinghuis, 2020-06-08:
Yeah yeah already fixed in #9884 .
Maybe a RC5 would give less issues repeating over and over again :trollface:

capposteve, 2020-06-09:
yea sorted, thanks

PR #9884 (2020-06-06): Fixed modes missing on startup.
Fixes #9883.

This is fallout from #9875. The problem is that `initActiveBoxIds()` was run too early during startup before the sensors were detected. Reinstating `initActiveBoxIds()` in `mspInit()` to fix this.