betaflight/betaflight #9885: Merge akk_vtx_fix branch with master

Diaoul, 2020-06-04:
Given the recent configurable VTX option and the small patch footprint on the code, I suggest to merge the branch `akk_vtx_fix` with a configuration option `vtx_flavor` that can be set to `default` or `akk`.

This would allow:
* Unified user experience
* Allow for those user to test pre-releases
* Decomissionning of the Jenkins job

Let me know if you're ok with this, I can start working on a PR.

RipperDrone, 2020-06-04:
haha, been proposing to give in with the akk/rdq glitch for literally years now. Irony it comes closer with more complex VTX interface now. In the past, the dev team always rejected due to grunt against cloner Co.'s doing inaccurate copies of products, thereby introducing additional coding work... has been a nice wave that the butter/emus of this world could ride on, making them look more flexible and listening to user requests. Let's see what the BF dev teams's judgement will be these days... ;-)

mikeller, 2020-06-06:
@Diaoul: VTX devices with this particular bug have not been sold for a number of years, and the number of users actually using these with Betaflight is declining.
Betaflight will continue to support users of this hardware who were left out in the rain by their suppliers and the manufacturer by providing custom builds of the firmware for the time being.

closed 2020-06-06 by mikeller

Diaoul, 2020-06-06:
I think we could use the same arguments to justify merging the branches, we're talking about a line of code here.
I bet there are a lot of those perfectly working VTX left on the shelf due to the BF stuff, people are basically buying new hardware and spending money because of this lack of direct support.

I understand why this was done and it worked as expected, manufacturers now sell compliant modules. This is why I thought it was time to move on, nowadays only users are penalized by this, not manufacturers that moved on.

But fair enough, a branch isn't the best but at least it's there.