betaflight/betaflight #9886: Error with Bidirectional DSHOT on SP Racing H7 EXTREME

Nico9n, 2020-06-04:
**Describe the bug**
I can not get Bidirectional DSHOT working on motor 1 and 3 with Betaflight 4.2 RC4. I have errors in the motor tab. Tested with dshot 150,300 and 600.  With Betaflight 4.1 rpm filter is working on all 4 motors. 

**To Reproduce**
Flash  SP Racing H7 EXTREME with Betaflight 4.2 RC4. Setup Bidirectional DSHOT.

**Expected behavior**
No errors in MotorTab

**Flight controller configuration**
'''
# 
# Building AutoComplete Cache ... Done!
# 
# diff

# version
# Betaflight / SPRACINGH7EXTREME (SP7E) 4.2.0 Jun  3 2020 / 11:50:18 (9718914fd) MSP API: 1.43

# start the command batch
batch start

board_name SPRACINGH7EXTREME

# master
set dshot_bidir = ON
set use_unsynced_pwm = OFF
set motor_pwm_protocol = DSHOT600

profile 0

rateprofile 0

# end the command batch
batch end

'''

**Setup / Versions**
 - Flight controller [ SP Racing H7 EXTREME  from https://shop.seriouslypro.com/sp-racing-h7-extreme];
 - Other components [only 4x  Siskin ST Blhei_32 v.32.7];

**Additional context**
With Betaflight 4.1.7 Bidirectional DSHOT is working without errors.
[image]
[image]

haslinghuis, 2020-06-04:
@hydra if you have the time - could you please test with the Siskin ESC 32.7 with BFC17.0RC2 / BF4.2RC4

hydra, 2020-06-05:
@haslinghuis can you get the manufacturer to send me one for testing?

hydra, 2020-06-05:
@haslinghuis @Nico9n can you confirm exactly which ESC you have please.  I can't find Siskin ST here...

http://www.sunrisemodel.com/en/product3.php?cid=103

haslinghuis, 2020-06-05:
@hydra - I already sent you the [Turnigy Multistar 32bit 51A ESC's](https://hobbyking.com/nl_nl/turnigy-multistar-32bit-51a-race-spec-esc-2-6s-opto.html?queryID=a49b5147f8cd5bfba4c39f2cea3c1ce6&objectID=74316&indexName=hbk_live_magento_nl_nl_products) for testing ;)

Nico9n, 2020-06-05:
> @haslinghuis @Nico9n can you confirm exactly which ESC you have please. I can't find Siskin ST here...
> 
> http://www.sunrisemodel.com/en/product3.php?cid=103

I don't now it exactly.  Here are some pictures. 
[image]
[image]
[image]

Nico9n, 2020-06-06:
Today I flashed Betaflight 4.1. Only esc and motors are connected. Bidirectional DSHOT was working without errors. Then I soldered the FrSky r9mini to the FC. Now I got the same Bidirectional DSHOT errors in Betaflight  4.1. I tested the r9mini at an other fc without an error. The r9mini is new and working correctly. 

Maybe the FC is defect? @hydra should I send it to you? It is the FC I ordered last week.

Nico9n, 2020-06-07:
The R9mini consumes 50mA. I replaced the R9min with a 100Ω resistor to be sure it is not faulty. The 100Ω resistor consumes 50mA too. The problem is still there. I think it must be the FC.

hydra, 2020-06-08:
@nico-dh the issue is so weird!  Can you please measure the voltage input to the FC's 3V3 LDOs with/without resistor or RX and measure both of the 3V3 LDO's voltage outputs, with and without the resistor.

Additionally where *exactly* are you connecting the RX and/or resistor?  Have you tried other 5V pads/connections?  There are quite a few, narrowing the issue down would help a lot.

Oscilloscope traces of the DSHOT signals and 3v3 LDO outputs would narrow down the problem.

Also, are you able to send the FC *and* the ESC to me, if so I can do the investigation here.

hydra, 2020-06-08:
> @hydra - I already sent you 4 [Turnigy Multistar 32bit 51A ESC's](https://hobbyking.com/nl_nl/turnigy-multistar-32bit-51a-race-spec-esc-2-6s-opto.html?queryID=a49b5147f8cd5bfba4c39f2cea3c1ce6&objectID=74316&indexName=hbk_live_magento_nl_nl_products) or didn't they survive your testing 😉

yes, I have the stuff you sent me still.  I can take another look at that on the scope with 4.2 RC4 and see if there's a change.  Those ESCs are different to @Nico9n 's ones.  I used them to help @sskuag with the ESC init issue in https://github.com/bitdump/BLHeli/issues/410

Nico9n, 2020-06-08:
Ok, here are some Oscilloscope traces. Connected to configurator and powered with 3S Lipo. No Motors are spinnig. Error is at 25%
[image]
First without load. 3.

3V measured at the 3.3V pad. 

[image]
[image]
[image]
[image]
[image]

Nico9n, 2020-06-08:
And here with resistor. I used ihe 5V Pin near the buzzer.
[image]
[image]
[image]
[image]
[image]

Nico9n, 2020-06-08:
> @nico-dh the issue is so weird! Can you please measure the voltage input to the FC's 3V3 LDOs with/without resistor or RX and measure both of the 3V3 LDO's voltage outputs, with and without the [...]

I will take the other Oscilloscope traces today evening. 
I read it too late.

Nico9n, 2020-06-08:
With load, 5V input 3.3V LDO
[image]
LDO:
3.3V LDO near gyro
[image]
3.3V LDO near osd
[image]

no load, 5V input 3.3V LDO
[image]
3.3V LDO near gyro
[image]
3.3V LDO near osd
[image]

hydra, 2020-06-09:
@Nico9n thanks for the traces, I'll review and digest ASAP.

WingsFishes, 2020-06-21:
Hi 

> > @nico-dh the issue is so weird! Can you please measure the voltage input to the FC's 3V3 LDOs with/without resistor or RX and measure both of the 3V3 LDO's voltage outputs, with and without the resistor.
> > Additionally where _exactly_ are you connecting the RX and/or resistor? Have you tried other 5V pads/connections? There are quite a few, narrowing the issue down would help a lot.
> > Oscilloscope traces of the DSHOT signals and 3v3 LDO outputs would narrow down the problem.
> > Also, are you able to send the FC _and_ the ESC to me, if so I can do the investigation here.
> 
> I will take the other Oscilloscope traces today evening.
> I read it too late.
> 
> This is my setup:
> [image]
> [image]

Hi Nico
Have same issues when i try to use RPM filter with my H7.
On which  UART nunero did you put your Esc in Betaflight config  ?

hydra, 2020-06-22:
> [image]
> [image]
> [image]

The first part of the signal is generated by the H7 where the slew rates is sharper than the second part of the signal, which is generated by the ESC.  Both appear to be fine.

Overall those traces look nice and sharp, certainly the slew-rate and timing seem correct to me, so electrically there's no issue.  Also of note is that there is no spike when the IO line transitions from OUTPUT to INPUT which used to be an issue on the H7 before it was fixed.

@joelucid As a DSHOT telemetry expert, can you see anything wrong with the traces above?

@Nico9n Did you use M1 to record the traces? did you notice anything different between M1 and M2,M3,M4 signals?

Nico9n, 2020-06-23:
@hydra yes, it was from M1. I did not notice any difference between M1 - M4. I have tested dys Aria 35A with the same problem. So I have replaced the Fc with the other you sent me back and there are no errors anymore. I think the fc is defect. It is on the way to you.

hydra, 2020-06-23:
> @hydra yes, it was from M1. I did not notice any difference between M1 - M4. I have tested dys Aria 35A with the same problem. So I have replaced the Fc with the other you sent me back and there are no errors anymore. I think the fc is defect. It is on the way to you.

Yup, I'll inspect it when it gets here but it's hard to imagine what could possibly be wrong with it.  Still, it will be interesting to see.

hydra, 2020-07-07:
@Nico9n was kind enough to send me his FC for investigation, after hooking it up to a Mamba 506 ESC and probing the M1-M4 signals at the ESC output this is what I found:

M1:
[image]

M2:
[image]

So the ESC isn't pulling the voltage on M1 down to 0v as part of the DSHOT telemetry response. 

The soldering on the FC's M1-M4 pins looks perfect too, attached.

[image]
[image]

At this time I have no idea why and I'll investigate a little more.  My current suspicion is that somehow the input circuitry of the M1 pin has been damaged, perhaps due to a wiring fault during installation.

I'm going to try using M5-M8 instead of M1-M4 and see what happens.

hydra, 2020-07-07:
ok, so using the CLI commands, as follows, to swap the motor outputs to use M5-M8 I confirm the same signal is seen at the output from ESC 1 on the Mamba 506.

```
resource MOTOR 1 B06
resource MOTOR 2 B07
resource MOTOR 3 C06
resource MOTOR 4 C07
resource MOTOR 5 A00
resource MOTOR 6 A01
resource MOTOR 7 A02
resource MOTOR 8 A03
```

This indicates to me that the issue is with either:
a) the ESC.
b) the firmware.

but rules out:
c) soldering issues of the MCU.
d) PCB manufacturing issues on the FC's M1 net.
e) faulty/bad MCU.

I'll try some other combinations, such as M3/M4 + M7/M8.

hydra, 2020-07-07:
So using M7/M8 and M3/M4 outputs, which are KNOWN GOOD doesn't help either.

[image]

```
resource MOTOR 1 A02
resource MOTOR 2 A03
resource MOTOR 3 C06
resource MOTOR 4 C07
resource MOTOR 5 A00
resource MOTOR 6 A01
resource MOTOR 7 B06
resource MOTOR 8 B07
```

There is still a bad output from the Mamba 506 4in1 ESC on the ESC 1 output.

So, to verify that the issue is ESC related, using the KNOWN GOOD output pins M3/M4 and M7/M8.  The signals from the FC to the pins on the 4in1 ESC for ESC 1 and ESC 2 are swapped, and sure enough the problem DOES NOT move to ESC 2.  There's still a bad signal on ESC 1 pin.  See GREEN wire (ESC 1 / M3) in previous photo.

See this photo where GREEN and ORANGE are swapped over.  Now ORANGE (ESC1 / M4) wire has a bad signal.

[image]

There doesn't seem to be any issue with the FIRST DSHOT signal coming from the FC firmware. (A00, A02 and B06 tested).

So I suspect an ESC manufacturing or ESC firmware issue.

hydra, 2020-07-07:
Doh!  So it turned out that on the 4in1 ESC I have here, ESC1 was flashed with BLHeli_32 32.6 and ESC2-4 were flashed with BLHeli_32 Beta 32.66.  Quite why ESC1 had older firmware then ESC 2-4 I don't know...

After updating them all to BLHeli 32.7 the ESC1 signal is now ok.  No DSHOT errors on M1-M4

[image]

I'll try another ESC on your FC @Nico9n But I'm not seeing any hardware issues on it.

Nico9n, 2020-07-07:
Thank you for testing. Can you try a 3s Lipo / 12V input voltage? With 3s,  I got more errors.

hydra, 2020-07-07:
Sure, no issues on 12V either.

[image]

Nico9n, 2020-07-07:
Crazy thing. I tried so many things, but I never got permanent 0% error. Only swapping the FC to the other H7 resolves my problem.

hydra, 2020-07-07:
[image]

Also tested with Tekko32 F3 4in1 with BLHeli 32.7 (factory firmware) and it's fine too.

hydra, 2020-07-07:
After some investigation, the cause of the errors on M4 are what appears to be a DSHOT response that is sent by the ESC too early.

In this first image you can see consecutive DSHOT commands (the first part of the signal, from the FC) followed by telemetry responses (the second part of the signal, from the ESC).  Note that most of them have 30us gap between them.

[image]

Here's the detail of a 'good' response.
[image]

Here's the detail of a 'bad' response, i.e. one that is set to early.
[image]

The FC has to transition from OUTPUT to INPUT, if the ESC sends the response too early then the FC will miss some of the signal from the ESC and thus you'll get errors.

I guess the BL Heli devs need to investigate this so that all the telemetry responses occur after the same delay of about 30uS (for a DSHOT 300 signal).

hydra, 2020-07-07:
I created a new issue for the BLHeli devs to investigate, here: https://github.com/bitdump/BLHeli/issues/468

hydra, 2020-07-08:
@Nico9n Please can you send me the ESCs you're using so I can investigate, your ESCs aren't F3 based, so it would be nice to know if they also have the problem mentioned in bitdump/BLHeli#468

If not there's not much more investigation I can do for you regarding this issue as electrically your FC seems fine here and works fine with the Mamba 506 ESC.

Nico9n, 2020-07-09:
@hydra yes, I will send you the ESC after the weekend

hydra, 2020-07-09:
@Nico9n 2 please, so I can compare, all 4 if possible.

jlhartley, 2020-07-12:
I have just completed a build with an H7 Extreme, and I believe I am experiencing a similar issue on Betaflight 4.2 running bidirectional DHOT 600.

The ESCs are T-Motor F35As. At first they appear to work with 0% errors, even when ramping up the motor speed in the motors tab. However, as soon as I power my transmitter and the receiver (R-XSR) starts receiving, I see the errors shoot up. After just a few seconds, 3 of the 4 motors stop spinning, and I see high error percentages for 2 ESCs. After this I can start and stop the one motor which continued to spin, but all others are unresponsive. Stranger still, the one spinning motor no longer corresponds to the correct motor as it is mapped in Betaflight.

I am using FPort with the receiver, but even when I switch off serialrx_inverted and serialrx_halfduplex to (deliberately) break the communication with the FC, the problem still occurs. So to me it seems as if the small additional current the receiver pulls from the 5V regulator when it is actively receiving / sending telemetry is enough to interfere with the ESC signalling in some way.

I see no issues at all when running DSHOT 600 with bidirectional DSHOT off.

I have tried different 5V / G pads for the reviewer with no difference. I have also tried DSHOT 300 and 150 with no success. I have the motor signal and signal ground wires twisted together, and have even tried twisting them tighter / more twists. I have ordered a capacitor to go across VBAT, and will test again when it arrives. Other than using shielded signal wires for the ESCs, I can't think what more can be done to minimise electrical noise.

hydra, 2020-07-12:
@jlhartley Can you look at the DSHOT signals on an oscilloscope please?  If not can you send your FC and a couple of your ESCs to me for investigation?

1. do you have 'Rev A' or 'Rev B' SPRacingH7EXTREME?
2. what firmware was on the ESC?

Yes, please try using shield signal wires for the ESCs to see if it helps.  Connect the signal *and* ground at both ends of the shielded wire otherwise you create an antenna.

Nico9n, 2020-07-13:
@hydra @sskaug I can reproduce this here. https://youtu.be/M95RbrCyN9E It takes some time to find the bad frame.

Nico9n, 2020-07-13:
> @Nico9n 2 please, so I can compare, all 4 if possible.

[image]
the escs will be on the way soon.

jlhartley, 2020-07-13:
@hydra I have a DS213 oscilloscope, but not much experience with this sort of in-depth electronics diagnostics. I've really only just worked out how to use it, however I hope the following images are of some use. I've looked at 26.7 microsecond intervals and I think I have spotted an issue with one frame:

Good Frame:
[image]

Bad Frame:
[image]

I could be completely wrong on that as I'm not very confident with what I'm doing.

1. How do I find the revision? I couldn't see any obvious markings on the board. Hopefully the photos below help.
2. The ESCs have BLHeli32 version 32.7 firmware.

I will attach some photos in case there is something obviously wrong that I have missed:

[image]

[image]

[image]

The soldering is less than perfect, but I checked for any shorts and solder blobs quite carefully and nothing seems to be wrong there.

jlhartley, 2020-07-13:
I've been back through the thread and re-read it, and now I think I better understand how bidirectional DSHOT works and what I'm looking at in the traces. I should be focusing on the timing between the commands from the FC and the reported telemetry data returned by the ESC, specifically the second bit being sent too early.

So far I haven't been able to spot this, but I will keep looking. What I have seen, which looks very odd, is the voltage occasionally dropping down to around 1.7V peak to peak. I've only managed to catch this happening twice:

[image]

[image]

When most of the time I'm seeing a more normal signal like this:

[image]

Testing on Betaflight 4.2 vs 4.1.7, the difference I'm noticing is that 4.1.7 seems a lot more tolerant of the high error %. I struggled to get the ESCs to 'lock up' running 4.1.7, but with 4.2 sometimes the ESCs will 'lock up' within a few seconds of running the motors in the motors tab. During the 'lock up' state the ESCs go into, I just see a flat signal line at around 3.5V in the scope.

hydra, 2020-07-15:
@jlhartley Exactly what ESCs are you using? Your ESCs look like they are T-Motor F35A 3-6S 32bit ESCs, please confirm.

A photo of an ESC and a screenshot of the BlHeli suite after connecting to the ESCs via the FC would be useful too.

The F35A's seem to be using an STM32F051K MCU, as per this random photo I found of one:

[image]

Reference: https://customquads.com.au/products/t-motor-f35a-3-6s-32bit

Given the nature of the fix it's possible that other MCU's that BlHeli supports are also affected by the bug too.

hydra, 2020-07-15:
@mikeller @etracer65 @joelucid You guys may want to investigate the BF 4.2 issue mentioned by @jlhartley in his last comment - https://github.com/betaflight/betaflight/issues/9886#issuecomment-657829904

Use the same ESC as him and BlHeli 32.7 or lower to re-produce the 'lock-up' issue, that needs some investigation regardless of the BlHeli 32.71 telemetry response delay fix that @sskaug has made here - https://github.com/bitdump/BLHeli/issues/468#issuecomment-658169960

@jlhartley Can you confirm that you *did* or *did not* manage to get the ESC to 'lock-up' with BF 4.1.7.

hydra, 2020-07-15:
@jlhartley Can you record a video of the DSHOT signal with 4.1.7 and 4.2.0 when it 'locks-up' please and provide a link here.

jlhartley, 2020-07-15:
@hydra I confirm that they are T-Motor F35A 3-6S 32bit ESCs. That image is accurate, it is the same chip.

Here is a screenshot from BLHeli32 Suite. Nothing changed from defaults other than motor direction for two motors:

[image]

To clarify, I did get the same behaviour to occur with 4.1.7, it just seemed to take longer with the motors spinning for it to happen. However, today by chance I powered up with a 6S rather than a 4S and found the problem did not occur! Furthermore, the lower the voltage on 4S the worse the problem seems to become. I tried a power supply to try to identify the exact voltage where I find problems - at around 20.5V and higher, the error % drops away until at 22V it is around 0% almost all the time, and I see no 'lock up' no matter how fast I spin the motors. Therefore my variations in testing between 4.1.7 and 4.2 could simply be battery voltage.

When I next have some more time, I will produce a video showing the problem in detail and all the troubleshooting steps I have tried. I have really struggled to get my oscilloscope to pause at exactly the right moment, the options for triggering seem quite limited, and the buffer seems to be very small. During the 'lock up' state the signal line is held high, but I can never manage to capture the frames which came before on the scope.

However, I also have a logic analyzer, and I think I have been able to capture the failure point twice. Each time it just looks like the frame was cut off early and the signal line held high.

Cut off partway through a command packet (FC --> ESC)

[image]

The frame just preceding this (final command and telemetry response):

[image]

I don't know how to decode the telemetry data, but I presume it is possible to see whether the ESC was returning nonsense RPM values by decoding those pulses? Certainly I can see nonsense RPM values in blackbox logging.

hydra, 2020-07-16:
@jlhartley please can you get the signal into 'lock-up' state, then disconnect the motor signal wire somewhere between the FC and the ESC and check BOTH the ESC and the FC on the scope.

This will tell us if it's the ESC holding the signal line high or if it is the FC holding the signal line high.

Use two channels of your scope for this connected to a simple toggle switch.

Also, to get your scope to pause at the right time, use the delay trigger to find HIGH signal that is longer than the normal amount of time between two consecutive DSHOT signals from the FC.

jlhartley, 2020-07-19:
@hydra I have put a switch on the M1 signal wire, and recorded this video: https://www.youtube.com/watch?v=bgw_iwQTpAk

Channel A in blue is the FC side, channel B in yellow is the ESC side. At around 15 seconds you can see me just checking with the switch, I toggle off and back on again and watch the ESC initialise.

The 'lock up' occurs 32 seconds in. 3 out of 4 motors stop spinning - it is always the same 3. After toggling the switch, the signal remains high on both the FC and ESC side. It's quite hard to see, but on the FC side there is some slight ripple. I toggle the switch on and off a few times, but the ESC does not initialise again.
When connected over USB, the FC has to be power cycled to clear the problem.

I still could not get my oscilloscope to trigger on the continuous high state, I only have these options:

[image]

The closest I could get was with the above voltage pulse width option, however because the signal doesn't go low again this wasn't useful. It has only been useful for finding when there is a longer than normal time between frames.
Tomorrow I have a proper bench scope arriving (Rigol) so I wont have to struggle with this small portable one anymore.

**Some further information, if it is useful:**

- I have re-examined the FC and found the marking for the revision - it is revision A.

- I have fitted the capacitor, but there has not been any improvement.

- The time taken for a 'lock up' to occur varies greatly. Sometimes the motors spin for a few minutes, sometimes just seconds. Sometimes they 'lock up' immediately after initialising and they don't spin at all. From my testing this seems partly dependent on battery voltage.

Frustratingly I have no spare FCs or ESCs to test different combinations. I'm reluctant to order more parts at this stage. However if I still can't get this setup to work or find some useful data when the scope arrives tomorrow, I think I will order more parts to get the quad flying.

jlhartley, 2020-07-21:
@hydra At last I have a proper scope, I can trigger and get that final frame before the problem occurs:

[image]

[image]

[image]

There seems to be some sort of anomaly before the telemetry packet:
[image]

hydra, 2020-07-21:
@jlhartley that is just the FC transitioning from output to input mode at the end.of the DMA DSHOT transfer. Safe to ignore.  @joelucid and I already investigated that.

jlhartley, 2020-07-22:
@hydra Does that mean that the FC is continuing to transition from output to input after I see the signal line held high, resulting in very similar looking spikes after the last frame, or is this likely to be caused by another source of noise?

[image]

The trace below is 6S. Am I correct in thinking that everything looks a lot less noisy at 6S voltage, which fits with my previous finding that errors stay close to 0% at 6S voltage and I cannot reproduce this problem?

[image]

ledvinap, 2020-07-23:
@jlhartley : You can try connection small (~ 100ohm) resistor in dshot connection. That way you can see collision (voltage drop will be larger on FC side)

Nico9n, 2020-07-23:
> @Nico9n Please can you send me the ESCs you're using so I can investigate, your ESCs aren't F3 based, so it would be nice to know if they also have the problem mentioned in [...]

@hydra Did you get the esc`s?

spracingofficial, 2020-12-09:
@nico-dh Yes.  Took a while for me to get round to testing them for you, sorry about that.  I replied by email to you earlier today, but for others following this thread:

* the FC he sent me that I put in a quad has been flying just fine once the firmware was upgraded with BLHeli test code.
* the ESCs he sent me don't report any telemetry errors when running the latest BLHeli 32.7 firmware.  The firmware that was on them also reported the version as 32.7 but that may have been some early release of 32.7 or some test code, only the manufactuer and @sskaug would know for sure.

Nico9n, 2020-12-10:
@spracingofficial @hydra you linked @nico-dh but I think it was for me.

jaisor, 2021-05-08:
Not sure what the latest on this is, but I might have experienced the same issue. Using SKYSTARSF7HD stack "SKYSTARS F7 F722 FLIGHT CONTROLLER AND 60A BLHELI-32 32BIT ESC". When running BLHeli 32.8, bidirectional RPM configured 4khz/DSHOT300, I kept getting jerks (snags, desyncs) on sharp flips (and rarely rolls). Happened often on fresh 6S and less as the voltage dropped. It did not happen on 5S. 

I noticed DShot Error on motor #1 was flaring up occasionally between 0 and around 5%, even when motors were not running. Rest of the motors were fine. Few blackbox logs I took showed Motor 1 flipping out part of the event which is what made me look at it closer .

Flashed BLHeli 32.7, no more DShot error on M1 and after 2 fresh 6S packs I have not been able to reproduce it.

hydra, 2021-05-09:
@jaisor If you reflash with 32.8 does the problem come back?

skrat, 2021-12-19:
I have similar issues I believe with MAMBA APP F722 MK1, MAMBA F50 _BL32 stack, running BLHeli 32.8. 2/4 motors report 100% errors, I tried re-flashing 32.8 to the ESC with no change. Suite doesn't show any possible upgrade/downgrade for my ESC.

dendraftsman, 2022-01-04:
@skrat I have similar issues I believe with MAMBA APP F722 MK1, MAMBA F50 _BL32 stack, running BLHeli 32.8. 2/4 motors report 100% errors. With disabled bidir, everything seems fine.

skrat, 2022-01-04:
@dendraftsman so, it turned out it's not a HW error, it's not BLHeli conf., it's BF config, kind people at Diatone had me Reset BF to defaults, then load their conf. and everything started to work.

dendraftsman, 2022-01-04:
@skrat Thanks, I will try resetting Bf

haslinghuis, 2022-04-02:
This issue should be resolved.

[image]

closed 2022-04-02 by haslinghuis

jlhartley, 2022-04-07:
Just to confirm after all this time, amazingly it is 4.3.0-RC4 that finally seems to resolve this issue (at least on the bench). I was following #11155 closely but I thought it was just a new issue with 4.3 that might not solve my problem. I could never make bidirectional DSHOT work with 4.2 or 4.1 with my SPRACINGH7EXTREME, even after trying F0 and F3 MCU based ESCs.

For future reference, if there are others who had issues with bidirectional DSHOT going all the way back to 4.1, it looks like they are resolved after 4.3.0-RC4.

haslinghuis, 2022-04-07:
Yes seems to have been a long road: https://github.com/bitdump/BLHeli/issues/410

hydra, 2022-04-07:
@jlhartley yes, the H7EXTREME was the first H7 board to ship and we worked out the DSHOT non-BITBANG (DPWM) issues and found bugs in BlHeli that have been fixed.  The issue in #11155 applied to BITBANG DShot only, which might be the default in 4.3.0-RC4 for the H7EXTREME where it would have been DPWM for 4.1.

Please can you confirm the dshot variant (DPWM or BITBANG) that you used for your test today and provide a diff. And then please test again with the OTHER mode and report back?

[1 more reply]