betaflight/betaflight #9888: Mamba 722 mini will not work correctly with Bidirectional Dshot bf4.17 rc1

capposteve, 2020-06-06:
Hi.

After having a days worth of issues on my GT R369 with bdshot not working i found alot of posts with the same issues,
my specs are as follows:

pre built dyatone GT R369
Mamba 722 mini stack running BF4.17 OC1
Mamba 306 blheli_32 esc running 32.7

i have set it up exactly as per instructions on the git hub page. but i am getting lots of random errors,
i can plug the battery into the quad and it will give me the start up tone, but no double beep, then it will start the start up tone again but doing it on the motors in random orders so they end up over lapping. sometimes constantly beeping the motors too. 
running DSHOT 300 and 4k4k to see if lowing it helped but it didnt.

other times if it did double beep it would loose 2 of the 4 escs and the two that where working did actualy work. 
its not consistent in the fact its never exactly the same fault. 

i have followed others advice with removing the camera etc which made no difference, i did however twist the esc cable 360 deg and that has made things much more stable but still not perfect.

i thought this issue had been resolved last year in older versions of bf but it apears that bdshot does not like any emf anywhere near the signal wires? is there a known fix for this issue?

cheers

steve

[image]

haslinghuis, 2020-06-06:
#9840

haslinghuis, 2020-06-06:
Duplicate of #9840

closed 2020-06-06 by mikeller

capposteve, 2020-06-06:
> Duplicate of #9840

How is it a duplicate? They are different boards?