betaflight/betaflight #9889: Locking up when creating rate modes in adjustments tab

Dwilson48625, 2020-06-06:
While configuring rate profile adjustments in BF 4.2 RC4, it will lock up when switching from "Adjustments" tab and "PID" tab. It will get stuck on PID tab and cant go to any other tabs. Refresh doesn't seem to get it unstuck. Restarting BF does get it working again.

(first ever bug report so I hope I did this correctly)

mikeller, 2020-06-06:
@Dwilson48625: Please read the instructions you are given when opening bug reports, and follow them:

**Describe the bug**
A clear and concise description of what the bug is.

**To Reproduce**
Steps to reproduce the behavior:

**Expected behavior**
A clear and concise description of what you expected to happen.

**Flight controller configuration**
Create a `diff` and post it here in a code block. Put ``` (three backticks) at the start and end of the `diff` block (instructions  on how to do a diff: https://oscarliang.com/use-diff-not-dump-betaflight/)

Use `resource show all` to create a resource allocation list and post it here in a code block. Put ``` (three backticks) at the start and end of the output block.

**Setup / Versions**
 - Flight controller [what type is it, where was it bought from];
 - Other components [RX, VTX, brand / model for all of them, firmware version where applicable];
 - how are the different components wired up.

**Additional context**
Add any other context about the problem here.

Dwilson48625, 2020-06-06:
While configuring rate profile adjustments in "Adjustments" tab on BF 4.2 RC4, it will lock up when switching from "Adjustments" tab and "PID" tab. It will get stuck on PID tab and cant go to any other tabs. Refresh doesn't seem to get it unstuck. Rates will change when flipping my 3 pos switch, it just gets stuck on that tab. Restarting BF does get it working again. 

To reproduce, set up rates profile selection for 3 rates in Adjustments tab. Then go to PIDS tab to see if the settings took. Everything in the PIDS tab will work as expected, but you may not be able to move to any other tabs.

I am also having random connection errors. For example, I just did the Diff and Resource dump for the above bug, then clicked "Adjustments tab" and it disconnected from the board, took me back to the Betaflight start up page, and no longer shows COM3 (My fc com) and only shows COM1 and manual). Closed BF and reopened, still no COM3. If I restart, it will come back and reconnect. I realized this issue after I made this bug report last night, so it was NOT included in the original post. This has happened several times using various "save and reboot" during setting up my board. I do not have any more specific info on what triggers it as it doesn't do it all the time and is random.

Windows 10 home, Ryzen5 1600 6 core 3.60ghz, 16gb ram, 64 bit, MSI gaming (270 I wanna say) MOBO, VGA gforce 1050ti.

FC is a Kakute f7, Tekko32 esc's remapped 90 degrees, Atlatl vtx, Avenger V2 2507 motors, R-XSR RX running ACCST Fport FCC on UART6, XLite TX running opentx ver.2.3.2.

Diff/Resource blocks to follow:

'''# diff

# version
# Betaflight / STM32F745 (S745) 4.2.0 Jun  3 2020 / 11:51:52 (9718914fd) MSP API: 1.43
# config: manufacturer_id: HBRO, board_name: KAKUTEF7, version: cc8e118f, date: 2020-03-04T02:37:20Z

# start the command batch
batch start

board_name KAKUTEF7
manufacturer_id HBRO

# resources
resource MOTOR 1 E09
resource MOTOR 2 B00
resource MOTOR 3 E11
resource MOTOR 4 B01
resource LED_STRIP 1 NONE
resource LED 1 NONE
resource CAMERA_CONTROL 1 D01

# feature
feature -RX_PARALLEL_PWM
feature RX_SERIAL
feature LED_STRIP
feature DISPLAY
feature ESC_SENSOR

# map
map TAER1234

# serial
serial 0 2048 115200 57600 0 115200
serial 5 64 115200 57600 0 115200
serial 6 1024 115200 57600 0 115200

# aux
aux 0 0 0 1450 2100 0 0
aux 1 1 1 900 1200 0 0
aux 2 35 3 1700 2100 0 0
aux 3 47 1 1275 1650 0 0

# adjrange
adjrange 0 0 2 1775 2100 12 2 0 0
adjrange 1 0 2 1300 1700 12 2 0 0
adjrange 2 0 2 900 1175 12 2 0 0

# vtxtable
vtxtable bands 6
vtxtable channels 8
vtxtable band 1 BAND     A CUSTOM  5865 5845 5825 5805 5785 5765 5745 5725
vtxtable band 2 BAND     B CUSTOM  5733 5752 5771 5790 5809 5828 5847 5866
vtxtable band 3 BAND     E CUSTOM  5705 5685 5665    0 5885 5905    0    0
vtxtable band 4 FATSHARK F CUSTOM  5740 5760 5780 5800 5820 5840 5860 5880
vtxtable band 5 RACEBAND R CUSTOM  5658 5695 5732 5769 5806 5843 5880 5917
vtxtable band 6 IMD6     I CUSTOM  5732 5765 5828 5840 5866 5740    0    0
vtxtable powerlevels 3
vtxtable powervalues 0 1 2
vtxtable powerlabels 25 200 800

# master
set acc_trim_roll = 2
set acc_calibration = 9,5,63,1
set baro_hardware = AUTO
set serialrx_provider = FPORT
set serialrx_inverted = ON
set serialrx_halfduplex = ON
set min_throttle = 1070
set dshot_bidir = ON
set use_unsynced_pwm = OFF
set motor_pwm_protocol = DSHOT600
set motor_pwm_rate = 480
set failsafe_throttle = 1250
set failsafe_procedure = AUTO-LAND
set current_meter = ESC
set yaw_motors_reversed = ON
set small_angle = 180
set osd_units = IMPERIAL
set osd_warn_batt_not_full = OFF
set osd_warn_visual_beeper = OFF
set osd_warn_crash_flip = OFF
set osd_warn_launch_control = OFF
set osd_warn_no_gps_rescue = OFF
set osd_warn_gps_rescue_disabled = OFF
set osd_warn_rssi = ON
set osd_cap_alarm = 1500
set osd_alt_alarm = 500
set osd_vbat_pos = 108
set osd_rssi_pos = 2066
set osd_link_quality_pos = 35
set osd_rssi_dbm_pos = 12
set osd_flymode_pos = 2061
set osd_throttle_pos = 2445
set osd_current_pos = 5
set osd_mah_drawn_pos = 108
set osd_altitude_pos = 2123
set osd_warnings_pos = 14762
set osd_avg_cell_voltage_pos = 2091
set osd_battery_usage_pos = 2505
set osd_disarmed_pos = 2283
set osd_stick_overlay_left_pos = 1
set osd_stat_tim_1 = ON
set osd_stat_tim_2 = OFF
set osd_stat_endbatt = ON
set osd_stat_battery = ON
set osd_stat_max_alt = ON
set osd_stat_bbox = OFF
set osd_stat_bb_no = OFF
set osd_stat_total_time = ON
set camera_control_key_delay = 125
set gyro_1_align_yaw = 2700

profile 2

rateprofile 0

# rateprofile 0
set roll_rc_rate = 130
set pitch_rc_rate = 130
set yaw_rc_rate = 130
set roll_expo = 50
set pitch_expo = 50
set yaw_expo = 50
set roll_srate = 67
set pitch_srate = 67
set yaw_srate = 67

# end the command batch
batch end'''

RESOURCE:

'''# resource
resource BEEPER 1 D15
resource MOTOR 1 E09
resource MOTOR 2 B00
resource MOTOR 3 E11
resource MOTOR 4 B01
resource MOTOR 5 C09
resource MOTOR 6 A03
resource PPM 1 E13
resource SERIAL_TX 1 A09
resource SERIAL_TX 2 D05
resource SERIAL_TX 3 B10
resource SERIAL_TX 4 A00
resource SERIAL_TX 6 C06
resource SERIAL_RX 1 A10
resource SERIAL_RX 2 D06
resource SERIAL_RX 3 B11
resource SERIAL_RX 4 A01
resource SERIAL_RX 6 C07
resource SERIAL_RX 7 E07
resource I2C_SCL 1 B06
resource I2C_SDA 1 B07
resource SPI_SCK 1 A05
resource SPI_SCK 2 B13
resource SPI_SCK 4 E02
resource SPI_MISO 1 A06
resource SPI_MISO 2 B14
resource SPI_MISO 4 E05
resource SPI_MOSI 1 A07
resource SPI_MOSI 2 B15
resource SPI_MOSI 4 E06
resource CAMERA_CONTROL 1 D01
resource ADC_BATT 1 C03
resource ADC_RSSI 1 C05
resource ADC_CURR 1 C02
resource SDCARD_CS 1 A04
resource SDCARD_DETECT 1 D08
resource OSD_CS 1 B12
resource GYRO_EXTI 1 E01
resource GYRO_CS 1 E04
resource USB_DETECT 1 A08'''

Hope that info helps.

Dale

Asizon, 2020-06-06:
@Dwilson48625 Which Betaflight Configurator version are you running?

Dwilson48625, 2020-06-06:
> @Dwilson48625 Which Betaflight Configurator version are you running?

Configurator: 10.7.0-RC1 (091ce5b5)

I see its out dated now, my bad. I will try the updated version.

Sorry for any confusion...

Dale

Dwilson48625, 2020-06-06:
Was running outdated configurator, my bad. 

Here was my confusion with RC$. In firmware flasher, I selected: 4.2.0-RC4- 03-06-20 08:16 for KAKUTEF7 board.. Im still a noob....

The above issues are now gone. Sorry about the confusions guys. 

Dale

closed 2020-06-06 by Dwilson48625

Asizon, 2020-06-06:
Dont worry, anyway thanks for reports.