betaflight/betaflight #9896: Could you show the current signal flow diagram for 4.2

and-sh, 2020-06-07:
Too many improvements have been made from a simple PID controller.
To understand how this affects flight performance, it may be helpful to show something like this diagram.
[ServoFlowDiagram](https://www.dynomotion.com/Help/ServoFlowDiagram.htm)

etracer65, 2020-06-07:
PID controller code is here: 

https://github.com/betaflight/betaflight/blob/master/src/main/flight/pid.c

Have at it.

and-sh, 2020-06-07:
Well, yes, only 1,700 lines of code.
Is it possible to depict this in the picture. Better multi-colored. It would be nice to sign which parameters control each block.

etracer65, 2020-06-07:
> Is it possible to depict this in the picture. Better multi-colored. It would be nice to sign which parameters control each block.

Sure, feel free to do this.

and-sh, 2020-06-08:
Do you have any drafts?

haslinghuis, 2020-06-08:
Please take a look at the [Betaflight wiki](https://github.com/betaflight/betaflight/wiki/BetaFlight-Deep-Dive).

and-sh, 2020-06-09:
Thank you, I saw it. Very good start.

I began to collect everything in one place:
[PID 4i2.pdf](https://github.com/betaflight/betaflight/files/4753409/PID.4i2.pdf)
Unfortunately, so far I do not have much free time.

haslinghuis, 2020-06-09:
I know what you mean but time is just an illusion ⌛

and-sh, 2020-06-11:
Draft N1 ready. 
Please check that the diagram (roughly) matches the code.
What important details did I miss?
[PID 4i2.pdf](https://github.com/betaflight/betaflight/files/4762492/PID.4i2.pdf)

and-sh, 2020-06-11:
Draft 1.01
[PID 4i2.pdf](https://github.com/betaflight/betaflight/files/4766322/PID.4i2.pdf)

and-sh, 2020-06-12:
Draft 1.5
[PID 4i2.pdf](https://github.com/betaflight/betaflight/files/4771537/PID.4i2.pdf)
 What important did I miss?

and-sh, 2020-06-13:
Ok. I think I've played enough.
Draft 1.6. Added some links.
[PID 4i2.pdf](https://github.com/betaflight/betaflight/files/4774860/PID.4i2.pdf)
Source [PID 4i2.zip](https://github.com/betaflight/betaflight/files/4774862/PID.4i2.zip)
I think this can be added to the wiki. This must be approved by programmers.

mikeller, 2020-06-14:
@and-sh: Thank you for your effort, this looks very good. And bonus points for doing it in LibreOffice!

haslinghuis, 2020-06-14:
Wow did you use some flowchart generator or did you do this all by manual code analysis? Very impressive diagram.

ctzsnooze, 2020-06-14:
That's really very good.  Maybe...

The box titled 'RPM Notch filters' includes settings for lowpass filters which aren't notches.  It contains a mix of filters and parameters to configure the filters.

The gyro filters themselves, with settings in brackets, are:
- RPM Notch filters (gyro_rpm_notch_harmonics, gyro_rpm_notch_q, gyro_rpm_notch_min)
- Dynamic Notch filter/s (dyn_notch_width_percent, set dyn_notch_q, set dyn_notch_min_hz, set dyn_notch_max_hz)
- Gyro Notch 1 and 2 (gyro_notch_hz, gyro_notch_cutoff)
- Dynamic Gyro Lowpass (gyro_lowpass_type, gyro_lowpass_hz, dyn_lpf_gyro_min_hz, dyn_lpf_gyro_max_hz)
- Static Gyro Lowpass (gyro_lowpass2_type, gyro_lowpass2_hz)

Although I like seeing the parameters, maybe just list the names in that box, all in one list?

There should be an input box titled DShot RPM telemetry at the left.  It should have two outputs.  One goes to the 'GyroFilters' box, and the other to the Dynamic Idle box.

VBat Sag compensation should be an input to motors.  VBat PID compensation is an input to mixers.  Both could have an input coming from a box on the left titled VBat.

motor_output_limit is an input to motors.  

The triangle with `?` in it can be named 'Mixer'.

ctzsnooze, 2020-06-14:
I think its wise to not mention D rpm filtering.
However the D filter box again is a mix of settings and filters.
As per gyro, here is a list of the filters with parameters in brackets:
- Dynamic D lowpass (dterm_lowpass_type, dyn_lpf_dterm_min_hz, dyn_lpf_dterm_max_hz, dyn_lpf_dterm_curve_expo)
- Static D lowpass (dterm_lowpass2_type, dterm_lowpass2_hz)
- Static D notch (dterm_notch_hz, dterm_notch_cutoff)

and-sh, 2020-06-14:
@haslinghuis: No. It was assembled in pieces from 3.4-4.2 Tuning notes, 3.0-4.1 Release notes and the links in them. 
@ctzsnooze : Ok. It will take some time. Do not close the topic yet.

IllusionFpv, 2020-06-14:
> 
> 
> I know what you mean but time is just an illusion ⌛

I'm illusion and I have time problems too.. 🤣

and-sh, 2020-06-17:
V 2.0 ready
[PID 4i2.pdf](https://github.com/betaflight/betaflight/files/4792550/PID.4i2.pdf)

haslinghuis, 2020-06-18:
@ctzsnooze please review 2.0

and-sh, 2020-06-19:
V2.01
[PID 4i2.pdf](https://github.com/betaflight/betaflight/files/4802235/PID.4i2.pdf)

mituritsyn, 2020-06-19:
[image]
Isn't it from BF docs? Are there sources anywhere?

and-sh, 2020-06-20:
Cool, I'm not the only one who loves pictures. Google found the following:
[uav tech facebook](https://www.google.com/imgres?imgurl=https%3A%2F%2Flookaside.fbsbx.com%2Flookaside%2Fcrawler%2Fmedia%2F%3Fmedia_id%3D2223167571277698&imgrefurl=https%3A%2F%2Fen-gb.facebook.com%2Fpages%2Fcategory%2FEducational-Consultant%2FUAVTech1%2Fposts%2F&tbnid=VbT6pB0voTQSCM&vet=12ahUKEwiP15G60o_qAhUHuCoKHUpNDd8QMygAegQIARB-..i&docid=VOs6s3S4JVo-uM&w=960&h=678&itg=1&q=betaflight&hl=ru&ved=2ahUKEwiP15G60o_qAhUHuCoKHUpNDd8QMygAegQIARB-)

ctzsnooze, 2020-06-30:
Looking good!

Some minor suggestions...

- throttle goes into the dynamic low passes
- it's also an input to anti-gravity
- there's no link between error and delta, either way.
- absolute control provides feedback for sustained errors to slightly change setpoint

I think its getting very complete now.

[image]

and-sh, 2020-07-03:
Ok. In progress

and-sh, 2020-07-12:
V3
[PID 4i2.pdf](https://github.com/betaflight/betaflight/files/4908787/PID.4i2.pdf)
Well, it became a little bit complicated. I think about lite version w/o parameters.

mituritsyn, 2020-07-12:
> V3
> [PID 4i2.pdf](https://github.com/betaflight/betaflight/files/4908787/PID.4i2.pdf)
> Well, it became a little bit complicated. I think about lite version w/o parameters.

Can you please share the source doc?

and-sh, 2020-07-13:
Of course. 
[PID 4i2 v3.zip](https://github.com/betaflight/betaflight/files/4910339/PID.4i2.v3.zip)
This is OpenOffice drawing. It has no direct analogue in MS office. You can use LibreOffice (it is free, small, quite usable, multi-planform (I use it on windows and Linux) ) to edit it,  or I can export drawing to eps or emf.

and-sh, 2020-07-13:
Light version.
[PID 4i2 lite.pdf](https://github.com/betaflight/betaflight/files/4912596/PID.4i2.lite.pdf)

mituritsyn, 2020-07-14:
[PID.4i2.v3.1.zip](https://github.com/betaflight/betaflight/files/4918129/PID.4i2.v3.1.zip)
Made some ''wiring" management. Seems it's needed to check d-term part. Is it error based or calculated directly from gyro?

and-sh, 2020-07-14:
Prettily. Please export this to something viewable.
I think it is so. Calculation of delta from pid.c:
`        gyroRateDterm[axis] = gyro.gyroADCf[axis] `
...........
`const float delta =
            - (gyroRateDterm[axis] - previousRawGyroRateDterm[axis]) * pidRuntime.pidFrequency / D_LPF_RAW_SCALE;
        previousRawGyroRateDterm[axis] = gyroRateDterm[axis];`
Before there was parameter  `delta_from_gyro`. There is no such parameter in 4.2.

and-sh, 2020-07-14:
I think this is right. Delta from setpoint now called feedforward. If you look on diagram there is a symmetry - gyro - D and setpoint - FF.

mituritsyn, 2020-07-14:
> Prettily. Please export this to something viewable.

Did you meant something like PDF? Maybe will be better to export this in something like draw.io or more maintanable format?
[PID 4i2.pdf](https://github.com/betaflight/betaflight/files/4920100/PID.4i2.pdf)

and-sh, 2020-07-15:
Thank you.
I think PDF is enough.  PDF can be viewed from here in single click. PDF is better than raster (jpg, png and so on)
because it is vector. 
You have lost wire from integral to summing.
If you have any interest feel free to develop this work.  You may use any soft you like. I chose Open Office just because it was installed on my pc. And Open Office quite cope with the task

mituritsyn, 2020-07-15:
[PID.4i2.v3.1.zip](https://github.com/betaflight/betaflight/files/4928067/PID.4i2.v3.1.zip)
[PID 4i2.pdf](https://github.com/betaflight/betaflight/files/4928068/PID.4i2.pdf)

ctzsnooze, 2020-07-16:
FF is a bit complicated.  The smoothing is applied in different places depending on whether the newer interpolated FF is active (the 4.2 default) or the older interpolation method is used.

With the new interpolated FF method, RC smoothing is applied in two very different locations.  For that, I think we will need two separate RC smoothing boxes.

The first box is about the smoothing applied to setpoint after it has been calculated. Its parameters are rc_smoothing_auto_smoothness, rc_smoothing_input_type, and rc_smoothing_input_hz.  It is applied after setpoint is calculated.

The second box is the smoothing applied to FF **after** it is calculated.  Its parameters are rc_smoothing_auto_smoothness, rc_smoothing_derivative_type, and rc_smoothing_derivative_hz.

The only user inputs to interpolated type FF are ff_interpolate_sp,  ff_spike_limit, ff_max_rate_limit and ff_smooth_factor.

The only user input to ff_boost is the ff_boost amount.

If ff_interpolate_sp = OFF, then the old method of linear setpoint interpolation is active.

The old method linearly interpolates setpoint itself.  Its inputs are rc_interp (mode), rc_interp_ch and rc_interp_int.  FF is calculated from the derivative of the interpolated setpoint and FF smoothing is then applied.  There is no boost in this mode, and none of the interpolate_ff parameters apply.

This makes this section quite complex.  I am not sure how best to do it.

I think it is best to show the default 4.2 behaviour, primarily.  The flow chart could be titled '4.2...' so it won't be confused with future or past diagrams.

In this diagram the old behaviour is in grey.

[image]

and-sh, 2020-07-21:
OK. FF corrected. V3.2. 
[PID 4i2.pdf](https://github.com/betaflight/betaflight/files/4953648/PID.4i2.pdf)
Interesting, parameter rc_smoothing_type in this case is obsolete?

ctzsnooze, 2020-08-05:
Sorry to be slow responding... the changes are good.

Some other small points I just realised...

iterm_windup doesn't have a throttle input.  It is a bit special, its input is motor differential.  So its path would be:

[mixer] ----> [iterm_windup] ----> [ iTerm ]

iterm_rotation doesn't have a throttle input either.  I think it has as an input the smoothed gyro data, from which it determines how the position of the quad changes in space, and the I vectors are oriented to be consistent in the world frame of reference rather than in the frame of reference of the quad.  So it needs its own little box....

--- [gyro filters] ---> [iterm_rotation] ---> [ iTerm ]

The only things that gain a throttle input and use this to change iTerm are anti-gravity and the iterm_relax parameters.

There are other I related modifiers like iterm limits, and acc_limit.  

acc_limit uses gyro derivative to limit iTerm.  the idea being to prevent iTerm accumulating when the rate of change of gyro is so high that likely a big error would develop.  

Getting better and better!

mrRobot62, 2020-08-09:
Hi guys, yeah that's very good approach to make a deep(er) dive into the BF 4.2 processing flow. Very cool.

Some questions to understand the picture better (u know: a pic should say more than 1000 words ;-) )

And maybe some improvements :-)

The line and box coloring says what?
Bold marked text says what?
Some boxes are filled only with a word like "Setpoint" others with parameters. Maybe it would be better to separate
- Like "functions" with sharp edged rectangles
- Parameters with round edged rectangles (or vice versa)
- Hardware input/output (like gyro, motors, accelerometer,...) with dotted rectangles 
- avoiding line mergers (if possible), because that can be obfuscated (e.g. 4 lines into "+" before mixer).
- if possible and for better understanding of flow direction use arrows allways

@and-sh good job man

and-sh, 2020-08-10:
@ctzsnooze  Ok. In progress.
@mrRobot62 Thank you. I will also try to make some cosmetic improvements.

mrRobot62, 2020-08-10:
@and-sh : you did this chart with what software?

What do you think about to repaint it with https://app.diagrams.net - so it's possible to convert this into pdf, html and others. This tool is not perfect but easy to use and helpful.

Maybe a co-working part - I can help you - I will do the "re-paint" it with above tool, based on your source and afterwards we can work together on this topic and in the meantime you work the extensions into the current flowchart.

A very cool feature is to convert into HTML and if the BF-Crew agree - it can be published on BF-site - however - it's an offer

I made dozens of flowcharts and diagrams on my job :-O

mrRobot62, 2020-08-10:
It's only a first draft - without review, coloring is not finished a.s.o
https://github.com/mrRobot62/betaflight_processing/blob/master/bf-4.2_processing-workflow.pdf

mrRobot62, 2020-08-11:
Version 0.7 - hope it's ok, maybe some colors are incorrect and arrows are to the wrong direction :-O
@ctzsnooze - I included your last requirement with Item_windup & item_rotation
@and-sh : it's ok for you? I put all the stuff on a github repo, Would you like to upload your sheet too?

and-sh, 2020-08-11:
@mrRobot62 Of course. I interesting in diagram firstly.  By collecting a lot of scattered information, I have a better understanding of how the Betaflight magic actually works. As you can see, I have little artistic flair.
Feel free to develop this work.  I guess it can be added to the wiki when completed. 
p.s.
The numbers in brackets are the version of Betaflight that introduced the feature with a hyperlink to the wiki. Better to keep the links.

mrRobot62, 2020-08-11:
@and-sh : yes, for me it's the same. If I can see a picture, it's much more easier to understand the background. I think it's a wonderful starting point for us and (hopefully) for others.

(x.y) numbers: yeah that was after some thinking about my understanding too :-)
(coloring): mhm - I'm not a native speaker - what do you thing with "There is no point in highlighting any part"?
My idea behind is: To colorize parts which are mostly influenced by something (like Gyro) ... something in this direction. You know what I mean.

My question - should we go on with this "online drawing software" and with this ? With git it's much more easier with version history a.s.o

and-sh, 2020-08-11:
@mrRobot62  I talking about gray rectangle "PID-controller". This all is PID controller.

mrRobot62, 2020-08-12:
new version available :-)

and-sh, 2020-08-12:
@mrRobot62 
Well.
I am not a native speaker too - if you are in doubt about something just ask: what do you mean.
ctzsnooze can provide details that only developers understand. This is not the main job for him, so the process will drag on for another 2-3 months. So be patient.
I think this is a fairly simple project and I see no point in organizing collective online work.
You can use any software you like.
I don't have a lot of free time and I would gladly leave this job to someone.
I will try to correct any inaccuracies that I understand.
I think it's better to put the results in one place. I think that this issue is quite suitable.
If there is no activity for about two to three weeks, I will simply continue this work.

mrRobot62, 2020-08-12:
hi @and-sh - yeah I read this thread and from @ctzsnooze that he some "absence" time :-)

However, I would like to continue this work - from my background I'm a developer but today I do a lot of other stuff around IT (IT-Security...).

BTW: I assume u did some re-engineering work - means looking into source code right ?- did(do) your use a software for that or "only walking around the code" ?

and-sh, 2020-08-13:
Only walking around.
Some notes
[bf-4.2_processing-workflow(1).pdf](https://github.com/betaflight/betaflight/files/5067076/bf-4.2_processing-workflow.1.pdf)
It should looks like
[PID 4i2.pdf](https://github.com/betaflight/betaflight/files/5067102/PID.4i2.pdf)

mrRobot62, 2020-08-13:
@and-sh thanks for this input. Implemented into the new image. Did you check your changes from source code? Maybe I misunderstood @ctzsnooze . However, I changed my image due to your changes

As additional UX - I created layers into the pic. If you download the HTML file and open it, you can enable/disable layers to remove parts from the whole picture. (maybe it's useful)  (currently only throttle)

[HTML - Layers](https://github.com/mrRobot62/betaflight_processing/blob/master/bf-4.2_processing-workflow.html)

and-sh, 2020-08-13:
@mrRobot62 No. It is clear from description and common sense. See for example [Settings to minimise bounceback](https://github.com/betaflight/betaflight/wiki/4.2-Tuning-Notes#settings-to-minimise-bounceback). In code everything is much more confusing. 
acc_limit acts in a similar way. In contrast Dmin Dmax and TPA changes D and P itself. 

[PID 4i2.pdf](https://github.com/betaflight/betaflight/files/5069329/PID.4i2.pdf)

About layers. I think it is not very correct to hide branches. We need to simplify the diagram. Something like [Light version](https://github.com/betaflight/betaflight/issues/9896#issuecomment-657548328)

mrRobot62, 2020-08-13:
and good idea with the light version - will implement this via layers, so it's possible to have both on the same source. Will do this later (weekend...)

mrRobot62, 2020-08-13:
huh - it was much more easier as assumend
https://github.com/mrRobot62/betaflight_processing/blob/master/bf-4.2_processing-workflow_light_V0.11.pdf (alpha)

mrRobot62, 2020-08-13:
@and-sh - cool thanks - one whish : if you made changes in your version, could your please make a "dot or something else" on this changes, much more easiere for me to find it :-)

and-sh, 2020-08-14:
I have read more carefully about iterm_windup. Rather, it affects the I, not the Iterm. 
Rebuilt links that relate to iterm_windup
[PID 4i2.pdf](https://github.com/betaflight/betaflight/files/5075544/PID.4i2.pdf)

mrRobot62, 2020-08-14:
that was my first impression too - with term_windup - I reengineer the graph.
BTW: news version on the git, it's very easy to switch from full view to light version. There is a layer which represent the "parameters (like ff_boost, ...)", if activated you see everything, if deactivated you have the "light" version.

BTW: I made some "walking around source code" and finde a different between rc_interpolation and rc_smooting. It depends what user set in parameter rc_interpolation_type.. I made some corrections due to this behaviour in the graph.

mrRobot62, 2020-08-15:
@and-sh : I believe there is something wrong in your graph. acc_limit / acc_limit_yaw are not behind ITerm and do not go into PIDSum.

As I understand source code right, both parameters are used during pid_init.c to limit velocity for axis Roll and Y.
`pidRuntime.maxVelocity[FD_ROLL]` = pidRuntime.maxVelocity[FD_PITCH] = pidProfile->rateAccelLimit * 100 * pidRuntime.dT;`. Inside P calculation acceleration limitation is done before P is calculated

Due to that, I think both parameter should go into the integral sign and are related to d/dt

mrRobot62, 2020-08-15:
@and-sh : I believe there is something wrong in the graph `acc_limit / acc_limit_yaw` are not behind ITerm and do not go (directly) into PIDSum.

As I understand source code right, both parameters are used during `pid_init.c` to limit velocity for axis Roll/Pitch and YAW (Y with acc_limit_yaw).
`pidRuntime.maxVelocity[FD_ROLL]` = pidRuntime.maxVelocity[FD_PITCH] = pidProfile->rateAccelLimit * 100 * pidRuntime.dT;`. 

Due to that, I think both parameter should go into the integral sign and are related to d/dt, because they are used to calculate the P/D and I values.

ìterm_limit` is only an ITerm limitation and is used during calculation of ITerm (`pid.c`) as a max negative or positiv value (-iterm_limit, +iterm_limit) and should connected to I

and-sh, 2020-08-15:
I don't understand C very well, especially taking into account the many compilation options.
1. acc_limit
I think there are several similar functions that limit iterm accumulation.
I think acc_limit, that you mention, change setpoint itself.
[line 545](https://github.com/betaflight/betaflight/blob/968e72fae88eea39ae5e3b936a47445362dd29fa/src/main/flight/pid.c#L545)
and next 
[line 891](https://github.com/betaflight/betaflight/blob/968e72fae88eea39ae5e3b936a47445362dd29fa/src/main/flight/pid.c#L891)

2.  It rotate accumulated iterm inside pid loop. 
[line 873](https://github.com/betaflight/betaflight/blob/968e72fae88eea39ae5e3b936a47445362dd29fa/src/main/flight/pid.c#L873)

3. Absolute control available, you can check it. Just enable it in CLI and fly.

Therefore, for now, I leave it as written in Wiki and notes by @ctzsnooze.

and-sh, 2020-08-16:
@mrRobot62 
Some notes
[bf-4.2_processing-workflow(3).pdf](https://github.com/betaflight/betaflight/files/5079950/bf-4.2_processing-workflow.3.pdf)

[15 more replies]

closed 2021-02-08 by github-actions[bot]