betaflight/betaflight #9897: GPS Rescue: Add Configurable Buffer to CURRENT_ALT

EggsBenedict, 2020-06-07:
**Describe the solution you'd like**
MAX_ALT adds a buffer of 15m once you break past the minimum value, I think. It would be really nice if CURRENT_ALT had either a fixed buffer of 15 meters (at least) or, ideally, a user configurable buffer.

Both MAX_ALT and CURRENT_ALT could leverage the same configurable value:
`gps_rescue_alt_buffer`: Additional height (in meters) to add to gps rescue return altitude.  Enabled for MAX_ALT only after `gps_rescue_initial_alt` is exceeded.  Enabled at all times for CURRENT_ALT.

**Describe alternatives you've considered**
For the most part, at distance we are limited to LOS flights.  I would think a pretty common scenario would be dipping behind a hill or unknown object suddenly where CURRENT_ALT could run you back into the hill.  Obviously MAX_ALT is an alternative (and what I use) but on some flights that have dramatic altitude changes the return to MAX_ALT can be pretty destructive. Yes, there are configurable throttle limits - it's nice to keep those pretty high so initial correction is fairly aggressive - to avoid bottoming out.

peteoz, 2020-06-08:
since I was considering to submit similar request myself,  just adding

**Additional context**
here I would like to quote @etracer65 from Dec 2018 (because he is exactly right) :
" It's a "rescue" function with the goal to get the quad back into range so the pilot can regain control. Popping up to the altitude recorded to clear obstacles aids in this as the radio link is easier to establish with a clear line of site."

back then we only had one GPS rescue mode "MAX_ALT". that was much better than nothing but as already mention by OP in some scenarios this can be destructive, imagine having just climbed a mountain 2km above the launch site and dived back down the mountain only to failsafe while still being 5km from home, the quad would attempt to climb back to 2km at the time when the battery may only have just about enough capacity to run home in a straight line. also considering many long range flyers using Li-Ion batteries these days that only take few seconds at slight throttle increase to drop voltage down to 2.5V or less.

the subsequent implementation of new GPS rescue mode "CURRENT_ALT" was a great news and had a potential to improve on the original "MAX_ALT", except for one small omission - safety margin/buffer. 
just like MAX_ALT, the CURRENT_ALT implementation should have a safety margin, without this safety buffer/margin we are likely to crash into a terrain/object/tree/rock or whatever else was the cause of the failsafe in the first place.

it is always our intention to fly LOS and when we do we almost never have a failsafe (some exceptions such as hardware failure, accidentally switching off radio, running out of radio battery), if failsafe occurred while LOS then yes existing implementation of "CURRENT_ALT" would be just fine, but this is hardly ever the case, we failsafe on long range flights mostly due to pilot error when we go out of LOS.
in this scenario we would ideally want the quad to climb up little bit, hence the CURRENT_ALT should be equal to "CURRENT_ALT + SAFETY_BUFFER"

now, I'm not exactly a fan of fixed "safety buffer" of 15m that was implemented with "MAX_ALT", I would honestly prefer CLI configurable parameter, say with possible values of 15m to 50m.
the speed at which we fly or dive, 15m is simply not enough to clear terrain if we accidentally dive behind a wrong ridge or mountain peak.

EggsBenedict, 2020-06-08:
@peteoz thanks for chiming in! I did attempt to implement this myself in the PR above. In this implementation the CLI/OSD configurable parameter applies to both MAX and CURRENT ALT options. I did set the allowable range to 0-100m. Perhaps allowing 100m is too generous...

peteoz, 2020-06-08:
@EggsBenedict 0-100m sounds good actually, when I see some guys diving fast and hard in those chutes, if they lose signal there, they need to clear not just the deep chute but also surrounding trees.

closed 2020-07-27 by stale[bot]