betaflight/betaflight #9902: AtlAtl HV v2 SmartAudio not setting power

augustozanellato, 2020-06-09:
**Describe the bug**
Power setting using SmartAudio doesn't work while channel setting works fine

**To Reproduce**
1. Try to set a different power level using OSD or Configurator
2. VTx leds and OSD both show that the current power level hasn't changed, 

**Expected behavior**
1. Try to set a different power level using OSD or Configurator
2. The correct power level is set

**Flight controller configuration**
```# diff

# version
# Betaflight / STM32F745 (S745) 4.1.7 May 28 2020 / 15:05:58 (9ba02a587) MSP API: 1.42
# manufacturer_id: HBRO   board_name: KAKUTEF7   custom defaults: YES

# start the command batch
batch start

board_name KAKUTEF7
manufacturer_id HBRO

# name: Rage5

# feature
feature -RX_PARALLEL_PWM
feature RX_SERIAL
feature ESC_SENSOR

# beeper
beeper -ON_USB

# serial
serial 0 2048 115200 57600 0 115200
serial 1 16384 115200 57600 0 115200
serial 5 64 115200 57600 0 115200
serial 6 1024 115200 57600 0 115200

# aux
aux 0 0 0 1800 2100 0 0
aux 1 27 4 900 1300 1 0
aux 2 27 5 900 1300 1 0
aux 3 13 2 1700 2100 0 0
aux 4 35 3 1800 2100 0 0

# vtxtable
vtxtable bands 6
vtxtable channels 8
vtxtable band 1 BOSCAM_A A FACTORY 5865 5845 5825 5805 5785 5765 5745 5725
vtxtable band 2 BOSCAM_B B FACTORY 5733 5752 5771 5790 5809 5828 5847 5866
vtxtable band 3 BOSCAM_E E FACTORY 5705 5685 5665    0 5885 5905    0    0
vtxtable band 4 FATSHARK F FACTORY 5740 5760 5780 5800 5820 5840 5860 5880
vtxtable band 5 RACEBAND R FACTORY 5658 5695 5732 5769 5806 5843 5880 5917
vtxtable band 6 IMD6     I CUSTOM  5732 5765 5828 5840 5866 5740    0    0
vtxtable powerlevels 3
vtxtable powervalues 0 1 2
vtxtable powerlabels 25 200 500

# rxfail
rxfail 13 s 1000

# master
set acc_calibration = -150,-151,-82
set baro_hardware = AUTO
set min_check = 1020
set rssi_channel = 14
set serialrx_provider = IBUS
set blackbox_p_ratio = 256
set dshot_bidir = ON
set motor_pwm_protocol = DSHOT600
set small_angle = 179
set pid_process_denom = 1
set osd_warn_rssi = ON
set osd_vbat_pos = 2401
set osd_rssi_pos = 2106
set osd_tim_2_pos = 2423
set osd_flymode_pos = 2081
set osd_crosshairs_pos = 2253
set osd_ah_pos = 2190
set osd_current_pos = 2454
set osd_mah_drawn_pos = 2136
set osd_craft_name_pos = 396
set osd_altitude_pos = 2113
set osd_warnings_pos = 14729
set osd_avg_cell_voltage_pos = 2433
set osd_battery_usage_pos = 2089
set osd_disarmed_pos = 2123
set osd_stat_endbatt = ON
set osd_stat_battery = ON
set osd_stat_max_alt = ON
set osd_stat_max_esc_temp = ON
set debug_mode = SMARTAUDIO
set vtx_band = 1
set vtx_channel = 1
set vtx_power = 1
set vtx_freq = 5865
set vcd_video_system = NTSC
set displayport_max7456_blk = 1
set gyro_1_align_yaw = 2700
set name = Rage5

profile 1

rateprofile 0

# rateprofile 0
set thr_expo = 40
set roll_rc_rate = 75
set pitch_rc_rate = 75
set yaw_rc_rate = 75
set roll_expo = 20
set pitch_expo = 20
set yaw_expo = 20

# end the command batch
batch end```

```# resource show all
Currently active IO resource assignments:
(reboot to update)
--------------------
A00: FREE
A01: FREE
A02: LED 1
A03: FREE
A04: SDCARD_CS
A05: SPI_SCK 1
A06: SPI_MISO 1
A07: SPI_MOSI 1
A08: FREE
A09: SERIAL_TX 1
A10: FREE
A11: USB
A12: USB
A13: SWD
A14: SWD
A15: FREE
B00: MOTOR 1
B01: MOTOR 2
B02: FREE
B03: FREE
B04: FREE
B05: FREE
B06: I2C_SCL 1
B07: I2C_SDA 1
B08: FREE
B09: FREE
B10: FREE
B11: FREE
B12: OSD_CS
B13: SPI_SCK 2
B14: SPI_MISO 2
B15: SPI_MOSI 2
C00: FREE
C01: FREE
C02: ADC_CURR
C03: ADC_BATT
C04: FREE
C05: FREE
C06: FREE
C07: SERIAL_RX 6
C08: FREE
C09: FREE
C10: FREE
C11: FREE
C12: FREE
C13: FREE
C14: FREE
C15: FREE
D00: FREE
D01: FREE
D02: FREE
D03: FREE
D04: FREE
D05: SERIAL_TX 2
D06: SERIAL_RX 2
D07: FREE
D08: SDCARD_DETECT
D09: FREE
D10: FREE
D11: FREE
D12: FREE
D13: FREE
D14: FREE
D15: BEEPER
E00: FREE
E01: GYRO_EXTI
E02: SPI_SCK 4
E03: FREE
E04: GYRO_CS 1
E05: SPI_MISO 4
E06: SPI_MOSI 4
E07: SERIAL_RX 7
E08: FREE
E09: MOTOR 3
E10: FREE
E11: MOTOR 4
E12: FREE
E13: FREE
E14: FREE
E15: FREE
F00: FREE
F01: FREE
F02: FREE
F03: FREE
F04: FREE
F05: FREE
F06: FREE
F07: FREE
F08: FREE
F09: FREE
F10: FREE
F11: FREE
F12: FREE
F13: FREE
F14: FREE
F15: FREE

Currently active Timers:
-----------------------
TIM1: FREE
TIM2: FREE
TIM3: FREE
TIM4: FREE
TIM5: FREE
TIM6: FREE
TIM7: FREE
TIM8:
    CH1: DSHOT_BITBANG 2
    CH2: DSHOT_BITBANG 5
TIM9: FREE
TIM10: FREE
TIM11: FREE
TIM12: FREE
TIM13: FREE
TIM14: FREE

Currently active DMA:
--------------------
DMA1 Stream 0: FREE
DMA1 Stream 1: FREE
DMA1 Stream 2: FREE
DMA1 Stream 3: FREE
DMA1 Stream 4: FREE
DMA1 Stream 5: FREE
DMA1 Stream 6: FREE
DMA1 Stream 7: FREE
DMA2 Stream 0: FREE
DMA2 Stream 1: FREE
DMA2 Stream 2: DSHOT_BITBANG 2
DMA2 Stream 3: DSHOT_BITBANG 5
DMA2 Stream 4: ADC
DMA2 Stream 5: FREE
DMA2 Stream 6: FREE
DMA2 Stream 7: FREE```

**Setup / Versions**
 - Holybro Kakute F7 AIO runnning 4.1.7
 - Holybro Atlatl HV v2, smartaudio connected to tx1

etracer65, 2020-06-09:
Check with the manufacturer to make sure you have the power levels in the vtxtable configured properly. Also check that the vtx is not "locked" or in "race" mode preventing power changes - again check with the manufacturer or see the manual.

augustozanellato, 2020-06-09:
Oh sorry, I just realized that I forgot to say that setting power via the VTx button works albeit that power level change isn't detected by Betaflight. I also tried to debug log SmartAudio and the debug[3] remains always 1 (25 mW) even if the power on the VTx is set to another level using the button.

dkustec, 2020-06-09:
You can try with 1.0 SA version:
`vtxtable powerlevels 4
vtxtable powervalues 7 16 25 40
vtxtable powerlabels 25 200 500 800
save`

dkustec, 2020-06-09:
Try with 1.0 SA version:

`vtxtable powerlevels 4`
`vtxtable powervalues 7 16 25 40`
`vtxtable powerlabels 25 200 500 800`
`save`