betaflight/betaflight #9906: Voltage /cell count causes incomplete boot

glwpie, 2020-06-12:
I am having an issue with my LuxMiniF7. It will not complete the boot sequence when i plug in a charged 6s pack. It successfully boots on a pack lower than 20 volts.
It shipped on 4.1.7 and i have also tried updated to 4.2.0 rc4 to see if it was a fixed bug, but saw no improvement

glwpie, 2020-06-13:
To reproduce issue:
 flash 4.1.7 or 4.2.0 rc4 to board
apply custom defaults
Unplug from PC and power board from 4in1 connector 
Escs will boot, video will appear, but betaflight logo in osd will not appear, and escs will not produce second set of tones indicating no signal from FC

etracer65, 2020-06-13:
Sounds like a hardware problem with the board, probably with the voltage regulator. Recommend you contact the manufacturer for hardware support.

glwpie, 2020-06-13:
I don't believe it is as I can boot it off the USB, plug the charged 6s battery in, and the system remains stable.

etracer65, 2020-06-13:
When connected to USB that supplies the power to the MCU and the voltage regulators aren’t used. It’s faulty hardware. Contact the manufacturer for a warranty return.

closed 2020-06-13 by etracer65