betaflight/betaflight #9907: Battery profiles into OSD profiles

kibinek, 2020-06-13:
**Is your feature request related to a problem? Please describe.**
I am using LiPo and LiIon batteries of different limit voltages and different capacities. E.g. when I do my setup in BetaFlight Configurator for LiPo bat (limit 3,5V) and then I fly with LiIon bat (limit 2,5V) the warnings start much earlier then required.
**Describe the solution you'd like**
To be able to swich among different battery setups.
**Describe alternatives you've considered**
Please either: move battery setup (limit voltages and stuff related to warnings) to OSD profiles to be able to switch (by means of stick commands) among battery types within OSD profiles or: create separate battery profiles switchable by means of stick commands like in INAV.
**Additional context**
I guess nothing more to note :) Thank you for your consideration of this suggestion.

haslinghuis, 2020-06-14:
For wider acceptance could you explain how Lithium Ion batteries benefits the current draw Betaflight requires?

kibinek, 2020-06-14:
I confirm, long range quads and cinewhoops. Racing most probably not.

dischinator, 2020-06-20:
I'd also like to have this eventually implemented in some way. On my cruising quad, which I fly with both LiPo and LiIon batteries I currently have set minimum voltages to 3V to keep the alarm going off early on the LiIon batteries. I got used to check voltages regulary  myself not relying on the alarm, which probably is good anyways.

closed 2020-07-27 by stale[bot]