betaflight/betaflight #9910: BTFL 4.2 RC4 FURYF4OSD No osd

marciano027, 2020-06-14:
After flashing 4.2 RC4 there is no OSD displayed.

Flashing back 4.1.7 resolves the problem
OSD in config is on
Under osd tab NTCS/PAL/AUTO selection is missing.

**Flight controller configuration**
'''
# 

# diff

# version
# Betaflight / STM32F405 (S405) 4.2.0 Jun  3 2020 / 11:51:11 (9718914fd) MSP API: 1.43
# config: manufacturer_id: DIAT, board_name: FURYF4OSD, version: 3a35e73b, date: 2019-09-30T05:46:12Z

# start the command batch
batch start

board_name FURYF4OSD
manufacturer_id DIAT

# name: Kabab 027 - 2

# resources
resource SDCARD_CS 1 B12
resource SDCARD_DETECT 1 D02
resource OSD_CS 1 NONE

# dma
dma SPI_TX 2 0
# SPI_TX 2: DMA1 Stream 4 Channel 0

# feature
feature GPS
feature TELEMETRY

# beacon
beacon RX_LOST
beacon RX_SET

# serial
serial 0 64 115200 57600 0 115200
serial 2 0 115200 57600 0 115200
serial 5 2 115200 57600 0 115200

# aux
aux 0 0 0 1700 2100 0 0
aux 1 2 1 1300 1700 0 0
aux 2 46 1 1700 2100 0 0
aux 3 13 3 1700 2100 0 0
aux 4 35 2 1700 2100 0 0
aux 5 36 3 1700 2100 0 0

# master
set acc_calibration = -20,40,-367,1
set mag_bustype = I2C
set mag_i2c_device = 1
set baro_bustype = I2C
set baro_i2c_device = 1
set rssi_channel = 12
set rssi_src_frame_errors = OFF
set serialrx_provider = CRSF
set blackbox_p_ratio = 64
set blackbox_device = SDCARD
set dshot_idle_value = 550
set dshot_bidir = ON
set motor_pwm_protocol = DSHOT300
set failsafe_procedure = GPS-RESCUE
set current_meter = VIRTUAL
set small_angle = 180
set gps_provider = UBLOX
set gps_sbas_mode = WAAS
set gps_auto_baud = ON
set gps_rescue_initial_alt = 20
set gps_rescue_descent_dist = 30
set gps_rescue_min_sats = 5
set pid_process_denom = 2
set sdcard_detect_inverted = ON
set sdcard_mode = SPI
set sdcard_spi_bus = 2
set osd_units = IMPERIAL
set osd_vbat_pos = 129
set osd_rssi_pos = 2113
set osd_tim_2_pos = 2166
set osd_flymode_pos = 2369
set osd_vtx_channel_pos = 56
set osd_craft_name_pos = 2081
set osd_gps_speed_pos = 2135
set osd_gps_lon_pos = 2446
set osd_gps_lat_pos = 2433
set osd_gps_sats_pos = 2401
set osd_home_dir_pos = 45
set osd_home_dist_pos = 46
set osd_altitude_pos = 2197
set osd_warnings_pos = 2282
set osd_avg_cell_voltage_pos = 2145
set osd_rtc_date_time_pos = 0
set max7456_spi_bus = 0
set gyro_1_align_yaw = 1800
set gyro_2_spibus = 0
set name = Kabab 027 - 2

profile 0

rateprofile 0

# end the command batch
batch end

# 

'''

'''
resource show all
Currently active IO resource assignments:
(reboot to update)
--------------------
A00: FREE
A01: FREE
A02: MOTOR 4
A03: MOTOR 1
A04: GYRO_CS 1
A05: SPI_SCK 1
A06: SPI_MISO 1
A07: SPI_MOSI 1
A08: BEEPER
A09: SERIAL_TX 1
A10: SERIAL_RX 1
A11: USB
A12: USB
A13: SWD
A14: SWD
A15: FREE
B00: MOTOR 2
B01: MOTOR 3
B02: FREE
B03: FLASH_CS
B04: LED 2
B05: LED 1
B06: I2C_SCL 1
B07: I2C_SDA 1
B08: FREE
B09: CAMERA_CONTROL
B10: FREE
B11: FREE
B12: SDCARD_CS
B13: SPI_SCK 2
B14: SPI_MISO 2
B15: SPI_MOSI 2
C00: INVERTER 1
C01: ADC_BATT
C02: FREE
C03: FREE
C04: GYRO_EXTI
C05: FREE
C06: SERIAL_TX 6
C07: SERIAL_RX 6
C08: FREE
C09: FREE
C10: SPI_SCK 3
C11: SPI_MISO 3
C12: SPI_MOSI 3
C13: FREE
C14: FREE
C15: FREE
D00: FREE
D01: FREE
D02: SDCARD_DETECT
D03: FREE
D04: FREE
D05: FREE
D06: FREE
D07: FREE
D08: FREE
D09: FREE
D10: FREE
D11: FREE
D12: FREE
D13: FREE
D14: FREE
D15: FREE
E00: FREE
E01: FREE
E02: FREE
E03: FREE
E04: FREE
E05: FREE
E06: FREE
E07: FREE
E08: FREE
E09: FREE
E10: FREE
E11: FREE
E12: FREE
E13: FREE
E14: FREE
E15: FREE
F00: FREE
F01: FREE
F02: FREE
F03: FREE
F04: FREE
F05: FREE
F06: FREE
F07: FREE
F08: FREE
F09: FREE
F10: FREE
F11: FREE
F12: FREE
F13: FREE
F14: FREE
F15: FREE

Currently active Timers:
-----------------------
TIM1: FREE
TIM2: FREE
TIM3: FREE
TIM4: FREE
TIM5: FREE
TIM6: FREE
TIM7: FREE
TIM8:
    CH1: DSHOT_BITBANG 1
    CH2: DSHOT_BITBANG 2
TIM9: FREE
TIM10: FREE
TIM11:
    CH1: CAMERA_CONTROL
TIM12: FREE
TIM13: FREE
TIM14: FREE

Currently active DMA:
--------------------
DMA1 Stream 0: FREE
DMA1 Stream 1: FREE
DMA1 Stream 2: FREE
DMA1 Stream 3: FREE
DMA1 Stream 4: FREE
DMA1 Stream 5: FREE
DMA1 Stream 6: FREE
DMA1 Stream 7: FREE
DMA2 Stream 0: FREE
DMA2 Stream 1: FREE
DMA2 Stream 2: DSHOT_BITBANG 1
DMA2 Stream 3: DSHOT_BITBANG 2
DMA2 Stream 4: ADC 1
DMA2 Stream 5: FREE
DMA2 Stream 6: FREE
DMA2 Stream 7: FREE

# 
'''

Diatone Mamba F405 Mamaba Stack

haslinghuis, 2020-06-14:
Did not test but which [configurator](https://github.com/betaflight/betaflight-configurator/releases) version are you using?

mikeller, 2020-06-14:
@marciano027:

> `set max7456_spi_bus = 0`

For whatever reason you are disabling the OSD in your configuration, this is working exactly as expected.

marciano027, 2020-06-14:
> Did not test but which [configurator](https://github.com/betaflight/betaflight-configurator/releases) version are you using? Also please don't import settings from previous versions.

Using BF configurator 10.7.0 RC2
Tried flashing again full chip erase same result, on 2 quads with the same board.

New diff: 

'''
# >
	reset
	load <mixer>
	reverse <servo> <source> r|n
status - show status
tasks - show task stats
timer - show/set timers
	<> | <pin> list | <pin> [af<alternate function>|none|<option(deprecated)>] | list | show
version - show version
vtx - vtx channels on switch
	<index> <aux_channel> <vtx_band> <vtx_channel> <vtx_power> <start_range> <end_range>
vtx_info - vtx power config dump
vtxtable - vtx frequency table
	<band> <bandname> <bandletter> [FACTORY|CUSTOM] <freq> ... <freq>

# # 0.5903776468856405

# diff

# version
# Betaflight / STM32F405 (S405) 4.2.0 Jun  3 2020 / 11:51:11 (9718914fd) MSP API: 1.43
# config: manufacturer_id: DIAT, board_name: FURYF4OSD, version: 3a35e73b, date: 2019-09-30T05:46:12Z

# start the command batch
batch start

board_name FURYF4OSD
manufacturer_id DIAT

# feature
feature TELEMETRY

# serial
serial 0 64 115200 57600 0 115200
serial 2 0 115200 57600 0 115200
serial 5 2 115200 57600 0 115200

# master
set acc_calibration = -54,39,-374,1
set serialrx_provider = CRSF
set dshot_bidir = ON
set motor_pwm_protocol = DSHOT300
set pid_process_denom = 2
set osd_vbat_pos = 14570
set gyro_1_align_yaw = 1800

profile 0

rateprofile 0

# end the command batch
batch end

# 
'''

etracer65, 2020-06-14:
Connect the battery (REMOVE PROPS!) **before** connecting the USB. The message at the top of the OSD tab is telling you this information.

marciano027, 2020-06-14:
Sorry guys, today flashed relase 4.2 working as expected TY

closed 2020-06-14 by marciano027