betaflight/betaflight #9916: Camera Control resource mapping missing in Kakute F7 Mini unified target configuration file

theely, 2020-06-15:
Kakute F7 Mini unified target configuration file is missing the resource mapping for Camera Control.
https://raw.githubusercontent.com/betaflight/unified-targets/master/configs/default/HBRO-KAKUTEF7MINI.config

The board has cam control pad as official documentation states:http://www.holybro.com/manual/HolybroKakuteF7miniManual.pdf

mikeller, 2020-06-15:
@theely: Do you have any idea what MCU pin is used for camera control - I could not find any manufacturer documentation for it.
If not, then maybe try `resource CAMERA_CONTROL 1 PE13` and report back if successful.

theely, 2020-06-15:
No unfortunately I don't, I did reach out to Holybro but got no reply so far.
Tried to set resource CAMERA_CONTROL 1 E13 but it does not seem to work.

theely, 2020-06-16:
Got words from Holybro!
Here is what I had to chance to enable camera control:

resource CAMERA_CONTROL 1 B03
timer B03 AF1
dma pin B03 0

I would submit a fix but I can't find the HBRO-KAKUTEF7MINI.config in the repository.

dkustec, 2020-06-16:
@theely, you looking in the wrong project, tho.
Here: [https://github.com/betaflight/unified-targets/blob/master/configs/default/HBRO-KAKUTEF7MINI.config](url)

dkustec, 2020-06-16:
@theely, you looking in the wrong project, tho.
There it is: [HBRO-KAKUTEF7MINI.config](https://github.com/betaflight/unified-targets/blob/master/configs/default/HBRO-KAKUTEF7MINI.config)

dkustec, 2020-06-16:
@theely, you're looking for it in the wrong project, tho.
There it is: [HBRO-KAKUTEF7MINI.config](https://github.com/betaflight/unified-targets/blob/master/configs/default/HBRO-KAKUTEF7MINI.config)

theely, 2020-06-16:
Oh thanks!
This is as far as I can get: https://github.com/betaflight/unified-targets/pull/292
Comments need fixing!

dkustec, 2020-06-16:
@theely: try this: `https://github.com/betaflight/unified-targets/blob/master/configs/default/HBRO-KAKUTEF7MINI.config`

theely, 2020-06-16:
@dkustec what do you mean?
I was able to submit a PR to fix HBRO-KAKUTEF7MINI.config:
betaflight/unified-targets#292.
But the code change is incomplete some comments need fixing as I don't have all the required information. See betaflight/unified-targets#292 for more information.

dkustec, 2020-06-16:
@theely: Sorry, I thought you couldn't get to the config. My mistake.

theely, 2020-06-16:
@dkustec no problem at all! Just checking if I might missed something.

haslinghuis, 2022-04-02:
Fixed in https://github.com/betaflight/unified-targets/pull/292

closed 2022-04-02 by haslinghuis