betaflight/betaflight #9918: tekKo32 F3 4in1 B

MimNanopirate, 2020-06-16:
**Describe the bug**
when i am at low rpm and give full throttle the quad stop and drop
it's just falling out of the sky like it shuts off.

https://youtu.be/s030Vkzg6Bg

**To Reproduce**
Turn ON RPM filter 
Whit no RPM filter is ok

**Expected behavior**
Rpm filter work.

**Flight controller configuration**
 ``` # diff all

# version
# Betaflight / STM32F745 (S745) 4.2.0 Jun 14 2020 / 03:05:04 (8f2d21460) MSP API: 1.43
# config: manufacturer_id: HBRO, board_name: KAKUTEF7HDV, version: cc8e118f, date: 2020-03-04T02:37:20Z

# start the command batch
batch start

# reset configuration to default settings
defaults nosave

board_name KAKUTEF7HDV
manufacturer_id HBRO
mcu_id 001f00273538510b31373730
signature 

# resources
resource MOTOR 1 E09
resource MOTOR 2 B00
resource MOTOR 3 E11
resource MOTOR 4 B01

# feature
feature GPS

# beeper
beeper -GYRO_CALIBRATED
beeper -DISARMING
beeper -ARMING
beeper -ARMING_GPS_FIX
beeper -BAT_LOW
beeper -GPS_STATUS
beeper -ACC_CALIBRATION
beeper -ACC_CALIBRATION_FAIL
beeper -READY_BEEP
beeper -DISARM_REPEAT
beeper -ARMED
beeper -SYSTEM_INIT
beeper -ON_USB
beeper -BLACKBOX_ERASE
beeper -CRASH_FLIP
beeper -CAM_CONNECTION_OPEN
beeper -CAM_CONNECTION_CLOSE
beeper -RC_SMOOTHING_INIT_FAIL

# beacon
beacon RX_LOST
beacon RX_SET

# serial
serial 2 2 115200 57600 0 115200
serial 3 64 115200 57600 0 115200
serial 5 0 115200 57600 0 115200

# aux
aux 0 0 0 1300 2100 0 0
aux 1 46 1 1300 2100 0 0
aux 2 13 2 1300 2100 0 0

# master
set gyro_lowpass2_hz = 300
set dyn_lpf_gyro_min_hz = 240
set dyn_lpf_gyro_max_hz = 600
set acc_calibration = 79,119,-327,1
set baro_hardware = AUTO
set rssi_channel = 12
set serialrx_provider = CRSF
set blackbox_p_ratio = 0
set blackbox_device = NONE
set dshot_bidir = ON
set failsafe_procedure = GPS-RESCUE
set ibata_scale = 240
set yaw_motors_reversed = ON
set small_angle = 180
set gps_provider = UBLOX
set gps_sbas_mode = EGNOS
set gps_ublox_use_galileo = ON
set gps_rescue_initial_alt = 30
set gps_rescue_descent_dist = 50
set gps_rescue_sanity_checks = RESCUE_SANITY_FS_ONLY
set gps_rescue_min_sats = 6
set deadband = 3
set yaw_deadband = 3
set osd_rssi_alarm = 75
set osd_cap_alarm = 1400
set osd_alt_alarm = 1000
set osd_tim1 = 1537
set osd_vbat_pos = 2246
set osd_rssi_pos = 2262
set osd_tim_1_pos = 2487
set osd_flymode_pos = 2434
set osd_throttle_pos = 409
set osd_current_pos = 2342
set osd_mah_drawn_pos = 2310
set osd_gps_speed_pos = 2325
set osd_gps_lon_pos = 2497
set osd_gps_lat_pos = 2512
set osd_gps_sats_pos = 2466
set osd_home_dir_pos = 2192
set osd_home_dist_pos = 2158
set osd_altitude_pos = 2291
set osd_warnings_pos = 2474
set osd_avg_cell_voltage_pos = 2278
set osd_stat_tim_1 = ON
set osd_stat_tim_2 = OFF
set osd_stat_max_dist = ON
set osd_stat_endbatt = ON
set osd_stat_max_alt = ON
set osd_stat_flight_dist = ON
set osd_stat_total_time = ON
set osd_stat_total_dist = ON
set gyro_1_align_yaw = 2700

profile 0

# profile 0
set dyn_lpf_dterm_min_hz = 84
set dyn_lpf_dterm_max_hz = 204
set dterm_lowpass2_hz = 180
set ff_interpolate_sp = AVERAGED_3

profile 1

profile 2

# restore original profile selection
profile 0

rateprofile 0

# rateprofile 0
set roll_rc_rate = 110
set pitch_rc_rate = 110
set yaw_rc_rate = 110
set roll_expo = 10
set pitch_expo = 10
set yaw_expo = 10

rateprofile 1

rateprofile 2

rateprofile 3

rateprofile 4

rateprofile 5

# restore original rateprofile selection
rateprofile 0 ```

**Setup / Versions**
 - Holybro F7 HDV and Tekko32 F3 4in1 B;
 - DJI Vista

**Additional context**
issue whit betaflight 4.1 et 4.2 BLheli 32.7

haslinghuis, 2020-06-16:
It looks like you imported a `diff` from a previous version? Please try with fresh config, apply defaults after flashing and enable `motor_pwm_protocol` in the configuration tab.

haslinghuis, 2020-06-16:
Please provide a blackbox log; this could possible be a single motor freaking out.

etracer65, 2020-06-16:
That's called a de-sync and is a problem with the ESCs and motors. Sometimes you can adjust settings in BLHeli to improve compatibility between the motors and ESC. Suggest you look for BLHeli32 support groups like: https://www.rcgroups.com/forums/showthread.php?2864611-BLHeli_32-Power-to-perform/page261

You can also try increasing the digital idle % on the Configuration tab as that might help some.

closed 2020-06-16 by etracer65

etracer65, 2020-06-16:
And there seem to be other people reporting problems with this particular ESC with failures in regards to bidirectional DSHOT. So I would suggest posting there with more feedback so the BLHeli32 developers know about the problem. You could also raise a bug report here: https://github.com/bitdump/BLHeli/issues