betaflight/betaflight #9919: Motor smoked on reboot after updating to 4.1

FPVShadow, 2020-06-16:
I'm hoping someone can help me. I just updated to 4.1 on my matekf405 osd. I have crossfire and noticed no input from my tx even though the receiver is bound. I went to the config tab and checked the protocol was  right(it was). I then went to the ports to check it was on the right uart. when I changed uart the quad restarted on it's own and 2 motors spun up so fast that one smoked. I cannot find a solution or explanation for this anywhere. P.S there was a battery connected so my tx would connect.

**Important: This is not the right place to get help with configuration issues. Your issue will be closed without further comment.**

Please get help from other user support options such as asking the manufacturer of the hardware you are using, RCGroups: https://rcgroups.com/forums/showthread.php?t=2464844, or Slack (registration at https://slack.betaflight.com/) or other user support forums & groups (e.g. Facebook).

augustozanellato, 2020-06-18:
(Not BF team member here)

It sounds more like it's an issue with your ESCs (or maybe motors), even if BF drove the motors to full power they shouldn't just smoke and die

FPVShadow, 2020-06-18:
I found out what caused it  switching ports with a battery connected

augustozanellato, 2020-06-18:
Not one of the safest things to do but that shouldn't smoke your motors

closed 2020-06-22 by no-response[bot]