betaflight/betaflight #9920: BF 4.2 AKK patch

alexdibi, 2020-06-16:
will the akk patched version of 4.2 ever be there?
now that there are vtx tables, wouldn't it make more sense to add an option directly using the main version of betaflight?

alexdibi, 2020-06-16:
👍

etracer65, 2020-06-16:
Duplicate of https://github.com/betaflight/betaflight/issues/9885

closed 2020-06-16 by etracer65

Diaoul, 2020-06-16:
I think the first question is unanswered, AKK patch for 4.2 isn't available yet. Any way to help out on this part?

etracer65, 2020-06-16:
Have patience.

mikeller, 2020-06-17:
@Diaoul: Not really sure what is going wrong here - the build for the patched version happened within an hour of the 4.2 release, something seems to be broken with the new version being found by configurator:

[image]

mikeller, 2020-06-17:
Gah, the curse of the rubber duck debugger hit again, fix identified and requested. (The hex file names produced by CI are wrong.)

mikeller, 2020-06-17:
Fixed now:
[image]