betaflight/betaflight #9921: Create manufacturer ID for Eachine

propson, 2020-06-16:
I see there is no unified target for the Eachine F405 FC which uses the old target MATEKF405.
Please create an id so that I can implement this into a target config.

Website: https://www.eachine.com/

Thanks

mikeller, 2020-06-21:
@propson: Are you a representative of eachine, and does this mean that Eachine will finally provide targets for hardware that was supplied without publicly available targets for it?

propson, 2020-06-23:
@mikeller: No I'm not a representative of Eachine. I cannot say what plans Eachine has or doesn't have. I'm simply a Betaflight user and I like to have a proper target for the FC I have in the Tyro 109. Existing targets aren't correct. The MATEK F405 STD comes close but isn't 100 correct. Gyro alignment isn't correct for example. I read the wiki about creating a config. I thought I could contribute by creating a config for this FC. The wiki mentioned I should create a ticket for getting the manufacturer id. So here I am.

mikeller, 2020-06-23:
@propson: The idea behind the manufacturer registration is that this should be used by the actual manufacturer, or somebody representing them or working with them and having access to detailed information like schematics for their hardware. And the URL should be an URL where users can go and get support for running Betaflight on this hardware.
So not sure that adding 'Eachine' as a manufacturer is the right thing to do if they are not willing to start supporting their users, or at least do the minimum to be compliant with the license of Betaflight and provide the sources for the modified versions of the firmware that they are distributing with their hardware.
Any community provided targets without support from Eachine could probably be added under the 'Legacy' manufacturer, or we could add a new generic manufacturer 'Community provided targets without manufacturer support' for these targets.

propson, 2020-06-24:
@mikeller I totally understand your point of view. 
I'm looking at it from the community. Our hobby has a pretty steep learning curve. We know that people often start their FPV hobby with cheap quads which might have 'clone' FC boards. I've see in the Facebook groups multiple people that think there is no BF 4.2 for their FC. Mainly because they were using MATEKF405 up to 4.1.1 and that one is considered legacy.

You final suggestion about 'community provided targets' could be a good idea.
I already created the file for the Tyro109. What I now only need is a place to store it so that other user can benefit from this.

haslinghuis, 2020-06-25:
I think the way to go is just ask Eachine for support. I have sent a request for creating an ID. If Eachine wants to be taken serious they will respond and looking at their site they are going forward.

propson, 2020-06-28:
@haslinghuis Good point. Let's see how this turns out.

mikeller, 2020-06-28:
Yeah, let's hope the best. For now we can use 'COMM', maybe if they EACHINE sees that people care about this they will come to the table.

haslinghuis, 2020-07-02:
No response from EACHINE. Sent another request. Thanks @mikeller for adding a community edition.

mikeller, 2020-07-06:
@propson: betaflight/unified-targets#300 is done now, you can use this for your target configuration.

@haslinghuis: Good luck. I have tried to contact them a number of times in the past. In some cases this was for clear breaches of Betaflight's license, where they are shipping hardware with custom builds of Betaflight without providing the source, which results in their users being stuck with insta-legacy versions of the firmware. I never got a single reaction from them, or any attempt to stop breaking our license.

haslinghuis, 2020-07-06:
😞 Yeah no reaction at all 👎 shame on Eachine. Still I like my tyro79 on a 4" armattan productions community frame.

propson, 2020-07-06:
@haslinghuis Too bad.
@mikeller I created the config and created a Pull Request. I'm still new to Github so I hope I've done it in a correct manner.

haslinghuis, 2020-07-09:
#10002 :)

closed 2020-08-16 by stale[bot]