betaflight/betaflight #9924: Stab and Horizon missing in betaflight 4.2 config and not working on OSD

prohelper, 2020-06-17:
The settings for Stabelize and Horizon are completely gone from the "mode" tab in betaflight configurator 10.7 and the indication on OSD is not working either. it just says "ACRO" either way. Not sure if this should be under betaflight configurator bugs or not but thought it was worth mentioning here. I am using DJI's system on an Iflight DC5 with IFF7_TWIN_G board. 

**To Reproduce**

**Expected behavior**
It should change on OSD when i switch form "acro" to "stab" or "acro" to "horizon". 

**Flight controller configuration**
```# 

# diff

# version
# Betaflight / STM32F7X2 (S7X2) 4.2.0 Jun  3 2020 / 11:52:14 (9718914fd) MSP API: 1.43
# config: manufacturer_id: IFRC, board_name: IFLIGHT_F722_TWING, version: bc19b7dc, date: 2020-04-01T04:37:12Z

# start the command batch
batch start

board_name IFF7_TWIN_G
manufacturer_id IFRC

# feature
feature -SOFTSERIAL
feature GPS

# beacon
beacon RX_LOST
beacon RX_SET

# serial
serial 0 64 115200 57600 0 115200
serial 1 1 115200 57600 0 115200
serial 3 2 115200 9600 0 115200

# aux
aux 0 0 3 1400 1600 0 0
aux 1 46 1 1700 2100 0 0
aux 2 46 1 900 1300 0 0
aux 3 13 2 1700 2100 0 0
aux 4 13 2 900 1300 0 0

# rxfail
rxfail 4 s 750
rxfail 5 s 2250

# master
set dyn_notch_width_percent = 0
set dyn_notch_q = 200
set dyn_notch_min_hz = 100
set dyn_notch_max_hz = 500
set acc_trim_pitch = -6
set acc_trim_roll = -2
set acc_calibration = 22,73,7,1
set mag_hardware = NONE
set rc_smoothing_derivative_type = BIQUAD
set rc_smoothing_auto_smoothness = 40
set sbus_baud_fast = ON
set failsafe_delay = 0
set failsafe_switch_mode = STAGE2
set failsafe_procedure = GPS-RESCUE
set beeper_dshot_beacon_tone = 5
set yaw_motors_reversed = ON
set small_angle = 100
set gps_provider = UBLOX
set gps_sbas_mode = AUTO
set gps_auto_baud = ON
set gps_ublox_use_galileo = ON
set gps_rescue_initial_alt = 60
set gps_rescue_descent_dist = 30
set gps_rescue_min_sats = 5
set osd_alt_alarm = 200
set osd_flymode_pos = 14498
set osd_mah_drawn_pos = 14829
set osd_gps_speed_pos = 14761
set osd_gps_lon_pos = 14368
set osd_gps_lat_pos = 14336
set osd_gps_sats_pos = 14530
set osd_home_dir_pos = 14414
set osd_home_dist_pos = 14446
set osd_flight_dist_pos = 47
set osd_altitude_pos = 14764
set osd_warnings_pos = 14697
set osd_esc_tmp_pos = 14594
set osd_stat_max_alt = ON
set gyro_2_align_yaw = 900
set gyro_rpm_notch_q = 900

profile 0

# profile 0
set dyn_lpf_dterm_curve_expo = 8
set feedforward_transition = 70
set iterm_relax_cutoff = 5
set iterm_windup = 75
set yaw_lowpass_hz = 50
set throttle_boost = 2
set throttle_boost_cutoff = 10
set p_yaw = 30
set ff_interpolate_sp = AVERAGED_4
set ff_spike_limit = 50
set ff_smooth_factor = 50
set ff_boost = 0

rateprofile 0

# rateprofile 0
set roll_expo = 10
set pitch_expo = 10
set yaw_expo = 10
set tpa_breakpoint = 1250

# end the command batch
batch end ``` 

**Setup / Versions**
 IFF7_TWIN_G from iflight
 DJI Fpv system, GPS
 Dji on UART2, GPS on UART4

haslinghuis, 2020-06-17:
Duplicate of #9883

haslinghuis, 2020-06-17:
This issue was recently fixed in #9884 and got into the BF4.2.0 release.
Please reflash the 4.2 release on Configurator 10.7.0

closed 2020-06-17 by etracer65