betaflight/betaflight #9926: XSR in Mamba F405 mkII Softserial and Smartport

ts-dr1, 2020-06-17:
[attention: newbie alert]

In BF 4.1.6 I had full telemetry from my old frsky XSR receiver, using Mamba f405 mkII, SBUS, softserial and smartport from the ppm port of the fc...

Then updated to BF4.2 and can´t make the fc transmit all sensors.

tried what used to work:
resource show all
(A15 is FREE)
resource motor 5 none
resource serial_tx 11 A15
save

then made Telemetry tab and Softserial tab both ON
On Ports: Softserial transmitter protocol Smartport
save

No new sensors from fc found on QX7S

then all four combinations with
tlm_inverted and tlm_halfduplex...

And also nothing... all basic 3 sensors from the receiver and none from the FC board.

How newbie am I?

Regards,

ts-dr1, 2020-06-17:
Sorry, what "Missing Information" means?
Should I dump the config here?
Why it worked in 4.1.6 and stoped in 4.2?

by the way... 4.2 rocks.

etracer65, 2020-06-17:
> **Describe the bug**
> A clear and concise description of what the bug is.
> 
> **To Reproduce**
> Steps to reproduce the behavior:
> 
> **Expected behavior**
> A clear and concise description of what you expected to happen.
> 
> **Flight controller configuration**
> Create a `diff` and post it here in a code block. Put ``` (three backticks) at the start and end of the `diff` block (instructions  on how to do a diff: https://oscarliang.com/use-diff-not-dump-betaflight/)
> 
> Use `resource show all` to create a resource allocation list and post it here in a code block. Put ``` (three backticks) at the start and end of the output block.
> 
> **Setup / Versions**
>  - Flight controller [what type is it, where was it bought from];
>  - Other components [RX, VTX, brand / model for all of them, firmware version where applicable];
>  - how are the different components wired up.
> 
> **Additional context**
> Add any other context about the problem here.
>

ts-dr1, 2020-06-17:
Describe the bug
Telemetry from FC sensors not sent to RC trough FrSky XSR. 
Using Softserial set to Tx Smartport protocol.

To Reproduce
1. Installed new BF 4.2, full chip erase.
2. On Ports defined UART1 as Rx. No SOFTSERIAL1 avaliable. Save & reboot.
3. On Configuration defined SBUS RC protocol. Defined Telemetry tab ON. Save & reboot.
4. Plugged 4S battery.
5. On Receptor, BF is receiving RC input, as normal.
6. On CLI set:
resource motor 5 none
resource SERIAL_TX 11 A15
save & reboot.
7. On Configuration defined Softserial tab ON. Save & reboot.
8. On Ports defined Telemetry Output Protocol to Smartport on SOFTSERIAL1. Save and Reboot.
9. On RC QX7S (opentx 2.3.7), page 11 TELEMETRY, just apears Receiver sensors: A2, RSSI and RxBt
10. Clicked "Discover new sensors". No new sensors.
11. Back to BF and CLI:
Typed all 4 combinations of ON/OFF command tlm_inverted and tlm_halfduplex.
Save & reboot for all 4 combinations.
12. No new sensors on TELEMETRY page of RC QX7S. 

Expected behavior
Expected that all available sensors data of FC sent to RC. 

Flight controller configuration
``` # diff

# version
# Betaflight / STM32F405 (S405) 4.2.0 Jun 14 2020 / 03:04:22 (8f2d21460) MSP API: 1.43
# config: manufacturer_id: DIAT, board_name: FURYF4OSD, version: 3a35e73b, date: 2019-09-30T05:46:12Z

# start the command batch
batch start

board_name FURYF4OSD
manufacturer_id DIAT

# name: ts_dr1><5diy

# resources
resource SERIAL_TX 11 A15

# feature
feature SOFTSERIAL
feature TELEMETRY

# beacon
beacon RX_LOST
beacon RX_SET

# serial
serial 0 64 115200 57600 0 115200
serial 2 0 115200 57600 0 115200
serial 30 32 115200 57600 0 115200

# aux
aux 0 0 0 1700 2100 0 0
aux 1 1 1 1700 2100 0 0
aux 2 13 2 1300 1700 0 0
aux 3 19 1 1300 1700 0 0
aux 4 35 2 1650 2100 0 0

# master
set gyro_lowpass2_hz = 325
set dyn_notch_width_percent = 1
set dyn_notch_q = 200
set dyn_notch_min_hz = 180
set dyn_notch_max_hz = 500
set dyn_lpf_gyro_min_hz = 260
set dyn_lpf_gyro_max_hz = 650
set acc_calibration = -131,34,239,1
set rssi_channel = 8
set rssi_src_frame_errors = OFF
set rc_smoothing_derivative_type = BIQUAD
set blackbox_device = NONE
set dshot_idle_value = 550
set dshot_bidir = ON
set motor_pwm_protocol = DSHOT300
set align_board_yaw = 180
set yaw_motors_reversed = ON
set small_angle = 60
set pid_process_denom = 2
set tlm_inverted = ON
set tlm_halfduplex = OFF
set osd_units = 
###ERROR: diff: CORRUPTED CONFIG: osd_units = 33###
set osd_rssi_alarm = 0
set osd_cap_alarm = 0
set osd_alt_alarm = 1300
set osd_rssi_pos = 14647
set osd_flymode_pos = 14786
set osd_throttle_pos = 14628
set osd_craft_name_pos = 14794
set osd_pidrate_profile_pos = 14810
set osd_warnings_pos = 329
set osd_avg_cell_voltage_pos = 14732
set gyro_1_sensor_align = DEFAULT
set gyro_1_align_yaw = 1800
set name = ts_dr1><5diy

profile 0

# profile 0
set dyn_lpf_dterm_min_hz = 91
set dyn_lpf_dterm_max_hz = 221
set dterm_lowpass2_hz = 195
set p_pitch = 60
set d_pitch = 49
set p_roll = 55
set d_roll = 46
set p_yaw = 59
set d_min_roll = 30
set d_min_pitch = 33

rateprofile 0

# rateprofile 0
set roll_rc_rate = 170
set pitch_rc_rate = 170
set yaw_rc_rate = 170
set roll_srate = 65
set pitch_srate = 65
set tpa_breakpoint = 1300

# end the command batch
batch end ```

```# resource show all
Currently active IO resource assignments:
(reboot to update)
--------------------
A00: FREE
A01: FREE
A02: MOTOR 4
A03: MOTOR 1
A04: GYRO_CS 1
A05: SPI_SCK 1
A06: SPI_MISO 1
A07: SPI_MOSI 1
A08: BEEPER
A09: FREE
A10: SERIAL_RX 1
A11: USB
A12: USB
A13: SWD
A14: SWD
A15: FREE
B00: MOTOR 2
B01: MOTOR 3
B02: FREE
B03: FLASH_CS
B04: LED 2
B05: LED 1
B06: I2C_SCL 1
B07: I2C_SDA 1
B08: FREE
B09: CAMERA_CONTROL
B10: FREE
B11: FREE
B12: OSD_CS
B13: SPI_SCK 2
B14: SPI_MISO 2
B15: SPI_MOSI 2
C00: INVERTER 1
C01: ADC_BATT
C02: FREE
C03: ADC_CURR
C04: GYRO_EXTI
C05: FREE
C06: FREE
C07: FREE
C08: FREE
C09: FREE
C10: SPI_SCK 3
C11: SPI_MISO 3
C12: SPI_MOSI 3
C13: FREE
C14: FREE
C15: FREE
D00: FREE
D01: FREE
D02: FREE
D03: FREE
D04: FREE
D05: FREE
D06: FREE
D07: FREE
D08: FREE
D09: FREE
D10: FREE
D11: FREE
D12: FREE
D13: FREE
D14: FREE
D15: FREE
E00: FREE
E01: FREE
E02: FREE
E03: FREE
E04: FREE
E05: FREE
E06: FREE
E07: FREE
E08: FREE
E09: FREE
E10: FREE
E11: FREE
E12: FREE
E13: FREE
E14: FREE
E15: FREE
F00: FREE
F01: FREE
F02: FREE
F03: FREE
F04: FREE
F05: FREE
F06: FREE
F07: FREE
F08: FREE
F09: FREE
F10: FREE
F11: FREE
F12: FREE
F13: FREE
F14: FREE
F15: FREE

Currently active Timers:
-----------------------
TIM1: FREE
TIM2: FREE
TIM3: FREE
TIM4: FREE
TIM5: FREE
TIM6: FREE
TIM7: FREE
TIM8:
    CH1: DSHOT_BITBANG 1
    CH2: DSHOT_BITBANG 2
TIM9: FREE
TIM10: FREE
TIM11:
    CH1: CAMERA_CONTROL
TIM12: FREE
TIM13: FREE
TIM14: FREE

Currently active DMA:
--------------------
DMA1 Stream 0: FREE
DMA1 Stream 1: FREE
DMA1 Stream 2: FREE
DMA1 Stream 3: FREE
DMA1 Stream 4: FREE
DMA1 Stream 5: FREE
DMA1 Stream 6: FREE
DMA1 Stream 7: FREE
DMA2 Stream 0: FREE
DMA2 Stream 1: FREE
DMA2 Stream 2: DSHOT_BITBANG 1
DMA2 Stream 3: DSHOT_BITBANG 2
DMA2 Stream 4: ADC 1
DMA2 Stream 5: FREE
DMA2 Stream 6: FREE
DMA2 Stream 7: FREE ```

Flight controller: 
Mamba F405 Mk2 from Banggood
Other components: 
RX: FrSky XSR.
Wired: 5v and GND. SBUS to SBUS pad. S.port to PPM pad.
VTX: Eachine VTX03.
Wired: 5V and GND. VTX in wired to VID pad

Mamba F405 MK2 pads below:
[https://cdn.shopify.com/s/files/1/0027/2708/4144/files/F405_MK2.png](url)

Softserial and Smartport worked fine in BF 4.1.6 using resource mapping described above.

etracer65, 2020-06-17:
Your config is corrupted.
```
set osd_units = 
###ERROR: diff: CORRUPTED CONFIG: osd_units = 33###
```
Reset to `defaults` and reconfigured from scratch.

Also there's something fishy with your resource mapping. Motor 5 is not showing released. Also A15 isn't associated with anything.
```
A15: FREE
```

Also not sure where you got A15 from in the first place. That pin is not even defined in the FURYF4OSD target. Also that target doesn't even define motor 5 let alone assign it to A15.