betaflight/betaflight #9927: Betaflight 4.2.0 Freezes up after crash

QuistFPV, 2020-06-18:
**Describe the bug**
A clear and concise description of what the bug is.
First flight after updating to 4.2.0 I crashed into a tree. The quad spun the motors wildly very briefly before everything want quiet. Everything had frozen-up OSD showed a constant 18A current draw and RSSI did not change. Furthermore it did not react on any input from the transmitter, like activate beeper, arm/diaarm and so on.

I thought it was just a one time glitch so I flew again. I crashed again (again my fault) and everything was again frozen-up.

**To Reproduce**
Just go out and crash your quad :-)
I am not sure what triggers this. As I understand Runaway detection is only active right after takeoff. 

**Expected behavior**
No freeze-up of the quad

**Flight controller configuration**
```
# diff

# version
# Betaflight / STM32F405 (S405) 4.2.0 Jun 14 2020 / 03:04:22 (8f2d21460) MSP API: 1.43
# config: manufacturer_id: DIAT, board_name: FURYF4OSD, version: 3a35e73b, date: 2019-09-30T05:46:12Z

# start the command batch
batch start

board_name FURYF4OSD
manufacturer_id DIAT

# name: GT-R349

# feature
feature -AIRMODE
feature TELEMETRY

# beeper
beeper -RX_LOST
beeper -DISARMING
beeper -ARMING
beeper -BAT_CRIT_LOW
beeper -BAT_LOW
beeper -GPS_STATUS
beeper -READY_BEEP
beeper -DISARM_REPEAT
beeper -ARMED
beeper -ON_USB
beeper -CAM_CONNECTION_OPEN
beeper -CAM_CONNECTION_CLOSE

# beacon
beacon RX_LOST
beacon RX_SET

# serial
serial 20 1 250000 57600 0 115200
serial 2 2048 115200 57600 0 115200
serial 5 64 115200 57600 0 115200

# aux
aux 0 0 0 1700 2100 0 0
aux 1 1 1 900 1300 0 0
aux 2 13 2 1700 2100 0 0
aux 3 26 3 1700 2100 0 0
aux 4 47 1 1300 1700 0 0

# vtxtable
vtxtable bands 6
vtxtable channels 8
vtxtable band 1 BOSCAM_A A FACTORY 5865 5845 5825 5805 5785 5765 5745 5725
vtxtable band 2 BOSCAM_B B FACTORY 5733 5752 5771 5790 5809 5828 5847 5866
vtxtable band 3 BOSCAM_E E FACTORY 5705 5685 5665    0 5885 5905    0    0
vtxtable band 4 FATSHARK F FACTORY 5740 5760 5780 5800 5820 5840 5860 5880
vtxtable band 5 RACEBAND R FACTORY 5658 5695 5732 5769 5806 5843 5880 5917
vtxtable band 6 IMD6     I CUSTOM  5732 5765 5828 5840 5866 5740    0    0
vtxtable powerlevels 3
vtxtable powervalues 0 1 2
vtxtable powerlabels 25 200 500

# master
set gyro_lowpass2_hz = 500
set dyn_notch_width_percent = 0
set dyn_notch_q = 250
set dyn_notch_min_hz = 60
set dyn_notch_max_hz = 300
set dyn_lpf_gyro_min_hz = 400
set dyn_lpf_gyro_max_hz = 1000
set acc_calibration = -23,-32,33,1
set min_check = 1010
set max_check = 1990
set rc_smoothing_derivative_type = PT1
set serialrx_provider = SRXL2
set dshot_idle_value = 550
set dshot_bidir = ON
set motor_pwm_protocol = DSHOT300
set motor_poles = 12
set bat_capacity = 650
set vbat_max_cell_voltage = 420
set ibata_scale = 450
set ibatv_scale = 153
set beeper_dshot_beacon_tone = 5
set small_angle = 180
set yaw_deadband = 1
set pid_process_denom = 2
set osd_warn_core_temp = OFF
set osd_cap_alarm = 550
set osd_logo_on_arming = ON
set osd_tim2 = 769
set osd_vbat_pos = 449
set osd_rssi_pos = 2458
set osd_tim_2_pos = 2519
set osd_flymode_pos = 2489
set osd_vtx_channel_pos = 408
set osd_current_pos = 2464
set osd_mah_drawn_pos = 2508
set osd_craft_name_pos = 54
set osd_pidrate_profile_pos = 409
set osd_warnings_pos = 2377
set osd_avg_cell_voltage_pos = 2497
set osd_stat_max_spd = OFF
set osd_stat_battery = ON
set osd_stat_max_g_force = ON
set debug_mode = DYN_IDLE
set vtx_band = 4
set vtx_channel = 1
set vtx_power = 2
set vtx_low_power_disarm = ON
set vtx_freq = 5740
set vtx_pit_mode_freq = 5740
set gyro_1_align_yaw = 1800
set gyro_rpm_notch_q = 600
set name = GT-R349

profile 2

# profile 2
set dyn_lpf_dterm_min_hz = 140
set dyn_lpf_dterm_max_hz = 340
set dyn_lpf_dterm_curve_expo = 7
set dterm_lowpass2_hz = 300
set vbat_pid_gain = ON
set vbat_sag_compensation = 100
set anti_gravity_gain = 7000
set feedforward_transition = 35
set pidsum_limit = 1000
set pidsum_limit_yaw = 500
set acro_trainer_angle_limit = 15
set p_pitch = 32
set i_pitch = 63
set d_pitch = 27
set f_pitch = 93
set p_roll = 29
set i_roll = 59
set d_roll = 25
set f_roll = 88
set p_yaw = 31
set i_yaw = 63
set f_yaw = 88
set d_min_roll = 16
set d_min_pitch = 18
set ff_interpolate_sp = AVERAGED_3
set idle_min_rpm = 15

rateprofile 1

# rateprofile 1
set thr_mid = 15
set thr_expo = 40
set rates_type = ACTUAL
set roll_rc_rate = 20
set pitch_rc_rate = 20
set yaw_rc_rate = 20
set roll_expo = 54
set pitch_expo = 54
set yaw_expo = 54
set roll_srate = 67
set pitch_srate = 67
set yaw_srate = 67

# end the command batch
batch end
```

```
# resource show all
Currently active IO resource assignments:
(reboot to update)
--------------------
A00: FREE
A01: FREE
A02: MOTOR 4
A03: MOTOR 1
A04: GYRO_CS 1
A05: SPI_SCK 1
A06: SPI_MISO 1
A07: SPI_MOSI 1
A08: BEEPER
A09: FREE
A10: FREE
A11: USB
A12: USB
A13: SWD
A14: SWD
A15: FREE
B00: MOTOR 2
B01: MOTOR 3
B02: FREE
B03: FLASH_CS
B04: LED 2
B05: LED 1
B06: I2C_SCL 1
B07: I2C_SDA 1
B08: FREE
B09: CAMERA_CONTROL
B10: SERIAL_TX 3
B11: FREE
B12: OSD_CS
B13: SPI_SCK 2
B14: SPI_MISO 2
B15: SPI_MOSI 2
C00: INVERTER 1
C01: ADC_BATT
C02: FREE
C03: ADC_CURR
C04: GYRO_EXTI
C05: FREE
C06: SERIAL_TX 6
C07: FREE
C08: FREE
C09: FREE
C10: SPI_SCK 3
C11: SPI_MISO 3
C12: SPI_MOSI 3
C13: FREE
C14: FREE
C15: FREE
D00: FREE
D01: FREE
D02: FREE
D03: FREE
D04: FREE
D05: FREE
D06: FREE
D07: FREE
D08: FREE
D09: FREE
D10: FREE
D11: FREE
D12: FREE
D13: FREE
D14: FREE
D15: FREE
E00: FREE
E01: FREE
E02: FREE
E03: FREE
E04: FREE
E05: FREE
E06: FREE
E07: FREE
E08: FREE
E09: FREE
E10: FREE
E11: FREE
E12: FREE
E13: FREE
E14: FREE
E15: FREE
F00: FREE
F01: FREE
F02: FREE
F03: FREE
F04: FREE
F05: FREE
F06: FREE
F07: FREE
F08: FREE
F09: FREE
F10: FREE
F11: FREE
F12: FREE
F13: FREE
F14: FREE
F15: FREE

Currently active Timers:
-----------------------
TIM1: FREE
TIM2: FREE
TIM3: FREE
TIM4: FREE
TIM5: FREE
TIM6: FREE
TIM7: FREE
TIM8:
    CH1: DSHOT_BITBANG 1
    CH2: DSHOT_BITBANG 2
TIM9: FREE
TIM10: FREE
TIM11:
    CH1: CAMERA_CONTROL
TIM12: FREE
TIM13: FREE
TIM14: FREE

Currently active DMA:
--------------------
DMA1 Stream 0: FREE
DMA1 Stream 1: FREE
DMA1 Stream 2: FREE
DMA1 Stream 3: FREE
DMA1 Stream 4: FREE
DMA1 Stream 5: FREE
DMA1 Stream 6: FREE
DMA1 Stream 7: FREE
DMA2 Stream 0: FREE
DMA2 Stream 1: FREE
DMA2 Stream 2: DSHOT_BITBANG 1
DMA2 Stream 3: DSHOT_BITBANG 2
DMA2 Stream 4: ADC 1
DMA2 Stream 5: FREE
DMA2 Stream 6: FREE
DMA2 Stream 7: FREE
```

**Setup / Versions**
Diatone GT-R349 with TBS Unify, everything stock, with a SPM4650 receiver connected to UART6.

**Additional context**
-

etracer65, 2020-06-18:
This is generally caused by a power problem. Either the voltage sagging too low due to a wiring problem or bad battery, or due to voltage spikes caused by the ESCs. In both cases the power problem can cause the microcontroller to lock up. So fo the first case check wiring, solder joints, possibly replace worn out batteries. In the second look for damaged ESCs (missing components) or try adding capacitors.

> **Important: This is not the right place to report faulty / broken hardware. Your issue will be closed without further comment.**
> 
> Contact the manufacturer or supplier of your hardware, or check RCGroups https://rcgroups.com/forums/showthread.php?t=2464844 to see if others with the same problem have found a solution. Also try our Slack https://slack.betaflight.com or Facebook group https://www.facebook.com/groups/betaflightgroup/

closed 2020-06-18 by etracer65

QuistFPV, 2020-06-18:
Thank you for your replay. I will of course try to look for a possible HW fault, but I do not think that is the problem(which is also why I posted here)
I have been flying betaflight 4.0.3 and 4.1.7 without problems, and the problem came after upgrading to 4.2.0.

stackola, 2020-07-16:
I have the same issue with the exact same hard-/software combination.

Mikepyt87, 2022-06-17:
Was this issue ever resolved for you? I am having the same exact thing happen, 100% as you describe with the new NewBeeDrone AIO V2.