betaflight/betaflight #9928: BetaFlight Configurator will not communicate via USB with LUXMINIF7 past BetaFlight 4.1.6

Sprokcet, 2020-06-18:
**Describe the bug**
A clear and concise description of what the bug is.
Betaflight configurator will not communicate via usb with LUXMINIF7 past BetaFlight 4.1.6. After upgrading to 4.2 the BetaFlight configurator would not communicate to FC. Tested all firmware between 4.1.6 and 4.2. The problem seems to start around 4.1.7 for LUXMINIF7.
**To Reproduce**
Install BetaFlight 4.1.7 or greater for the problem to be reproduced.

**Expected behavior**
Resolve the perceived bug in a later release of BetaFlight 4.2.x

**Flight controller configuration**
Baseline configuration

Use `resource show all` to create a resource allocation list and post it here in a code block. Put ``` (three backticks) at the start and end of the output block.

**Setup / Versions**
 - Flight controller - Lumenier LUX Mini F7 20x20 Stack from GetFPV
 - Other components - Standalone FC Problem

**Additional context**
Reloaded all driver supporting Betaflight. Computer used for BetaFlight configurator is Windows 10

Sprokcet, 2020-06-18:
👍

etracer65, 2020-06-18:
I tested with a prototype LUXMINIF7 supplied by GetFPV and upgrading to BF4.2 works with no problems. Suggest you raise an issue with GetFPV and ask them to review their target config they supplied in light of any hardware revisions that may have been made to the board.

closed 2020-06-22 by no-response[bot]