betaflight/betaflight #9929: Bidirectional DSHOT and ESC_SENSOR conflict?

Diaoul, 2020-06-18:
**Describe the bug**
* With bidirectionnal DSHOT enabled and ESC_SENSOR disabled I have no error as I increase the throttle in the Motor tab: 0.00%
* With bidirectionnal DSHOT enabled and ESC_SENSOR enabled I have no some errors as I increase the throttle in the Motor tab: 0.05% to 0.30%

**To Reproduce**
Do the above

**Expected behavior**
Same error rate 0.00%

**Flight controller configuration**
```# diff

# version
# Betaflight / STM32F7X2 (S7X2) 4.2.0 Jun 17 2020 / 07:01:41 (604ba5d01) MSP API: 1.43
# config: manufacturer_id: MTKS, board_name: MATEKF722, version: f173f9bc, date: 2019-10-02T14:43:13Z

# start the command batch
batch start

board_name MATEKF722
manufacturer_id MTKS

# name: ChameleonTi

# resources
resource MOTOR 6 NONE
resource CAMERA_CONTROL 1 A08

# feature
feature GPS
feature ESC_SENSOR

# serial
serial 1 2 115200 57600 0 115200
serial 2 64 115200 57600 0 115200
serial 3 1024 115200 57600 0 115200
serial 4 2048 115200 57600 0 115200

# aux
aux 0 0 0 1700 2100 0 0
aux 1 1 1 1700 2100 0 0
aux 2 2 1 1300 1700 0 0
aux 3 13 2 1700 2100 0 0
aux 4 35 2 1300 1700 0 0

# master
set acc_calibration = 15,12,-29,1
set rssi_channel = 8
set serialrx_provider = FPORT
set serialrx_inverted = ON
set serialrx_halfduplex = ON
set min_throttle = 1070
set dshot_bidir = ON
set use_unsynced_pwm = OFF
set motor_pwm_rate = 480
set camera_control_key_delay = 110
set gyro_1_align_yaw = 1800

profile 0

rateprofile 0

# end the command batch
batch end
```

```
# resource show all
Currently active IO resource assignments:
(reboot to update)
--------------------
A00: FREE
A01: SERIAL_RX 4
A02: SERIAL_TX 2
A03: SERIAL_RX 2
A04: FREE
A05: SPI_SCK 1
A06: SPI_MISO 1
A07: SPI_MOSI 1
A08: CAMERA_CONTROL
A09: FREE
A10: FREE
A11: USB
A12: USB
A13: SWD
A14: LED 2
A15: FREE
B00: FREE
B01: FREE
B02: FREE
B03: SPI_SCK 3
B04: SPI_MISO 3
B05: SPI_MOSI 3
B06: I2C_SCL 1
B07: I2C_SDA 1
B08: FREE
B09: LED 1
B10: OSD_CS
B11: FREE
B12: FREE
B13: SPI_SCK 2
B14: SPI_MISO 2
B15: SPI_MOSI 2
C00: ADC_BATT
C01: SDCARD_CS
C02: GYRO_CS 1
C03: GYRO_EXTI
C04: ADC_CURR
C05: FREE
C06: MOTOR 1
C07: MOTOR 2
C08: MOTOR 3
C09: MOTOR 4
C10: SERIAL_TX 3
C11: FREE
C12: SERIAL_TX 5
C13: BEEPER
C14: FREE
C15: FREE
D00: FREE
D01: FREE
D02: FREE
D03: FREE
D04: FREE
D05: FREE
D06: FREE
D07: FREE
D08: FREE
D09: FREE
D10: FREE
D11: FREE
D12: FREE
D13: FREE
D14: FREE
D15: FREE
E00: FREE
E01: FREE
E02: FREE
E03: FREE
E04: FREE
E05: FREE
E06: FREE
E07: FREE
E08: FREE
E09: FREE
E10: FREE
E11: FREE
E12: FREE
E13: FREE
E14: FREE
E15: FREE
F00: FREE
F01: FREE
F02: FREE
F03: FREE
F04: FREE
F05: FREE
F06: FREE
F07: FREE
F08: FREE
F09: FREE
F10: FREE
F11: FREE
F12: FREE
F13: FREE
F14: FREE
F15: FREE

Currently active Timers:
-----------------------
TIM1:
    CH1: CAMERA_CONTROL
TIM2: FREE
TIM3: FREE
TIM4: FREE
TIM5: FREE
TIM6: FREE
TIM7: FREE
TIM8:
    CH1: DSHOT_BITBANG 3
TIM9: FREE
TIM10: FREE
TIM11: FREE
TIM12: FREE
TIM13: FREE
TIM14: FREE

Currently active DMA:
--------------------
DMA1 Stream 0: FREE
DMA1 Stream 1: FREE
DMA1 Stream 2: FREE
DMA1 Stream 3: FREE
DMA1 Stream 4: FREE
DMA1 Stream 5: FREE
DMA1 Stream 6: FREE
DMA1 Stream 7: FREE
DMA2 Stream 0: ADC
DMA2 Stream 1: FREE
DMA2 Stream 2: DSHOT_BITBANG 3
DMA2 Stream 3: FREE
DMA2 Stream 4: FREE
DMA2 Stream 5: FREE
DMA2 Stream 6: FREE
DMA2 Stream 7: FREE
```

**Setup / Versions**
 - FC Matek F722
 - ESC: AIRBOT Wraight32 V2 35A
- ESC signal wires twisted with ground and connected with the others on UART4

**Additional context**
N/A

etracer65, 2020-06-18:
The two functions are different and unrelated so there should be no conflict. However it's true that the MCU on the ESC is doing more work and therefore might be overloaded leading to packet errors. You could try running at a slower DSHOT speed to see if that helps. Also you could raise the issue the the BLHeli32 developers.

Diaoul, 2020-06-19:
Cannot the RPM reported by ESC telemetry be used for RPM filtering instead of using DSHOT ? This could avoid doing the work twice.
Is 0.30% error rate significant enough to cause some troubles in the RPM filtering, i.e. could I fly this without issues?

I tested with DSHOT300 and the error rate is now steady at 0.00% 👍 
At DSHOT300 is it useful still to run 8k/8k ?

McGiverGim, 2020-06-19:
No, the RPM reported by the ESC is very low rate, it is slow. It can't be used for RPM filtering.
Usually an error about 1% is not a problem, but if it is 0%, better.

Diaoul, 2020-06-19:
So I've tried to lower the PWM Frequency of the ESC from 48kHz to 24kHz and it didn't change a thing regarding the error rate.
If the load was the issue, shouldn't that relieve the ESC and allow it to report without errors?

etracer65, 2020-06-19:
> Cannot the RPM reported by ESC telemetry be used for RPM filtering instead of using DSHOT ? This could avoid doing the work twice.

No. The ESC SENSOR telemetry is far too slow to be used for RPM filtering.

> So I've tried to lower the PWM Frequency of the ESC from 48kHz to 24kHz and it didn't change a thing regarding the error rate.
If the load was the issue, shouldn't that relieve the ESC and allow it to report without errors?

I wouldn't expect that it would since the PWM frequency is a function of the hardware gate driver and shouldn't have any affect on the ESC MCU. However the fact that you reported the error rate increasing at higher motor speeds is a confirmation of the MCU load affecting the error rate since as RPM increases so does the load on the ESC MCU.

Diaoul, 2020-06-19:
Thanks for the help and clarifications. 
Is that a limitation of all BLHeli32 ESCs or this particular model? Should I go back to DSHOT300 and 4k/4k or stick with DSHOT600 8k/8k despite the error rate in your opinion? Or maybe just drop ESC_SENSOR? I don't know if that's useful anyways.

I hope this can be optimized on BLHeli32 🤞

etracer65, 2020-06-19:
I haven't head of any generalized problems with running bidirectional DSHOT and ESC sensor telemetry. But that's really more of a question for the BLHeli32 devs. I would recommend you raise an issue with them. https://github.com/bitdump/BLHeli/issues

I wouldn't recommend using it in a mode where the errors are increasing with RPM as you really won't know what's going on in flight. So if you want to keep using this combo i would either turn off ESC Sensor or run DSHOT300.

Diaoul, 2020-06-20:
I did report it on the BLHeli repo, see the linked issue above. I will disable ESC Sensor as I do not use the temperature reported from it anyways.

Diaoul, 2020-07-19:
@sskaug investigated this (thank you!) and seems to have found the culprit. Increasing the tolerance for delay should solve the issue.

haslinghuis, 2020-07-19:
Test code addressing this issue can be found here: https://github.com/bitdump/BLHeli/tree/master/BLHeli_32%20ARM/Rev32.7.2%20SBUS%20and%20S.PORT%20testcode

Beware this is for testing purposes only and not an official release.

sskaug, 2020-07-19:
Note that the Re32.7.2 addresses an issue of too short delay between dshot input and response. While this issue is about too long delay

closed 2020-08-29 by stale[bot]