betaflight/betaflight #9931: FuryF4OSD - OSD Not Working on 4.2

blauver, 2020-06-20:
OSD is not functioning in 4.2. The OSD tab in the configurator does not have the normal options such as selecting NTSC or PAL nor is the Font Manager button selectable. Placing OSD items on the display does not have any effect on the output. Of note, the OSD works fine when running 3.5.7 and 4.1.7; The 4.2.X versions are the only ones with issues.

# If your issue looks like a hardware fault or a configuration problem please don't raise an issue here.

## Please consider using other user support options such as asking the manufacturer of the hardware you are using, RCGroups: https://rcgroups.com/forums/showthread.php?t=2464844, or Slack (registration at https://slack.betaflight.com/) or other user support forums & groups (e.g. Facebook).

## If you believe there is an issue with the firmware itself please follow these steps:
1. Describe your problem;
2. Include ways to reproduce the problem;
3. Provide as much information as possible about your hardware and software, including:
- what hardware / software you are using;
- the version of all the software used;
- how things are connected / wired up.
4. Create a `diff` and post it here in a code block. Put ``` (three backticks) at the start and end of the `diff` block (instructions  on how to do a diff: https://oscarliang.com/use-diff-not-dump-betaflight/);
5. Remove this Text :).

blauver, 2020-06-20:
Update: I downgraded to 4.1.7 and flashed the font from the Font Manager. This option was not available when running 4.2.0. Going back to 4.2.0, the Font Manager option is now present and permits the font to again be flash. All is working now.

etracer65, 2020-06-20:
You just need to have the battery connected BEFORE connecting the USB. That flight controller doesn't power the OSD chip on USB.

closed 2020-06-20 by etracer65

blauver, 2020-06-21:
Thank you @etracer65 !! Makes perfect sense now!

calebunleashed, 2020-07-12:
I had the same issue with a FuryF4OSD firmware on one drone, and the OMNIBUSF4 firmware on another drone.

It seems this 4.2.0 firmware is buggy. Downgrading to 4.1.7 fixes the issue on both flight controllers. Plugging in the battery does not make a difference in this case.

FC's in use:
- MAMBA F405 MK2 (Fury)
- HGLRC XJB F413-TX20.V2 (Omni)

@etracer65 can you please reopen this issue?

etracer65, 2020-07-12:
@calebunleashed There is no problem. Just connect the battery first, wait a second or two, then plug in the USB.

calebunleashed, 2020-07-12:
@etracer65 Why does the previous firmware version (4.1.7) work without the battery connected? Was there a change done in 4.2.0 to make this mandatory?

calebunleashed, 2020-07-12:
For example I have downgraded to 3.1.7, I do not have the battery connected, and I can see the expected OSD settings:

[image]

etracer65, 2020-07-12:
@calebunleashed Because 4.2 added actual detection for a responding OSD chip - something that wasn't present in earlier versions. So older versions wouldn't recognize that the OSD chip wasn't responding and would allow features to seem to work when they really weren't. Some flight controller (like the ones you've listed) don't power the OSD chip when only connected via USB. There is no problem and the features are working as designed.

calebunleashed, 2020-07-13:
Hi @etracer65 - thanks for the explanation, I appreciate your time and I understand this now.

Could it be worth adding in one of the following suggestions:

  1. "greying out" the other OSD options instead of removing them

  2. Having a warning message displayed on the OSD screen to advise the user something along the lines of "No OSD chip detected, please plug in your battery to power on the OSD chip for configuring". Currently the users sees this for 4.2.0 firmware on Betaflight Configurator 10.7.0:
[image]

&nbsp;&nbsp;&nbsp;&nbsp;3. Or, if the user turns on OSD in the Configuration screen, then display all the OSD settings:
[image]

The number 3 suggestion seems to be more inline with the user experience of the Betaflight, for example:

**Step 1: Set the configuration to be a quad:**
[image]

**Step 2: Play with the motors settings like an ESC exists:**
[image]

**Step 3: Observe the bare FC is plugged in to the computer with no ESCs or battery:**
[image]

Thanks for your consideration.

mikeller, 2020-07-14:
@calebunleashed:

> Having a warning message displayed on the OSD screen to advise the user something along the lines of "No OSD chip detected, please plug in your battery to power on the OSD chip for configuring". Currently the users sees this for 4.2.0 firmware on Betaflight Configurator 10.7.0:

Is something like this what you had in mind:

[image]

This is what the official Betaflight configurator, with the official firmware, with default settings and no battery connected gives you for your target. If you had posted the information we asked you to post when opening this issue then maybe somebody could have helped you determine what is wrong with your configuration that stops this from showing...