betaflight/betaflight #9932: Why there is a softserial limit?

dgrat, 2020-06-20:
Are more softserial ports possible on F4+ Hardware?

tenten8401, 2020-06-20:
Can't you just assign more ports with the resource command? How many do you have?

haslinghuis, 2020-06-21:
Software serial has a heavy load on the MCU hence only 2 are currently allowed (supported) in the code.

mikeller, 2020-06-21:
You _could_ probably increase the number to more than 2 on the MCU types that are currently supported - this seems to be barely needed, and most boards probably don't have enough pins with available timers to use more than 2 SOFTSERIALs, but there is nothing to say that this won't work.

dgrat, 2020-06-21:
@mikeller You are probably right. After struggling to use RunCam Prot, ESC Duplex, Tramp/Smartport on a F4, I was wondering why there is this limit, because the MCU Load is dependent on many factors and two seems arbitrary. On the other hand I just noted that two might be sufficient for my needs and with the F7 boards softserial might become less important.

mikeller, 2020-06-21:
@dgrat: The higher flexibility of hardware UARTS on F7 boards should allow you to work around the need for more SOFTSERIALs indeed.

closed 2020-06-22 by dgrat