betaflight/betaflight #9933: BetaFlight 3.5.7 missing from releases page

evilC, 2020-06-20:
Also seems to not be showing up in the BF configurator
The releases page jumps from 3.5.6 to 4.0.0 RC1, although the 3.5.7 tag does still exist, and the release is downloadable if you [craft the appropriate URL](https://github.com/betaflight/betaflight/releases/tag/3.5.7)

bizmar, 2020-06-20:
Can't check configurator now, but it's there in releases, it's a maintenance update that was published between 4.0 RCs, back porting some of the bugs fixed during 4.0 development... It's just not directly after 3.5.6 ... https://github.com/betaflight/betaflight/releases?after=4.0.0-RC3

etracer65, 2020-06-20:
It's not in the Configurator because it's more than 2 major releases old. If you want it then download from GitHub.

closed 2020-06-20 by etracer65

mikeller, 2020-06-21:
@evilC: I think GitHub was having some issues in the last few days, looks like this was a glitch: https://github.com/betaflight/betaflight/releases?after=4.0.0-RC3