betaflight/betaflight #9934: New Failsafe mode: Right and drop

brunofporto, 2020-06-20:
Just drop the drone can have consequences. If the drone is fast or in the middle of a stunt it can drop at a nonfavorable angle.

It would be nice if the drone level itself and then drop.

We could even choose some angle as a preference instead of flat as perfectly flat is not always good, Some angle is better to absorb the hit energy.

If the drone has GPS and can read speed than it would be nice for it to brake first, level, and then drop.

But basically: Level and drop would be a nice start.

etracer65, 2020-06-20:
You can already do this. Just set the stage 1 channel fallback value (also available in the CLI as `rxfail`) for the aux channel you have angle mode assigned so that it activates on stage 1 failsafe. Then set stage 2 to DROP. When you failsafe the quad will enter stage 1 and self-level. Then 0.4 seconds later it will drop.

brunofporto, 2020-06-20:
Nice! Thanks :)

I'll close this issue, sorry. :)

closed 2020-06-20 by brunofporto