betaflight/betaflight #9939: Gps/Navigation Features Please

JCFPV, 2020-06-22:
I really like betaflight and would love to see more gps features such as position hold or a similar program of mission control to be able to have more features since the new flight controllers are coming out with more space. A lot of times when I am flying someone walks by or a car comes down the road and it would be nice to be able to flip a switch and it stay in place or possibly fly a set course  Also I really like betaflight and the features with it and ease of use it would be great to use cinewoops on with more gps/ Navigation Features Thanks Betaflight for your time thanks for everything.

# If your issue looks like a hardware fault or a configuration problem please don't raise an issue here.

## Please consider using other user support options such as asking the manufacturer of the hardware you are using, RCGroups: https://rcgroups.com/forums/showthread.php?t=2464844, or Slack (registration at https://slack.betaflight.com/) or other user support forums & groups (e.g. Facebook).

## If you believe there is an issue with the firmware itself please follow these steps:
1. Describe your problem;
2. Include ways to reproduce the problem;
3. Provide as much information as possible about your hardware and software, including:
- what hardware / software you are using;
- the version of all the software used;
- how things are connected / wired up.
4. Create a `diff` and post it here in a code block. Put ``` (three backticks) at the start and end of the `diff` block (instructions  on how to do a diff: https://oscarliang.com/use-diff-not-dump-betaflight/);
5. Remove this Text :).

augustozanellato, 2020-06-22:
Do you mean something like GPS+baro position/altiutude hold? In that case I don't think it would be feasible to hold the position just using GPS because it doesn't have enough precision, an optical flow sensor would be better suited to that job

closed 2020-06-23 by no-response[bot]

richardbulger, 2020-11-17:
I think position hold in particular would be a very useful feature to have, especially now GPS rescue mode has been implemented and more widely used.
Is wouldn't need to super precise. Like the GPS rescue mode, just good enough to get you out of tight spots where you can't immediately land (bug in your goggles, anyone?)
It could augment GPS rescue mode feature, by being able to hover (roughly) over the home point