betaflight/betaflight #9940: OSD: Crosshair is overwrited by artificial horizon

McGiverGim, 2020-06-22:
**Describe the bug**
The Crosshair symbol is being overwrited by the Artificial Horizon. This has changed since 4.1. Now the Crosshair is part of the "background" of the OSD and is being drawn earlier than the Artificial Horizon.
The Crosshair is "useful" under two circumstances specially: 
1. To help maintain a direction: this continue working because the Artificial Horizon is way off when you go straight forward.,
2. To help landing. Aligning the Crosshair with the Artificial Horizon helps to land.

The second part now is difficult because the Crosshair is not visible. We all know that is there, but is not natural and changes the way it is since the beginning..

**To Reproduce**
Use the Crosshair and the Artificial Horizon in the OSD.

**Expected behavior**
The Crosshair must be visible over the Artificial Horizon. 

**Flight controller configuration**
Here it is, but I doubt is useful
```
# 

# diff all defaults

# version
# Betaflight / STM32F7X2 (S7X2) 4.2.0 Apr 29 2020 / 08:09:54 (b480103d4) MSP API: 1.43
# config: manufacturer_id: MTKS, board_name: MATEKF722SE, version: f173f9bc, date: 2019-10-02T14:43:13Z

# start the command batch
batch start

# reset configuration to default settings
defaults nosave

board_name MATEKF722SE
manufacturer_id MTKS
mcu_id 0048003b3237510c31313935
signature 

# name: MIV-MIGMUL
#feature -GPS

# feature
feature -AIRMODE
feature GPS
#feature AIRMODE
#beeper ON_USB

# beeper
beeper -ON_USB
#beacon -RX_SET

# beacon
beacon RX_SET

# serial
#serial 0 0 115200 57600 0 115200
serial 0 2 115200 57600 0 115200
#serial 2 0 115200 57600 0 115200
serial 2 2048 115200 57600 0 115200
#serial 3 0 115200 57600 0 115200
serial 3 16384 115200 57600 0 115200

# aux
#aux 0 0 0 900 900 0 0
aux 0 0 0 1400 1850 0 0
#aux 1 0 0 900 900 0 0
aux 1 1 1 1275 1425 0 0
#aux 2 0 0 900 900 0 0
aux 2 1 0 1900 2100 0 0
#aux 3 0 0 900 900 0 0
aux 3 2 1 1425 1600 0 0
#aux 4 0 0 900 900 0 0
[... 365 lines cut ...]
set yaw_srate = 51

rateprofile 2

rateprofile 3

rateprofile 4

rateprofile 5

# restore original rateprofile selection
rateprofile 0

# save configuration
save
# 
```

**Setup / Versions**
 - Flight controller [what type is it, where was it bought from]; Matek F722SE
 - Other components [RX, VTX, brand / model for all of them, firmware version where applicable];
 - how are the different components wired up.

**Additional context**
I know that this was done to "optimize" the speed in the OSD. But I think the elements first must be useful, and after that, try to optimize them.

dkustec, 2020-06-23:
@etracer65 , crosshairs shouldn't be in the background when the acc is used. It's pretty annoying when the crosshairs disappears from the horizon line when landing.
This can solve the problem easily:

```
const osdElementDrawFn osdElementDrawFunction[OSD_ITEM_COUNT] = {
    [OSD_CAMERA_FRAME]            = NULL,  // only has background. Added first so it's the lowest "layer" and doesn't cover other elements
    [OSD_RSSI_VALUE]              = osdElementRssi,
    [OSD_MAIN_BATT_VOLTAGE]       = osdElementMainBatteryVoltage,
#ifdef USE_ACC
    [OSD_ARTIFICIAL_HORIZON]      = osdElementArtificialHorizon,
    [OSD_CROSSHAIRS]              = osdBackgroundCrosshairs, // draw croshairs in the foreground if the acc is used
#else
    [OSD_CROSSHAIRS]              = NULL,  // only has background
#endif
```

McGiverGim, 2020-06-26:
Pushed a PR to "change" this behaviour. Tested and it works.

closed 2020-07-05 by mikeller