betaflight/betaflight #9941: Failsafe for bad gyro

dgrat, 2020-06-22:
During the winter period I was not flying. Later, I noticed that one of my copters had a bad gyro. Ocassionally on start while stationary, the z-axis was drifting with 15 deg/s and later also x and y axis (5-30 deg/s). The copter could be armed without a problem but flying became hard. 
Does it make sense to add a failsafe, hindering the user from arming when the gyro has a constant drift before arming? I would suggest to couple the failsafe with the max arm angle value and disable when the value is set to 180 degree. Naively, one could maybe make use of a very low frequenzy LPF (< 0.01 - 0.1 Hz) to "average" the drift rate per axis before arm.

etracer65, 2020-06-22:
How exactly would the flight controller determine the "static drift"? The only sensor we have is the gyro. How can we use the gyro sensor to determine the gyro sensor is bad? Also most of the time when sensors fail this way it's related to heat so initially they don't show much problem but rapidly degrade after being powered on for a while. So at that point you're likely already flying.

dgrat, 2020-06-22:
All other quads have a neglectable random drift around 0.0x. If you have a constant drift of >1 or <1 it is bad.

etracer65, 2020-06-22:
You're missing the point. The gyro is the only sensor we have to measure angular rotation. We can't differentiate between "real" and "failing sensor" because we have no other frame of reference.

dgrat, 2020-06-22:
Shouldn't a stationary gyro return always on average close to zero values?

etracer65, 2020-06-22:
Again, how does the flight controller know it's stationary? The gyro says it's not. What reason do we have to not trust the gyro?

Also this type of gyro failure is exceedingly rare. Building in complicated detection logic for this is definitely not needed.

dgrat, 2020-06-22:
Well theoretically, the accelerometer could be used to check whether the copter is moved.

etracer65, 2020-06-22:
What if the user has the accelerometer disabled? Anyway rotation is not acceleration.  For example a flight controller rotating at a constant speed around the Z axis (yaw) will show no accelerometer activity.

dgrat, 2020-06-22:
> What if the user has the accelerometer disabled?  
I think, while disarmed, it could make sense to let the accel run (maybe sample slow)
>  For example a flight controller rotating at a constant speed around the Z axis (yaw) will show no accelerometer activity.
correct me if i am wrong, but i thought there would be a centrifugal acceleration like with gravity.

etracer65, 2020-06-22:
> Also I think a failing gyro is not very rare. 

And note that I said **this type** of gyro failure is exceedingly rare. The case where the gyro shows an increasing bias on one of more axis. In fact I've only ever seen 2 confirmed cases.  Other types of failures are more common such as complete failure (no response) or random high-rate glitches on a single axis.

dgrat, 2020-06-22:
Would commits for rare issues be accepted, or are firmware size constraints mor important?

etracer65, 2020-06-22:
For all the reasons I've already mentioned (and a few others - like being able to rearm after an in-flight disarm, arming after "tossing" the quad, etc.) I don't think there's a viable and reliable way to do this. Compounded by the exceedingly rare case of this type of gyro failure it really doesn't seem worthwhile. For example how would anybody test it? It's not like we can get a gyro that has failed in this specific way. But if you want to work on it feel free. But yes adding a solution that uses a significant amount of flash space wouldn't be warranted for the exceptionally rare and specific case.

dgrat, 2020-06-22:
Sounds reasonable. I was expecting issues like this would be more frequent, probably, because it was hitting me

ctzsnooze, 2020-06-30:
Its very rare indeed.

the most common gyro failures are random noise spikes on one axis when shaken at certain external frequencies.

gyro drift is very unusual.

I agree with eTracer; I suggest we close the issue.

closed 2020-07-01 by dgrat