betaflight/betaflight #9942: Diatone MAMBAF722

davidbabreu, 2020-06-22:
**Describe the bug**
In OSD tab, the font manager is not showing. Was working on earlier versions (4.1.7)

**To Reproduce**
Just flash the 4.2 RC or final and button turns grey.

**Expected behavior**
Font manager should be available?

**Flight controller configuration**
Settings does not affect. Button is never available.

Use `resource show all` to create a resource allocation list and post it here in a code block. Put ``` (three backticks) at the start and end of the output block.

**Setup / Versions**
 - Diatone MambaF722 (mini)
 - Crossfire / Rush Tank ultimate
 - everything is soldered up

**Additional context**
Add any other context about the problem here.

Asizon, 2020-06-22:
And, can you read the Osd warning message at the top?

davidbabreu, 2020-06-22:
Yes, I have a warning saying that OSD font may not match the one on the FC.
Also missing it's the OSD option for the video format (Auto, PAL, NTSC).

à(s) 17:12:

davidbabreu, 2020-06-24:
Think this is automatic, don't know what information was missing.
Just wanted to report the bug of the OSD tab being incomplete. No place to select NTSC or PAL, and font manager isn't available.
Again, I'm using the MAMBAF722 target as told on the official Diatone website, but I have the mini stack.
If helpfull, I can stream the problem or try to record it if easier.

haslinghuis, 2020-06-24:
BFC 10.7.0 / BF 4.2
Please try attaching LiPo before connecting USB. But even then I get the OSD warning:
```
Note: OSD preview may not show the actual font that is installed on the flight controller. The layout of individual elements may look different when older versions of the firmware are used - please check the look through your goggles before flying.
```
Font Manager Button also not available.

Asizon, 2020-06-24:
Yes, this osd note and not warning is only for osd font preview. The font manager button and the new warning messages appears both when osd chip is not detected, like warning says:
`<b>WARNING:</b></span> No OSD chip was detected. Some flight controllers will not properly power the OSD chip unless connected to battery power. Please connect the battery before connecting the USB (PROPS REMOVED!)."`

Asizon, 2020-06-24:
@davidbabreu please follow our instructions ore Warning message instructions and you are done

Asizon, 2020-06-24:
Weird!! In some update we have break the Warning message, some strange is happens

Asizon, 2020-06-24:
I think I have fixed this behiavour:
https://github.com/betaflight/betaflight-configurator/pull/2084

haslinghuis, 2020-06-24:
Sorry to confuse this issue. It seems I forgot to reset `osd_displayport_device = AUTO`
Now it is working!

davidbabreu, 2020-06-24:
Only now managed to try things out.
haslinghuis post was the one targeting the issue and provided the solution. Don't know why it didn't work before, because I had power before connecting usb (to check motor direction and to install OSD font) and the button was still grey and OSD tab seemed incomplete. But today, pluged everything on before connecting USB and starting Betaflight and indeed the OSD tab was complete, and font manager button working. Today I did it on a different computer, but don't believe that is the issue here. I might have plug the battery while USB and Betaflight were already opened.

Still that error was not showing, the only one I had was this:
Nota: A fonte da pré-visualização do OSD pode não ser a mesma da fonte instalada no controlador de voo. O layout de alguns elementos podem ser diferentes em versões mais antigas do Betaflight. Sempre verifique o visual em seus óculos antes de voar.
(Note: OSD preview may not show the actual font that is installed on the flight controller. The layout of individual elements may look different when older versions of the firmware are used - please check the look through your goggles before flying.)

Can't thank enought to you all for taking the time to help.
Kind regards and I wish you the best flights
David

Asizon, 2020-06-24:
Ok, thankyou @haslinghuis and @davidbabreu for your great tests, im making changes on my PR trying to avoid these issues.

closed 2020-08-01 by stale[bot]