betaflight/betaflight #9943: Max Throttle Issue

tedwanday, 2020-06-23:
Hi everyone,

I just flashed BF4.2 on my Mamba F405 and didnt copy any of my old settings to this firmware, I set everything up again. However, when I arm my board the motors would spin even at throttle zero and I realised this is due to AirMode. But when I raise my throttle past a certain value, it goes to max and sticks there permanently and if I reduce the throttle to zero it doesnt cut out, i have to disarm the board to turn it off. Could someone please help me identify the cause of this?

McGiverGim, 2020-06-23:
Are you on bench or it is flying? If it is on the bench it is habitual that airmode produces this. It will work ok flying.
Have you tested the motors tab (propellers off)?

tedwanday, 2020-06-23:
Yes im on the bench but ive tried it with and without the propellors off.
I read a similar post where the issue was that it was being tied down, but /i tried all the fixes listed there and nothing changed

McGiverGim, 2020-06-23:
It's easy to check: disable airmode and check again to see if the problem persists.

tedwanday, 2020-06-23:
It does persist. When I pass a certain throttle value it throttles up to maximum and stays there

haslinghuis, 2020-06-23:
When testing the quad on the bench: without props or with props and tied down won't matter.
When armed the FC receives throttle input and wants to move the quad. If it doesn't move, motor output goes up. On some point you can not go lower without disarm. With airmode or angle/horizon mode enabled this behavior will be exaggerated.

closed 2020-06-24 by no-response[bot]