betaflight/betaflight #9945: Replacing broken Betaflight OSD with FrskyOSD.

ben975, 2020-06-23:
Hi,

I broke the OSD on my F7 FC running betaflight, but everything else works. The OSD still appears configurable in betaflight etc so all is good there. However I would like to run a Matek Analog Pixel OSD instead just to get it all working again. As the FrskyOSD is a new feature, I was wondering if there is a way for me to switch off the inbuilt OSD and instead use the external one in Betaflight Configurator. I also noticed it said graphical features aren't currently supported, and am wondering if this allows me to view just the voltage for now in the feed?

If this doesn't exist is it possible that the feature could be added in a future update as I have 2 boards with ripped pads around the osd (crashes) and it would be very good to be able to fix these boards and add an external osd.

Thanks.

haslinghuis, 2020-06-23:
Just connect the Pixel OSD onto a free UART (Rx/Tx), connect the camera and VTX to the OSD instead of using the FC.
On the `ports tab` under peripherals select OSD (FrSky Protocol) on the corresponding UART.

You still can use SmartAudio on another UART (Tx) but you'll have to connect the FC to the OSD and the OSD to the VTX. Connecting SmartAudio Tx directly to the VTX would result in electrical noise.

I am unsure if its needed to disable the onboard OSD with `osd_displayport_device=NONE` because configuring the OSD at the ports tab should disable the use of the onboard OSD.

haslinghuis, 2020-06-23:
Important: This is not the right place to get help with configuration issues. Your issue will be closed without further comment.

Please get help from other user support options such as asking the manufacturer of the hardware you are using, RCGroups: https://rcgroups.com/forums/showthread.php?t=2464844, or Slack (registration at https://slack.betaflight.com/) or other user support forums & groups (e.g. Facebook).

ben975, 2020-06-27:
> Just connect the Pixel OSD onto a free UART (Rx/Tx), connect the camera and VTX to the OSD instead of using the FC.
> For details visit the manufacturer website.
> 
> In BFC on the `ports tab` under peripherals select OSD (FrSky Protocol) on the corresponding UART.
> 
> You still can use SmartAudio on another UART (Tx) but you'll have to connect the FC to the OSD and the OSD to the VTX. Connecting SmartAudio Tx directly to the VTX would result in electrical noise.
> 
> I am unsure if its needed to disable the onboard OSD with `osd_displayport_device=NONE` because configuring the OSD at the `ports tab` should disable the use of the onboard OSD.
> 
> First connect LiPo, than connect USB to upload font into PXOSD module. `save`

Thanks for this!

I have an issue where my video flickers with the new osd in flight and the osd moves around the screen. I can send dvr if you would like?

haslinghuis, 2020-06-27:
Please try NTSC on both CAM and in OSD settings. This could also indicate a ground problem. This is not firmware issue.

Which CAM are you using? Could you show a picture of the wiring?

Asizon, 2020-06-28:
@ben975 Try the new V2 firmware version with the new gui, i think it solved your issues
https://github.com/FrSkyRC/FrSkyOSDApp/releases/tag/v2.0.1

ben975, 2020-06-29:
> @ben975 Try the new V2 firmware version with the new gui, i think it solved your issues(maybe it need to adapt also in Betaflight firmware side)
> https://github.com/FrSkyRC/FrSkyOSDApp/releases/tag/v2.0.1

Thanks both of you!

I actually totalled this FC last weekend :( and so the need for the OSD, but I think the issue was the common ground issue as it only became really bad on 5s

I am planning to build a wing soon and I feel that the issue is both the v1 firmware and the common ground as I had symptoms of both the second there was high current or noise the osd would move around and the screen flicker, but it was always faint anyway (seems like firmware). Either way I am putting this board in a fixed wing or 7" haven't decided yet within the next 6-8 weeks and will retest the firmware etc as I want a graphical osd with inav and I have a spare osd now anyway.

Do you guys want the DVR of the issue?

closed 2020-08-08 by stale[bot]