betaflight/betaflight #9950: Matek F405-STD broken barometer? How to check?

MR444, 2020-06-25:
Hi everyone, I have a new Matek F405 -STD which does not show any values for the altitude. It is always zero. 

My older Matek F405-STD works without problems with BF 4.2.0. How can I check if the sensor is broken?

Barometer is activated in configuration tab

Thanks.

[image]
[image]

haslinghuis, 2020-06-25:
Did you try `set debug_mode = altitude` ?

Otherwise if the config is exactly the same I think you should request a RMA.
It could also be a HW revision difference I am not aware of.

MR444, 2020-06-25:
Okay I found out what is the problem. There was a small difference with huge effect. 

Barometer stops working when I activate LED strip in configuration tab.

So only LEDs OR barometer work. Why is it like that? Before I had Betaflight 3_5_2 installed and both worked without any problems.

Betaflight 3_5_2 LEDs _and_ barometer working:
[image]

haslinghuis, 2020-06-26:
Use target “MATEKF405STD (MTKS)
On BF Pins Definition tab:

- 2812LED_Strip share I2C1_SCL pad as default.  either enable LED_Strip, or enable Baro.
- 2812LED can be resourced to S7 (PB8) if using Baro the same time with BF firmware

MR444, 2020-06-26:
Thanks for your help. I also read this on Matek product description but how to resource to S7? I always get a parsing error.

When I type in "resoucre" I get the following options for resourcing the LED strip:
resource LED_STRIP 1 B06

And for status LEDs:
resource LED 1 B09
resource LED 2 A14

Which command should be the correct one?

haslinghuis, 2020-06-26:
Please try:

```
resource motor 7 none
resource LED_STRIP 1 PB8
save
```

haslinghuis, 2020-06-26:
If that doesn't work please provide a `diff` and use `resource show all` to create a resource allocation list and post it here in a code block. Put ``` (three backticks) at the start and end of the output blocks on its own line.

MR444, 2020-06-27:
Nope still doesn't work. That's also what I tried. Resource parsing failed.

```
# 

# resource motor 7 none
Resource is freed

# resource LED_STRIP 1 PB8
###ERROR: resource: PARSING FAILED###

# resource show all
Currently active IO resource assignments:
(reboot to update)
--------------------
A00: FREE
A01: FREE
A02: FREE
A03: SERIAL_RX 2
A04: FREE
A05: SPI_SCK 1
A06: SPI_MISO 1
A07: SPI_MOSI 1
A08: FREE
A09: FREE
A10: FREE
A11: USB
A12: USB
A13: SWD
A14: LED 2
A15: FREE
B00: FREE
B01: FREE
B02: FREE
B03: SPI_SCK 3
B04: SPI_MISO 3
B05: SPI_MOSI 3
B06: LED_STRIP
B07: I2C_SDA 1
B08: FREE
B09: LED 1
B10: OSD_CS
B11: FREE
B12: FREE
B13: SPI_SCK 2
B14: SPI_MISO 2
B15: SPI_MOSI 2
C00: FREE
C01: SDCARD_CS
C02: GYRO_CS 1
C03: GYRO_EXTI
C04: ADC_CURR
C05: ADC_BATT
C06: MOTOR 1
C07: MOTOR 2
C08: MOTOR 3
C09: MOTOR 4
C10: SERIAL_TX 3
C11: FREE
C12: SERIAL_TX 5
C13: BEEPER
C14: FREE
C15: FREE
D00: FREE
D01: FREE
D02: FREE
D03: FREE
D04: FREE
D05: FREE
D06: FREE
D07: FREE
D08: FREE
D09: FREE
D10: FREE
D11: FREE
D12: FREE
D13: FREE
D14: FREE
D15: FREE
E00: FREE
E01: FREE
E02: FREE
E03: FREE
E04: FREE
E05: FREE
E06: FREE
E07: FREE
E08: FREE
E09: FREE
E10: FREE
E11: FREE
E12: FREE
E13: FREE
E14: FREE
E15: FREE
F00: FREE
F01: FREE
F02: FREE
F03: FREE
F04: FREE
F05: FREE
F06: FREE
F07: FREE
F08: FREE
F09: FREE
F10: FREE
F11: FREE
F12: FREE
F13: FREE
F14: FREE
F15: FREE

Currently active Timers:
-----------------------
TIM1: FREE
TIM2: FREE
TIM3:
    CH1: MOTOR 1
TIM4:
    CH1: LED_STRIP
TIM5: FREE
TIM6: FREE
TIM7: FREE
TIM8:
    CH2: MOTOR 2
    CH3: MOTOR 3
    CH4: MOTOR 4
TIM9: FREE
TIM10: FREE
TIM11: FREE
TIM12: FREE
TIM13: FREE
TIM14: FREE

Currently active DMA:
--------------------
DMA1 Stream 0: LED_STRIP
DMA1 Stream 1: FREE
DMA1 Stream 2: TIMUP 3
DMA1 Stream 3: FREE
DMA1 Stream 4: FREE
DMA1 Stream 5: FREE
DMA1 Stream 6: FREE
DMA1 Stream 7: FREE
DMA2 Stream 0: ADC 1
DMA2 Stream 1: TIMUP 8
DMA2 Stream 2: FREE
DMA2 Stream 3: FREE
DMA2 Stream 4: FREE
DMA2 Stream 5: FREE
DMA2 Stream 6: FREE
DMA2 Stream 7: FREE

# 
```

haslinghuis, 2020-06-27:
I got this answer from MATEKSYS:
```
This conflict has been on F405-STD/F405-CTR since they were released in 2017
Just follow the tips in their page.

resource motor 7 none
restore led_strip 1 B08

After remapping,  must solder LED DIN to S7 pad,  rather then 'LED" pad,  otherwise it will still effect SCL bus.
```
On the Matek website it specifies PB8 en in their answer it is B08?

MR444, 2020-06-27:
Yes strange. PB8 must be wrong. I will try again after I soldered LED strip to S7 later today. I really wonder how I made it working with BF 3.5.3 on original LED pin.

MR444, 2020-06-28:
Okay I think what Matek calls PB8 is B08 and PB6 is B06.

So I freezed B08, resourced LED strip to B08, saved, soldered LED strip to S7 connected and... magic smoke.

I thougt it would be a good idea to buy an old FC which I just have to exchange one by one but now causing more problems than checking wiring of a new FC.

haslinghuis, 2020-06-28:
Hope the smoke was on the led strip, Do you use a 2812?
The resourced pin is a data to led din pin, not for power.

Did you connect like in the [picture](http://www.mateksys.com/?portfolio=2812-8slim) --> in your case replace LED DIN with S7.

MR444, 2020-06-28:
Yes me LEDs are connected like on the picture. I use 2812 similiar to these https://www.banggood.com/4-PCS-PandaRC-5V-RGB-WS2812-LED-Strip-w-or-100mm-Cable-for-F3-F4-Flight-Controller-RC-Drone-FPV-Racing-p-1672588.html?rmmds=search&cur_warehouse=CN

Sure S7 is data pin and not for power but I wanted to check if anything "comes out" from data pin. 

Could it be that my multimeter works somehow like a resistance? I cannot explain why it only works when I put a multimeter between data pin and +5V. On LED pin (same that is used for barometer) me LEDs work without problems.

I don't know where the smoke came from. Everything still works. Really strange how these small things can keep me so busy.

haslinghuis, 2020-06-28:
You should also test continuity with your multimeter and maybe use a smokestopper.
I see you enable the `LED_STRIP` feature so I assume you actually configured the LEDs in the LED Strip tab.
What could work I found on a related issue #4988 is to power the LED Strip from the 4V5 pad.
Also try with a default config and configure the led strip before configuring anything else.

MR444, 2020-06-29:
4V5 makes not difference.

Looks like @alexz-kh has the same problem since he updated. 

I think I tried now everything and try to find some non-programmable 5V LEDs.

Thank you for your help!

brainbubblersbest, 2020-06-30:
[image]

Anisotropic, 2020-07-01:
Measure voltage coming to led. If this >=5v this is the answer. You can solder a diod between led strip and powered pin. So coming voltage became 4.7-4.8v and led would work. 2812 needs less then 5v in some situations. For more details you can google this.

MR444, 2020-07-02:
> 
> 
> Measure voltage coming to led. If this >=5v this is the answer. You can solder a diod between led strip and powered pin. So coming voltage became 4.7-4.8v and led would work. 2812 needs less then 5v in some situations. For more details you can google this.

What a simple solution for such an annoying problem! Soldering one Zener diode in 5 minutes and the problem is solved! Thank you very much, you are the best.  :))

The voltage on S7 is really slightly higher than on LED pin. Obviously slightly enough for creating this problem. It still doesn't explain why LED and Baro worked before with older betaflight versions withoud diode but the problem is solved..

closed 2020-07-02 by MR444

MATEKSYS, 2020-07-02:
Nothing wrong with voltage on S7,  It is PWM signal.  multimeter can't read right voltage.

I am sure it is LEDs tolerant issue,   Quality difference between different LED factory.

In order to prove what I said. I tested it just now with our LED strip
LED is powered by 5.17V,  it works fine.
[image]

200ohm on S7 rail. I didn't change anything.
[image]

If you hook 2812LED to "LED" pad of F405-STD/F405-CTR,  no matter what betaflight version you use.  LED and Baro 100% can't work together.

brainbubblersbest, 2020-07-02:
A normal Diode should work. Why a Zehner Diode?
A Standard Diode reduce the voltage at 0, 7v
5,1-0,7 and you have 4,4V
Or you get matek led stripes. These work with 5V also

Anisotropic, 2020-07-02:
>Or you get matek led stripes. These work with 5V also
It depends. I have old matek f405 plate and matek leds (were bought separately). If you solder just leds (like MATEKSYS did) they work. If you solder all quad stuff like receiver, video transmitter, camera - the leds don't work and need a diod. :neutral_face: