betaflight/betaflight #9952: Cannot regain control after failsafe induced GPS Rescue

Daemon42, 2020-06-26:
**Describe the bug**
After triggering GPS Rescue mode by inducing a Stage 2 failsafe with RC link loss, when the RC link was restored, it refused to exit failsafe, thus refused to exist GPS rescue.  It flew home, attempted to "land" and disarmed 10ft above the pavement.  

**To Reproduce**
Following normal testing with GPS Rescue mode which worked as expected, I performed a final test of failsafe induced GPS Rescue, by turning off my transmitter with flying quad, and then back on again a few seconds later (to simulate true loss and restoration of signal).
As soon as signal was lost, it went through Stage 1 and 2 failsafe, and activated GPS Rescue, climbed and began flying home.    After I turned Tx back on, I verified that RC link was re-established (showed active telemetry on the Tx) but Betaflight would not exit failsafe.   I have a black box log of the flight that indicates that the FC saw the RC signal restored, followed by various mode changes, stick movements, etc, but it refused to exit failsafe.  

**Expected behavior**
Per the GPS Rescue wiki, I expected to regain control immediately after RC link was restored and FC detected that it was receiving a valid signal from the Rx. 

**Flight controller configuration**
```
# 

# diff all

# version
# Betaflight / STM32F745 (S745) 4.2.0 May 13 2020 / 14:27:23 (6aeac3caa) MSP API: 1.43
# config: manufacturer_id: HBRO, board_name: KAKUTEF7HDV, version: cc8e118f, date: 2020-03-04T02:37:20Z

# start the command batch
batch start

# reset configuration to default settings
defaults nosave

board_name KAKUTEF7HDV
manufacturer_id HBRO
mcu_id 0041002d3438511033343635
signature 

# feature
feature -AIRMODE
feature GPS

# serial
serial 1 64 115200 57600 0 115200
serial 2 2 115200 57600 0 115200
serial 5 0 115200 57600 0 115200
serial 6 1024 115200 57600 0 115200

# aux
aux 0 0 0 1275 2100 0 0
aux 1 1 1 1300 1700 0 0
aux 2 2 1 1700 2100 0 0
aux 3 46 2 1825 2100 0 0
aux 4 13 3 1775 2100 0 0
aux 5 28 0 1750 2100 0 0

# master
set dyn_notch_width_percent = 1
set dyn_notch_min_hz = 120
set dyn_notch_max_hz = 375
set acc_calibration = 79,27,-219,1
set baro_hardware = AUTO
set rssi_scale = 125
set fpv_mix_degrees = 25
set serialrx_provider = CRSF
set dshot_idle_value = 600
set dshot_bidir = ON
set failsafe_procedure = GPS-RESCUE
set vbat_warning_cell_voltage = 360
set vbat_scale = 112
set yaw_motors_reversed = ON
set small_angle = 180
set gps_provider = UBLOX
set gps_rescue_descent_dist = 50
set gps_rescue_min_sats = 6
set gps_rescue_allow_arming_without_fix = ON
set osd_units = IMPERIAL
set osd_vbat_pos = 14724
set osd_rssi_pos = 14788
set osd_tim_2_pos = 14740
set osd_flymode_pos = 14731
set osd_current_pos = 14755
set osd_gps_lon_pos = 14385
set osd_gps_lat_pos = 14353
set osd_gps_sats_pos = 14792
set osd_home_dir_pos = 2407
set osd_altitude_pos = 2476
set osd_warnings_pos = 14697
set debug_mode = GYRO_RAW
set vcd_video_system = PAL
set gyro_1_align_yaw = 2700

profile 0

profile 1

profile 2

# restore original profile selection
profile 0

rateprofile 0

rateprofile 1

rateprofile 2

rateprofile 3

rateprofile 4

rateprofile 5

# restore original rateprofile selection
rateprofile 0

# save configuration
save
# 
```
```
Currently active IO resource assignments:
(reboot to update)
--------------------
A00: FREE
A01: FREE
A02: LED 1
A03: FREE
A04: SDCARD_CS
A05: SPI_SCK 1
A06: SPI_MISO 1
A07: SPI_MOSI 1
A08: FREE
A09: SERIAL_TX 1
A10: SERIAL_RX 1
A11: USB
A12: USB
A13: SWD
A14: SWD
A15: FREE
B00: MOTOR 1
B01: MOTOR 2
B02: FREE
B03: FREE
B04: FREE
B05: FREE
B06: I2C_SCL 1
B07: I2C_SDA 1
B08: FREE
B09: FREE
B10: SERIAL_TX 3
B11: SERIAL_RX 3
B12: PREINIT 3
B13: SPI_SCK 2
B14: SPI_MISO 2
B15: SPI_MOSI 2
C00: FREE
C01: FREE
C02: ADC_CURR
C03: ADC_BATT
C04: FREE
C05: FREE
C06: FREE
C07: FREE
C08: FREE
C09: FREE
C10: FREE
C11: FREE
C12: FREE
C13: FREE
C14: FREE
C15: FREE
D00: FREE
D01: FREE
D02: FREE
D03: FREE
D04: FREE
D05: SERIAL_TX 2
D06: SERIAL_RX 2
D07: FREE
D08: SDCARD_DETECT
D09: FREE
D10: FREE
D11: FREE
D12: FREE
D13: FREE
D14: FREE
D15: BEEPER
E00: FREE
E01: GYRO_EXTI
E02: SPI_SCK 4
E03: FREE
E04: GYRO_CS 1
E05: SPI_MISO 4
E06: SPI_MOSI 4
E07: FREE
E08: FREE
E09: MOTOR 3
E10: FREE
E11: MOTOR 4
E12: FREE
E13: FREE
E14: FREE
E15: FREE
F00: FREE
F01: FREE
F02: FREE
F03: FREE
F04: FREE
F05: FREE
F06: FREE
F07: FREE
F08: FREE
F09: FREE
F10: FREE
F11: FREE
F12: FREE
F13: FREE
F14: FREE
F15: FREE

Currently active Timers:
-----------------------
TIM1: FREE
TIM2: FREE
TIM3: FREE
TIM4: FREE
TIM5: FREE
TIM6: FREE
TIM7: FREE
TIM8:
    CH1: DSHOT_BITBANG 2
    CH2: DSHOT_BITBANG 5
TIM9: FREE
TIM10: FREE
TIM11: FREE
TIM12: FREE
TIM13: FREE
TIM14: FREE

Currently active DMA:
--------------------
DMA1 Stream 0: FREE
DMA1 Stream 1: FREE
DMA1 Stream 2: FREE
DMA1 Stream 3: FREE
DMA1 Stream 4: FREE
DMA1 Stream 5: FREE
DMA1 Stream 6: FREE
DMA1 Stream 7: FREE
DMA2 Stream 0: FREE
DMA2 Stream 1: FREE
DMA2 Stream 2: DSHOT_BITBANG 2
DMA2 Stream 3: DSHOT_BITBANG 5
DMA2 Stream 4: ADC
DMA2 Stream 5: FREE
DMA2 Stream 6: FREE
DMA2 Stream 7: FREE
```

**Setup / Versions**
 - Flight controller [what type is it, where was it bought from]:   Kakute F7 HDV from RDQ running Betalflight 4.2
 - Other components [RX, VTX, brand / model for all of them, firmware version where applicable]:   
VTX/Cam: DJI FPV AU plug and play to FC
 Beitian BN-170 GPS wired to UART3 
Crossfire Nano Rx wired to UART2 and talking CRSF protocol (full telemetry). 

 - 

**Additional context**
Black box log of the flight available here: https://drive.google.com/file/d/1rl_7x4EsOY0scQlLXOkxuaXrub3bUS4J/view?usp=sharing
Suggest turning on
RC Command [throttle]
Failsafe Phase
Rx Signal Received
Mode changes before 5:08.373 are manual tests of GPS Rescue which were all successful. 
5:08.373 RC Signal lost after turning Tx off
5:09.247 Failsafe activated
5:16..471   RC Signal restored after RC Tx was turned back on. 
Quickly tried giving it various stick inputs, and cut and restored throttle, to no effect.   Failsafe still active. 
05:22.349 Tried disarming and re-arming several times (knowing it might drop).   These were ignored. 
06:07.701 gave up trying to send commands to the quad, and simply watched it attempt to land. 
07:26.180 It disarmed itself when it judged it was close enough to the home altitude and position. 

Others have said this works in BF4.2 but none that I know of have tested by turning Tx off and back on again.  

On a side note, if GPS rescue is enabled, and I do a bench test of stage 2 failsafe (also by turning Tx off after arming) it disarms and I cannot re-arm after I reconnect, drop throttle, disarm and re-arm.   I have to power cycle the quad to re-gain control.
I have to think this is the same issue. 
I verified that if Stage 2 failsafe is set to drop, then this test works fine.   Arm - tx off - failsafe - disarms itself - tx on - wait 2 seconds - drop throttle - re-arm and the motors are spinning normally.

McGiverGim, 2020-06-26:
Your attachment is not public, need to ask for permission. You can attach it directly in github with drag&drop.

marekwiecek, 2020-06-26:
@Daemon42 I had the same behavior when instead of turning off the transmitter, I used FAILSAFE on the switch. What helped was enabling this option.

set gps_rescue_sanity_checks = RESCUE_SANITY_FS_ONLY

Also, on failsafe in betaflight, I am setting AUX to GPS-RESCUE MODE and once GPS rescue is engaged, I set that switch to the same position and once the quad is close by, I switch it off on the switch and regain control without any problem.

I hope this helps.

McGiverGim, 2020-06-26:
This option has nothing to do with this in theory. What it does is disarm the quad if something is wrong (quad not going to home, detect a crash, etc.)

marekwiecek, 2020-06-26:
@McGiverGim I understand that from the documentation: https://github.com/betaflight/betaflight/wiki/GPS-rescue-mode
However, I am just sharing the experience I have and a method that allowes me to regain control each time. Perhaps SANITY CHECK has an unintended consequence (just speculating) that prevents from being able to regain control. Above description of the process works for me.

etracer65, 2020-06-26:
You just need to move the flight control sticks (roll, pitch or yaw) to retake control. Sticks need to be moved as least 15% to signal that you're ready to retake control and then GPS Rescue will disconnect.

Daemon42, 2020-06-26:
@etracer65 , yep I did that, but it didn't work. 
I made sure the BBL link is shareabled and also attached it in github.  
If you add the RC Commands in a separate graph, can see at 5:17.082 I moved yaw quite a lot and moved both roll and pitch a little bit after that (wanted to avoid it just randomly flipping over if control was intermittent).  
The FC saw all my inputs, but everything was ignored.   It refused to exit failsafe.  

Oh, and just to rule it out, I do not have a mode switch to manually enable failsafe, so there's no way I could force it to get stuck in failsafe condition as the BBL shows.

Daemon42, 2020-06-26:
@marekwiecek If that sanity checks setting really has anything to do with this, then it's most certainly broken.    
I do have GPS rescue on a switch and the one thing I did not try was enabling it while it was stuck in failsafe, and then disabling it to see if would regain control.  Of course the documentation does not say that is necessary and I've seen videos indicating that it's not.    I've seen someone do the exact test I attempted.  Turn Tx off, let FS trigger GPS rescue, turn Tx back on, regain link, regain control in whatever mode it was previously in and fly away.   In my case, something prevented that.

etracer65, 2020-06-26:
@Daemon42 

> If you add the RC Commands in a separate graph, can see at 5:17.082 I moved yaw quite a lot and moved both roll and pitch a little bit after that

At the time you referenced you moved the yaw stick a maximum of 11%.

robert-b, 2020-06-26:
i had a fly away.
do not know exactly what to blame ...
but for sure there is something not so nice.
or how do call a fly away ...

Daemon42, 2020-06-26:
> 
> 
> @Daemon42
> 
> > If you add the RC Commands in a separate graph, can see at 5:17.082 I moved yaw quite a lot and moved both roll and pitch a little bit after that
> 
> At the time you referenced you moved the yaw stick a maximum of 11%.

I moved it to the normal extent I would move the sticks while flying as you can see from the earlier flying in the log.   I didn't bury the sticks in the corners because then I'd come out of failsafe doing 2 rolls/flips per second.   Are the RC Command input values it's looking at before or after the rates/expo have been applied?

etracer65, 2020-06-26:
> Are the RC Command input values it's looking at before or after the rates/expo have been applied?

Stick deflection percentage. Actually I was wrong about the 15%. The default is actually 30% stick deflection. See the `failsafe_stick_threshold` setting. Failsafe will not hand control back unless you exceed the stick deflection threshold. This is to prevent it immediately giving control back after signal returns where you may not be ready to take control. Based on the log everything looks normal to me and you just didn't move the sticks enough.

Daemon42, 2020-06-26:
FWIW, the fact that you have to move the sticks more than 15% from center to cancel failsafe, isn't in the GPS rescue documentation.
It says only "With this setting, GPS Rescue will activate in the event of a failsafe. However, it will return control to the user **as soon as the radio link comes back**. "   with no qualifications.

Daemon42, 2020-06-26:
Looking at the code I'm pretty sure that setting  is in uS not %, and I moved it a maximum of 57uS off center. 
It also does an interesting thing where it adds up the total deflection of all sticks to exceed that value, although I was only moving one at a time.

etracer65, 2020-06-26:
> Looking at the code I'm pretty sure that setting is in uS not %

No, it's a deflection percentage.

Daemon42, 2020-06-26:
Hmm, you're right, but this check wasn't always there in the context of GPS rescue.
From 4.0.x maintenance branch. 
```
 case FAILSAFE_GPS_RESCUE:
                if (receivingRxData) {
                    failsafeState.phase = FAILSAFE_RX_LOSS_RECOVERED;
                    reprocessState = true;
}
```
Then changed in 4.1 to 
```
 case FAILSAFE_GPS_RESCUE:
                if (receivingRxData) {
                    if (areSticksActive(failsafeConfig()->failsafe_stick_threshold)) {
                        failsafeState.phase = FAILSAFE_RX_LOSS_RECOVERED;
                        reprocessState = true;
                    }
}
```
I notice that iNav has a very similar threshold setting with check in failsafe.c in the same place.  But it is a uS value from 0-500 and it default to 50uS which is only 10%. 
I wonder if when this check was added in BF 4.1, a bit was lost in translation and the default value should have been 
closer to 10 instead of 30.    30% really is a lot for some quads.

etracer65, 2020-06-26:
It was added in 4.1 and is as designed. It was not a “mistake” ported from INAV. Look up the PR if you want more details. As I said earlier, everything worked as designed.

Daemon42, 2020-06-26:
Ok thanks.    Perhaps someone can update the GPS rescue documentation page.  It's still describing BF4.0 behavior, not BF4.1/2.  
I'll do a PR if you want.     I still assert that 30% deflection is way too big of a default.    All we're trying to do is detect
that someone has purposefully moved the sticks, not initiate a 200 degree per second roll the moment they regain control.

haslinghuis, 2020-06-26:
PR #7936

Daemon42, 2020-06-26:
> 
> 
> PR #7936

Very interesting.   iNav's method was specifically referenced as a possible solution, others agreed that it was the way to do it, 10% minimum deflection was suggested (which matches iNav default 50/500), and then it was implemented in BF.   Except the default was set to 30 instead of 10 without being mentioned anywhere in the PR.

etracer65, 2020-06-26:
> Ok thanks.    Perhaps someone can update the GPS rescue documentation page.  It's still describing BF4.0 behavior, not BF4.1/2.
>
> I'll do a PR if you want.     I still assert that 30% deflection is [...]

Feel free to update it. Betaflight is a community project and much of the documentation is provided by the users.

McGiverGim, 2020-06-26:
True. And for update the wiki, you have an edit button directly on the wiki. You don't need a PR or any kind of approval.

haslinghuis, 2020-06-26:
Thanks @Daemon42 for updating the wiki, I've added the PR to the 4.1 history

closed 2020-08-03 by stale[bot]

ollie2612, 2021-05-12:
Appreciate this is closed but can anyone confirm it was simply down to stick deflection? I had the exact same issue today.