betaflight/betaflight #9954: OSD: Laptime in the OSD via the craftname

realhuno, 2020-06-26:
**Is your feature request related to a problem? Please describe.**
working great with all other betaflight versions. but betaflight 4.2.0 broke this feature.
Working example with Betaflight 4.0.0! 
[image]
(on the top the time in ms.. craftname osd element overwrite via lua script)

**Describe the solution you'd like**
Move craftname back to active elements and not background like #9949 

**Describe alternatives you've considered**
create custom MSP command and OSD element

**Additional context**
hardware mod´s
https://github.com/realhuno/taranis_esp32_lua
https://github.com/realhuno/Chorus-RF-Laptimer/tree/telemetry

etracer65, 2020-06-26:
The Craft Name is a static field and appropriate to be part of the background as that reduces CPU load. The fact that you're using a custom Lua script to update the field in an unintended and unsupported way doesn't justify removing the optimization.

realhuno, 2020-06-27:
Is it possible to copy the craftname function call it laptimes and move to active osd elements? @etracer65

etracer65, 2020-06-27:
@realhuno Unless you could demonstrate an easy way that people could implement this on their own and it could become widely used, then it wouldn't make sense to create a new OSD element for this purpose. To me it seems like this is very much custom and not something that's well supported for general use. Like maintenance and distribution of the Lua script, the tools/hardware/? to connect some lap timer to the radio to get the time to send back via map, etc. If this could be turned into a project that was well documented and easily used by people then we might consider a dedicated element.

However if you wish to compile a custom version of the firmware for yourself that moved the craft name back to the foreground elements then the changes are small and you're welcome to do this for your own consumption. It's basically 2 changes to the `osd/osd_elements.c` file (one line updated, one line removed).

[image]

HazardCreative, 2020-06-28:
I'm definitely in favor of having an OSD string that can be written to.

The [RotorHazard race timer](https://github.com/RotorHazard/RotorHazard/) has been working toward lap times and splits displayed as OSD elements. Our most recent beta release publishes the necessary information to the network where other system can pick it up. 

The system works now with the ClearView2.0 receiver and publishes to its OSD. An ESP32 connects to the RotorHazard server over WiFi and outputs the serial commands that the CV needs. Here's some race footage from it in use yesterday: [3-consecutive qualifier](https://www.youtube.com/watch?v=wkvr3sqAN4A) | [Heads-up race](https://www.youtube.com/watch?v=QseZF8sXgp8).

We've been working toward publishing this information to Betaflight by routing the data over telemetry. The latest OpenTX version supports reading from a serial device, so our ESP32 system can be adapted to this protocol. @realhuno has been working on this and taken it to proof of concept—up to 4.2 release. Once proven functional, we will begin stabilizing the code and bringing it into the RotorHazard timer project.

etracer65, 2020-06-29:
See #9967. This makes changes so that the background Craft name element will be updated whenever the MSP command to update it is processed. This keeps the element in the background for the performance benefits and only causes a hit when the field is updated (when doesn't happen in normal use therefore not affecting the majority of users).

However in the long term this is not really the best way to implement the lap timer functionality. Because of your repurposing the field for unexpected use there's no way that we would know that other improvements would break this unplanned use. Once the lap timer project is more formalized then it would be best to revisit this in consideration of a general updatable OSD element field that is intended to updated via MSP during flight.

realhuno, 2020-06-30:
@etracer65 very nice ;-) thank you....... here i have the hardware mod for testing. https://github.com/realhuno/taranis_esp32_lua

closed 2020-07-05 by mikeller

mikeller, 2020-07-06:
@HazardCreative: This is actually a pretty neat idea for a home brew timing system. We could probably look into adding a dedicated OSD element to support this, if there are any benefits to it.

HazardCreative, 2020-07-06:
One thing we're trying to work though now is that sending the data over telemetry is restricted to numeric and not a character string. I wonder if a dedicated feature could allow us to encode data instead and get access to more characters that way? Or if the BetaFlight team had any other ideas on how a string might be transmitted via a radio and displayed?

mikeller, 2020-07-06:
What exactly are you trying to send from where to where?
In the case of SmartPort, arbitrary data can be sent by defining a custom 'sensor type', and then using the frames for this sensor as a container to transport the data - this is what is done for the 'MSP over SmartPort' feature at the core of the TX lua script.

HazardCreative, 2020-07-06:
My portion of the project is the timing, UI, and sending data out to the network—I'm less familiar with what happens after it's read by another device. @realhuno can speak more to the limitation he ran into here. Maybe this will fix it. I'll let him chime in.

realhuno, 2020-07-06:
i have here a very basic function...... so i just overwrite the craftname via msp. 

[image]

HazardCreative, 2020-07-06:
Ah, okay; I understand now that particular limitation is for getting data into the radio from certain methods (SPort).

Smeat, 2020-07-06:
Hi, @realhuno asked me to talk about my experience
I've been using a custom MSP message along with a custom OSD element. I am sending a 6 byte packet containing the laptime in ms, the lap number and pilot id. So from a technical standpoint it is pretty straight forward. 
Works without any problem using FrSky and the new time is pretty much instantly displayed.
But due to limited free time, I never managed to go beyond a simple proof of concept.

mikeller, 2020-07-07:
@realhuno: Understood. And to get data into Betaflight (either in the name OSD element or in some other form with a new MSP command, MSP is the correct protocol to be used. But apart from that, if you need to transfer data over SmartPort, you might be better off using a different protocol, like @Smeat explains, as the overhead for MSP is not needed for this purpose.

PR #9967 (2020-07-05): Refresh OSD background after MSP change to craft name
Fixes #9954 (even though that's using the field in an unintended way)

The OSD element "Craft name" is part of the OSD background so it's normally only rendered once. However that field can technically be updated at runtime using `MSP_SET_NAME`. Re-render the background as a result of this MSP command.

Reenables previously unexpected use of the Craft name field as a way to display updated data via Lua MSP messaging (see #9954).