betaflight/betaflight #9955: No Telemetry, No Buzzer, No GPS data

RookieFPV, 2020-06-27:
Hardware: 
ESC: AK32PIN 35A 4IN1 6S V2
FC: AIKON F7 MINI W/OSD V2HD [Target AIKONF7]
RX: Crossfire Nano (Crossfire MicroTX)
GPS: Beitan BN-180
DJI Vista

Issue: No Telemetry (voltage f.e. isnt detected at all in Betaflight configurator, voltage indicator at the very top at the configurator tells "0.00V(USB)"), No GPS (no data), Buzzer/Beeper doest respont. With Betaflight 4.1 everything works without any hardware/wiring changes.
[image]

Mayk-CLP, 2020-06-27:
wird das Board auch direkt über die ESC also ohne UBS Verbindung versorgt? Sieht irgendwie so aus als wenn die 5V Versorgung fehlt.

RookieFPV, 2020-06-27:
> wird das Board auch direkt über die ESC also ohne UBS Verbindung versorgt? Sieht irgendwie so aus als wenn die 5V Versorgung fehlt.

Wie gesagt, mit Betaflight 4.1 funktioniert alles. Ich gehe daher davon aus, das verkabelung und auch die hardware technisch ok ist.

Mayk-CLP, 2020-06-27:
Empfänger geht also du siehst wie sich die Knüppel bewegen? 

Flash mal zurück und schau ob es immer noch geht. Beim flashen immer "Chip vollständig löschen" anwählen! 

Wenn es mit 4.1 wieder geht kann es ja fast nur an einem falschen Pin Layout im 4.2 Aikon Target liegen. Wundert mich nur das es dann nicht mehr Meldungen gibt. Du könntest über dump das Pinning von 4.1 zu 4.2 prüfen ist aber etwas mühselig.

RookieFPV, 2020-06-27:
> Empfänger geht also du siehst wie sich die Knüppel bewegen?
>
> Flash mal zurück und schau ob es immer noch geht. Beim flashen immer "Chip vollständig löschen" anwählen!
>
> Wenn es mit 4.1 wieder geht [...]

reproduzierbar: mit 4.2 geht es nicht, mit 4.1 geht es. 2 mal hin und her geflshed, bleibt konstant so. ich kann mit 4.1 und mit 4.2 fliegen. lediglich habe ich mit 4.2 keinerlei telemetrie und keinen beeper/buzzer.

Mayk-CLP, 2020-06-27:
Hast du immer neu Konfiguriert oder mit CLI ein Backup hochgeladen?

RookieFPV, 2020-06-27:
> Hast du immer neu Konfiguriert oder mit CLI ein Backup hochgeladen?
> […](#)

neu konfiguriert. zum testen reichte mir ja der buzzer. die voltage sehe ich ja direkt in betaflight ohne etwas einstellen zu müssen.

etracer65, 2020-06-27:
As requested when you opened the issue, please provide the follow information:

> **Flight controller configuration**
> Create a `diff` and post it here in a code block. Put ``` (three backticks) at the start and end of the `diff` block (instructions  on how to do a diff: https://oscarliang.com/use-diff-not-dump-betaflight/)
> 
> Use `resource show all` to create a resource allocation list and post it here in a code block. Put ``` (three backticks) at the start and end of the output block.

RookieFPV, 2020-06-27:
```
board_name AIKONF7
manufacturer_id AIKO

# dma
dma ADC 1 NONE

# feature
feature -RX_PARALLEL_PWM
feature RX_SERIAL
feature GPS
feature TELEMETRY
feature OSD

# beeper
beeper -GYRO_CALIBRATED
beeper -RX_LOST_LANDING
beeper -DISARMING
beeper -ARMING
beeper -ARMING_GPS_FIX
beeper -BAT_CRIT_LOW
beeper -BAT_LOW
beeper -GPS_STATUS
beeper -ACC_CALIBRATION
beeper -ACC_CALIBRATION_FAIL
beeper -DISARM_REPEAT
beeper -ARMED
beeper -SYSTEM_INIT
beeper -ON_USB
beeper -BLACKBOX_ERASE
beeper -CRASH_FLIP
beeper -CAM_CONNECTION_OPEN
beeper -CAM_CONNECTION_CLOSE
beeper -RC_SMOOTHING_INIT_FAIL

# serial
serial 1 64 115200 57600 0 115200
serial 2 1 115200 57600 0 115200
serial 4 2 115200 57600 0 115200

# aux
aux 0 0 0 1950 2050 0 0
aux 1 1 1 1450 2050 0 0
aux 2 27 3 1950 2050 0 0
aux 3 46 1 1950 2050 0 0
aux 4 13 3 1450 2050 0 0
aux 5 26 2 1950 2050 0 0
aux 6 35 2 1450 1550 0 0

# master
set dyn_notch_width_percent = 0
set dyn_notch_q = 250
set acc_calibration = -10,-22,41,1
set mag_hardware = NONE
set baro_hardware = NONE
set min_check = 1005
set max_check = 1995
set rssi_channel = 9
set serialrx_provider = CRSF
set min_throttle = 1070
set dshot_bidir = ON
set use_unsynced_pwm = OFF
set motor_pwm_protocol = DSHOT300
set motor_pwm_rate = 480
set failsafe_switch_mode = KILL
set align_board_yaw = 180
set yaw_motors_reversed = ON
set small_angle = 180
set gps_provider = UBLOX
set gps_sbas_mode = EGNOS
set gps_ublox_use_galileo = ON
set gps_rescue_allow_arming_without_fix = ON
set deadband = 1
set yaw_deadband = 1
set pid_process_denom = 2
set osd_warn_batt_not_full = OFF
set osd_warn_batt_warning = OFF
set osd_warn_batt_critical = OFF
set osd_warn_visual_beeper = OFF
set osd_warn_esc_fail = OFF
set osd_warn_core_temp = OFF
set osd_warn_rc_smoothing = OFF
set osd_warn_gps_rescue_disabled = OFF
set osd_warn_rssi = ON
set osd_rssi_alarm = 40
set osd_cap_alarm = 1400
set osd_tim1 = 2561
set osd_vbat_pos = 2538
set osd_rssi_pos = 2543
set osd_tim_1_pos = 342
set osd_flymode_pos = 2330
set osd_mah_drawn_pos = 2506
set osd_gps_speed_pos = 2546
set osd_gps_sats_pos = 2362
set osd_home_dir_pos = 2512
set osd_home_dist_pos = 2533
set osd_altitude_pos = 2512
set osd_warnings_pos = 363
set osd_disarmed_pos = 2349
set osd_stat_tim_1 = ON
set osd_stat_tim_2 = OFF
set osd_stat_max_dist = ON
set osd_stat_battery = ON
set osd_stat_max_alt = ON
set osd_stat_bbox = OFF
set osd_stat_bb_no = OFF
set osd_stat_max_g_force = ON
set osd_stat_max_esc_temp = ON
set osd_stat_max_esc_rpm = ON
set osd_stat_min_link_quality = ON
set osd_stat_total_time = ON
set osd_stat_total_dist = ON
set osd_stat_min_rssi_dbm = ON
set gyro_rpm_notch_harmonics = 1

profile 0

rateprofile 0

# rateprofile 0
set roll_rc_rate = 90
set pitch_rc_rate = 90
set roll_srate = 80
set pitch_srate = 80
set yaw_srate = 75

---------------------------------

# Building AutoComplete Cache ... Done!
# 
# resource show all
Currently active IO resource assignments:
(reboot to update)
--------------------
A00: FREE
A01: FREE
A02: SERIAL_TX 2
A03: SERIAL_RX 2
A04: GYRO_CS 1
A05: SPI_SCK 1
A06: SPI_MISO 1
A07: SPI_MOSI 1
A08: FREE
A09: FREE
A10: FREE
A11: USB
A12: USB
A13: SWD
A14: SWD
A15: FREE
B00: FLASH_CS
B01: FREE
B02: PREINIT 2
B03: SPI_SCK 3
B04: SPI_MISO 3
B05: SPI_MOSI 3
B06: FREE
B07: FREE
B08: FREE
B09: FREE
B10: I2C_SCL 2
B11: I2C_SDA 2
B12: OSD_CS
B13: SPI_SCK 2
B14: CAMERA_CONTROL
B15: SPI_MOSI 2
C00: ADC_BATT
C01: ADC_CURR
C02: SPI_MISO 2
C03: FREE
C04: GYRO_EXTI
C05: FREE
C06: MOTOR 1
C07: MOTOR 2
C08: MOTOR 3
C09: MOTOR 4
C10: SERIAL_TX 3
C11: SERIAL_RX 3
C12: SERIAL_TX 5
C13: LED 1
C14: PINIO 1
C15: BEEPER
D00: FREE
D01: FREE
D02: SERIAL_RX 5
D03: FREE
D04: FREE
D05: FREE
D06: FREE
D07: FREE
D08: FREE
D09: FREE
D10: FREE
D11: FREE
D12: FREE
D13: FREE
D14: FREE
D15: FREE
E00: FREE
E01: FREE
E02: FREE
E03: FREE
E04: FREE
E05: FREE
E06: FREE
E07: FREE
E08: FREE
E09: FREE
E10: FREE
E11: FREE
E12: FREE
E13: FREE
E14: FREE
E15: FREE
F00: FREE
F01: FREE
F02: FREE
F03: FREE
F04: FREE
F05: FREE
F06: FREE
F07: FREE
F08: FREE
F09: FREE
F10: FREE
F11: FREE
F12: FREE
F13: FREE
F14: FREE
F15: FREE

Currently active Timers:
-----------------------
TIM1: FREE
TIM2: FREE
TIM3: FREE
TIM4: FREE
TIM5: FREE
TIM6: FREE
TIM7: FREE
TIM8:
    CH1: DSHOT_BITBANG 3
TIM9: FREE
TIM10: FREE
TIM11: FREE
TIM12:
    CH1: CAMERA_CONTROL
TIM13: FREE
TIM14: FREE

Currently active DMA:
--------------------
DMA1 Stream 0: FREE
DMA1 Stream 1: FREE
DMA1 Stream 2: FREE
DMA1 Stream 3: FREE
DMA1 Stream 4: FREE
DMA1 Stream 5: FREE
DMA1 Stream 6: FREE
DMA1 Stream 7: FREE
DMA2 Stream 0: FREE
DMA2 Stream 1: FREE
DMA2 Stream 2: DSHOT_BITBANG 3
DMA2 Stream 3: FREE
DMA2 Stream 4: FREE
DMA2 Stream 5: FREE
DMA2 Stream 6: FREE
DMA2 Stream 7: FREE

RookieFPV, 2020-06-27:
```
A00: FREE
A01: FREE
A02: SERIAL_TX 2
A03: SERIAL_RX 2
A04: GYRO_CS 1
A05: SPI_SCK 1
A06: SPI_MISO 1
A07: SPI_MOSI 1
A08: FREE
A09: FREE
A10: FREE
A11: USB
A12: USB
A13: SWD
A14: SWD
A15: FREE
B00: FLASH_CS
B01: FREE
B02: PREINIT 2
B03: SPI_SCK 3
B04: SPI_MISO 3
B05: SPI_MOSI 3
B06: FREE
B07: FREE
B08: FREE
B09: FREE
B10: I2C_SCL 2
B11: I2C_SDA 2
B12: OSD_CS
B13: SPI_SCK 2
B14: CAMERA_CONTROL
B15: SPI_MOSI 2
C00: ADC_BATT
C01: ADC_CURR
C02: SPI_MISO 2
C03: FREE
C04: GYRO_EXTI
C05: FREE
C06: MOTOR 1
C07: MOTOR 2
C08: MOTOR 3
C09: MOTOR 4
C10: SERIAL_TX 3
C11: SERIAL_RX 3
C12: SERIAL_TX 5
C13: LED 1
C14: PINIO 1
C15: BEEPER
D00: FREE
D01: FREE
D02: SERIAL_RX 5
D03: FREE
D04: FREE
D05: FREE
D06: FREE
D07: FREE
D08: FREE
D09: FREE
D10: FREE
D11: FREE
D12: FREE
D13: FREE
D14: FREE
D15: FREE
E00: FREE
E01: FREE
E02: FREE
E03: FREE
E04: FREE
E05: FREE
E06: FREE
E07: FREE
E08: FREE
E09: FREE
E10: FREE
E11: FREE
E12: FREE
E13: FREE
E14: FREE
E15: FREE
F00: FREE
F01: FREE
F02: FREE
F03: FREE
F04: FREE
F05: FREE
F06: FREE
F07: FREE
F08: FREE
F09: FREE
F10: FREE
F11: FREE
F12: FREE
F13: FREE
F14: FREE
F15: FREE

Currently active Timers:
-----------------------
TIM1: FREE
TIM2: FREE
TIM3: FREE
TIM4: FREE
TIM5: FREE
TIM6: FREE
TIM7: FREE
TIM8:
    CH1: DSHOT_BITBANG 3
TIM9: FREE
TIM10: FREE
TIM11: FREE
TIM12:
    CH1: CAMERA_CONTROL
TIM13: FREE
TIM14: FREE

Currently active DMA:
--------------------
DMA1 Stream 0: FREE
DMA1 Stream 1: FREE
DMA1 Stream 2: FREE
DMA1 Stream 3: FREE
DMA1 Stream 4: FREE
DMA1 Stream 5: FREE
DMA1 Stream 6: FREE
DMA1 Stream 7: FREE
DMA2 Stream 0: FREE
DMA2 Stream 1: FREE
DMA2 Stream 2: DSHOT_BITBANG 3
DMA2 Stream 3: FREE
DMA2 Stream 4: FREE
DMA2 Stream 5: FREE
DMA2 Stream 6: FREE
DMA2 Stream 7: FREE
```

etracer65, 2020-06-27:
This is your problem:

```
dma ADC 1 NONE

```
Reset the configuration by doing the `defaults` command in the CLI and reconfigure manually.

RookieFPV, 2020-06-27:
> This is your problem:
> 
> ```
> dma ADC 1 NONE
> ```
> 
> Reset the configuration by doing the `defaults` command in the CLI and reconfigure manually.

thank you.
what causes that issue?
could you describe which command after "defaults" i had to add in the cli for manually reconfiguration?

etracer65, 2020-06-27:
I don't know you you got this bad setting configured. The `defaults` command resets all the settings to the default values - just like after flashing. You will have to manually reconfigure all settings. Do not put the `dma ADC 1 NONE` command back in or it will stop working again.

RookieFPV, 2020-06-27:
ok. i think i did it right. all working now.

closed 2020-06-27 by etracer65