betaflight/betaflight #9971: no osd rssi with s bus telemetry ok

AkaNyclo, 2020-07-01:
I have a problem with the new betaflight 4.2,in my osd I see 0 RSSI or -130dbm but on my taranis it work without a problem and the link is stable, even if I put the rssi on ch16 it doesn't work in my osd

I have an r xsr with sbus and smartport, the telemetry work as I can see the ampere altitude on the osd

**Setup / Versions**
 - Flywoo goku f7 mini stack
 - Frsky R xsr
 - s bus and smartport

# version
# Betaflight / STM32F7X2 (S7X2) 4.2.0 Jun 14 2020 / 03:05:26 (8f2d21460) MSP API: 1.43
# config: manufacturer_id: FLWO, board_name: FLYWOOF7DUAL, version: 2ca86ba0, date: 2020-03-08T08:55:50Z

# start the command batch
batch start

# reset configuration to default settings
defaults nosave

board_name FLYWOOF7DUAL
manufacturer_id FLWO
mcu_id 003400393338510d32343933
signature 

# feature
feature -RX_PARALLEL_PWM
feature -RSSI_ADC
feature -LED_STRIP

# beeper
beeper -ARMING
beeper -ARMING_GPS_FIX
beeper -BAT_CRIT_LOW
beeper -BAT_LOW
beeper -GPS_STATUS
beeper -ACC_CALIBRATION
beeper -ACC_CALIBRATION_FAIL
beeper -READY_BEEP
beeper -DISARM_REPEAT
beeper -ARMED
beeper -ON_USB
beeper -CRASH_FLIP
beeper -CAM_CONNECTION_OPEN
beeper -CAM_CONNECTION_CLOSE
beeper -RC_SMOOTHING_INIT_FAIL

# map
map RETA1234

# serial
serial 0 64 115200 57600 0 115200
serial 1 32 115200 57600 0 115200
serial 4 2048 115200 57600 0 115200

# aux
aux 0 0 0 1025 1425 0 0
aux 1 1 1 900 1100 0 0
aux 2 2 1 1400 1600 0 0
aux 3 13 0 1900 1950 0 0
aux 4 13 0 1375 1425 0 0
aux 5 13 0 1675 1725 0 0
aux 6 13 0 1175 1225 0 0
aux 7 35 0 1725 1875 0 0
aux 8 35 0 1250 1325 0 0
aux 9 36 2 1525 1700 0 0

# adjrange
adjrange 0 0 3 900 1225 12 3 0 0
adjrange 1 0 3 1300 1675 12 3 0 0
adjrange 2 0 3 1750 2100 12 3 0 0

# rxrange
rxrange 0 987 2011
rxrange 1 987 2011
rxrange 2 987 2011
rxrange 3 987 2011

# vtxtable
vtxtable bands 6
vtxtable channels 8
vtxtable band 1 BOSCAM_A A FACTORY 5865 5845 5825 5805 5785 5765 5745 5725
vtxtable band 2 BOSCAM_B B FACTORY 5733 5752 5771 5790 5809 5828 5847 5866
vtxtable band 3 BOSCAM_E E FACTORY 5705 5685 5665    0 5885 5905    0    0
vtxtable band 4 FATSHARK F FACTORY 5740 5760 5780 5800 5820 5840 5860 5880
vtxtable band 5 RACEBAND R FACTORY 5658 5695 5732 5769 5806 5843 5880 5917
vtxtable band 6 IMD6     I CUSTOM  5732 5765 5828 5840 5866 5740    0    0
vtxtable powerlevels 4
vtxtable powervalues 14 20 26 30
vtxtable powerlabels 25 100 400 1W

# master
set gyro_lowpass2_hz = 375
set dyn_notch_width_percent = 0
set dyn_notch_q = 250
set dyn_notch_min_hz = 70
set dyn_notch_max_hz = 350
set dyn_lpf_gyro_min_hz = 300
set dyn_lpf_gyro_max_hz = 750
set acc_calibration = 109,-6,-123,1
set fpv_mix_degrees = 20
set blackbox_device = NONE
set dshot_bidir = ON
set motor_poles = 12
set bat_capacity = 850
set vbat_min_cell_voltage = 350
set vbat_warning_cell_voltage = 370
set vbat_scale = 113
set small_angle = 180
set osd_warn_batt_not_full = OFF
set osd_logo_on_arming = FIRST_ARMING
set osd_tim1 = 769
set osd_tim2 = 2560
set osd_vbat_pos = 14710
set osd_rssi_pos = 2145
set osd_rssi_dbm_pos = 129
set osd_tim_1_pos = 2402
set osd_flymode_pos = 2113
set osd_g_force_pos = 55
set osd_throttle_pos = 2370
set osd_vtx_channel_pos = 2081
set osd_ah_sbar_pos = 8398
set osd_ah_pos = 8270
set osd_current_pos = 2133
set osd_mah_drawn_pos = 2391
set osd_compass_bar_pos = 8234
set osd_altitude_pos = 2101
set osd_pidrate_profile_pos = 2118
set osd_warnings_pos = 14601
set osd_avg_cell_voltage_pos = 342
set osd_disarmed_pos = 14570
set osd_stat_battery = ON
set osd_stat_max_alt = ON
set osd_stat_max_g_force = ON
set vtx_band = 5
set vtx_channel = 5
set vtx_power = 1
set vtx_freq = 5806
set vcd_video_system = NTSC
set displayport_max7456_wht = 3
set gyro_1_align_pitch = 1800
set gyro_1_align_yaw = 1800
set gyro_2_align_yaw = 2700

profile 0

# profile 0
set dyn_lpf_dterm_min_hz = 105
set dyn_lpf_dterm_max_hz = 255
set dterm_lowpass2_hz = 225
set iterm_relax_type = GYRO
set p_pitch = 37
set d_pitch = 33
set f_pitch = 114
set p_roll = 34
set d_roll = 31
set f_roll = 108
set p_yaw = 36
set f_yaw = 108
set d_min_roll = 20
set d_min_pitch = 22
set ff_interpolate_sp = AVERAGED_3

profile 1

profile 2

# restore original profile selection
profile 0

rateprofile 0

# rateprofile 0
set roll_rc_rate = 124
set pitch_rc_rate = 124
set yaw_rc_rate = 174
set roll_expo = 30
set pitch_expo = 30
set roll_srate = 69
set pitch_srate = 69
set yaw_srate = 71

rateprofile 1

# rateprofile 1
set roll_rc_rate = 126
set pitch_rc_rate = 126
set yaw_rc_rate = 175
set roll_expo = 40
set pitch_expo = 40
set roll_srate = 71
set pitch_srate = 71
set yaw_srate = 74

rateprofile 2

# rateprofile 2
set roll_rc_rate = 120
set pitch_rc_rate = 120
set yaw_rc_rate = 175
set roll_expo = 5
set pitch_expo = 5
set roll_srate = 72
set pitch_srate = 64
set yaw_srate = 39

rateprofile 3

rateprofile 4

rateprofile 5

# restore original rateprofile selection
rateprofile 0

haslinghuis, 2020-07-01:
Please try `set rssi_invert = ON`
Be sure to use latest firmware for R-XSR and Taranis.

etracer65, 2020-07-02:
The FRSky receivers don't provide RSSI over telemetry via smartport. You can switch to the FPort firmware on the receiver and that will provide RSSI natively. Or you can stick with the smartport firmware and use alternate methods to get RSSI like using Lua scripts on the radio or rerouting the RSSI back on a channel in the receiver (look for youtube videos).

haslinghuis, 2020-07-02:
FPort is really preferred yes. Thanks @etracer65 for setting this straight, I missed that part. It even seems RSSI gets back to the radio even without S.Port configured [infofpv](https://intofpv.com/t-r-xsr-no-rssi-on-osd). Only have to configure an AUX on the transmitter to send the RSSI data back to the receiver/flight controller. That's why I suggested the missing `rssi_channel` setting.

More info: [5 Ways to Setup RSSI in your OSD, Analog, TX Telemetry, Receiver Channel, FPort, SBUS Packet Loss](https://www.youtube.com/watch?v=eMIExxWUlH4)

closed 2020-08-08 by stale[bot]