betaflight/betaflight #9973: improved "crash_recovery" feature in case of stuck props

Daniel-1276, 2020-07-02:
**Is your feature request related to a problem? Please describe.**
If you crash a quad it is most likely that props get stuck somehow and motors may get burned until you manually disarm.

**Describe the solution you'd like**
As we now have (free) rpm filters for nearly all BLHELI_S and BLHELI_32 ESCs it may be considered to add/extend crash_recovery feature that would turn of all motors if a single rpm measure is below idle rpm while armed. Configuration parameters could include a threshold in seconds until this feature kicks in (default could be something like 0.5sec). This delay time could prevent the activation while hitting a gate for a fraction of a second. This may correlates also with "crash_recovery = DISARM" feature.

**Describe alternatives you've considered**
"crash_recovery = DISARM" as far as I understand depends on gyro data to detect a crash and is prone to false positive detections, if drone got stuck in something soft for example.

**Additional context**
N/A

maybenikhil, 2020-07-04:
this is really interesting, it would probably have to be a CLI implementation for compatibility's sake, because it requires dshot, which not every quad is running

mikeller, 2020-07-06:
@Daniel-1276: This has been discussed a number of times (see #7854 for example) and has had at least one change implemented based on this: #7924. Have a look at this and see if it provides what you want.
Of course this feature could also be done based on measured RPM, but in this case it should be implemented in the ESC and not in the flight controller.

Daniel-1276, 2020-07-06:
Hi Mike, many thanks for looking into this topic. In the reference post you mentioned @etracer65 stated "And it's quite easy to have crash recovery settings that can be triggered in flight without a crash - even through quick rolls/flips."... That was my point raising this feature request to include another variable (motor RPM) to improve reliability of crash detection. If you think this is of no benefit, no worries :)

mikeller, 2020-07-06:
@Daniel-1276: It could potentially be added - but since there was pretty much no feedback whatsoever to the original 'crash disarm' implementation, and it was not possible to find anybody willing to test this, I am not sure if there is any value to adding something to functionality that has never been proven to actually work. I guess the point is that nobody is willing to crash and damage their equipment in order to verify and improve this.

closed 2020-08-16 by stale[bot]