betaflight/betaflight #9976: Request link in flight controller design resources page.

ModdQuad, 2020-07-03:
I was wondering if I could get a link on the 
https://github.com/betaflight/betaflight/wiki/Flight-Controller-Design
page.  As a firmware engineer, this page was very helpful to validate my hardware design.

My site offers a schematic for a full featured flight controller that has been built and [flown](https://www.youtube.com/watch?v=MtMT04ecjL4).  I decided not to go to market with the design because I did not see enough interest in yet another flight controller with software osd.  Also, I find sales and marketing painful and would rather work on other engineering projects.

ModdQuad
https://www.moddquad.com/

mikeller, 2020-07-06:
@ModdQuad: Feel free to add a link to the page - the wiki is editable by all GitHub users. Also, we are definitely interested in adding a target configuration for your flight controller if it is an open source design that can be used by other hobbyist - encouraging homebrew and open source is one of our project goals.

ModdQuad, 2020-07-06:
If I am reading drivers/mco.c correctly,  there is no way to add a target config for the board.   I need PC9 to output an 8MHz clock to drive the osd mcu.  (to save 50 cents off the bom)   old [issue](https://github.com/betaflight/betaflight/issues/9346)

mikeller, 2020-07-06:
@ModdQuad: Some progress was made towards this in #9464. Happy to accept pull requests to improve the configurability of this.

ModdQuad, 2020-07-07:
I only made three boards, never went into production.  I caught on early that runtime binding aka unified targets meant picking winners and losers without some sort of plugin interface.  No big deal, Its my normal job to do things like this, and they often fail for the oddest reasons.  As digital video systems will likely dominate the market in a few years, making this tech obsolete,  the window of opportunity has closed.  No time to play catch up.

closed 2020-07-07 by ModdQuad