betaflight/betaflight #9977: Uninverted SPort doesn't want to work

luqasz, 2020-07-04:
**Describe the bug**
SPort telemetry doesn't want to work on F4 flight controller with Jumper R1+ receiver (uninverted serial)

**To Reproduce**
Removed components from receiver:
- inverters 
- buffer
Connect RX from receiver to TX on FC, TX from receiver to RX on FC.

**Expected behavior**
SPort should work with uninverted serial.

**Flight controller configuration**
```
# diff all

# version
# Betaflight / STM32F405 (S405) 4.2.0 Jun 14 2020 / 03:04:22 (8f2d21460) MSP API: 1.43
# config: manufacturer_id: MTKS, board_name: MATEKF405STD, version: c4140206, date: 2020-05-02T03:17:47Z

# start the command batch
batch start

# reset configuration to default settings
defaults nosave

board_name MATEKF405
manufacturer_id MTKS
mcu_id 003100563152500e20333647
signature

# name: WSK

# resources
resource MOTOR 5 NONE
resource SERIAL_TX 11 A15

# feature
feature -SOFTSERIAL
feature -AIRMODE
feature GPS
feature RSSI_ADC

# beeper
beeper -GYRO_CALIBRATED
beeper -GPS_STATUS
beeper -ACC_CALIBRATION
beeper -ON_USB
beeper -BLACKBOX_ERASE

# beacon
beacon RX_LOST
beacon RX_SET

# serial
serial 0 2 115200 57600 0 115200
serial 2 2048 115200 57600 0 115200
serial 4 32 115200 57600 0 115200

# aux
aux 0 0 0 900 1200 0 0
aux 1 1 2 900 1200 0 0
aux 2 13 1 1700 2100 0 0
aux 3 26 0 900 1200 0 0

# rxrange
rxrange 0 987 2011
rxrange 1 987 2011
rxrange 2 987 2011
rxrange 3 987 2011

# vtxtable
vtxtable bands 5
vtxtable channels 8
vtxtable band 1 A        A FACTORY 5865 5845 5825 5805 5785 5765 5745 5725
vtxtable band 2 B        B FACTORY 5733 5752 5771 5790 5809 5828 5847 5866
vtxtable band 3 C        C FACTORY 5705 5685 5665    0 5885 5905    0    0
vtxtable band 4 D        D FACTORY 5740 5760 5780 5800 5820 5840 5860 5880
vtxtable band 5 E        E FACTORY 5658 5695 5732 5769 5806 5843 5880 5917
vtxtable powerlevels 3
vtxtable powervalues 0 1 2
vtxtable powerlabels 25 200 600

# master
set acc_calibration = 24,92,-133,1
set min_check = 1010
set max_check = 1990
set rssi_scale = 200
set bat_capacity = 1550
set vbat_scale = 113
set ibata_scale = 600
set small_angle = 180
set gps_provider = UBLOX
set gps_sbas_mode = AUTO
set gps_auto_baud = ON
set gps_ublox_use_galileo = ON
set gps_rescue_angle = 40
set gps_rescue_initial_alt = 80
set gps_rescue_descent_dist = 30
set tlm_inverted = ON
set tlm_halfduplex = OFF
set osd_warn_visual_beeper = OFF
set osd_warn_core_temp = OFF
set osd_warn_rssi = ON
set osd_cap_alarm = 1300
set osd_alt_alarm = 0
set osd_rssi_pos = 27
set osd_link_quality_pos = 27
set osd_flymode_pos = 2081
set osd_vtx_channel_pos = 2506
set osd_ah_sbar_pos = 2318
set osd_gps_speed_pos = 2113
set osd_gps_sats_pos = 2137
set osd_home_dir_pos = 2094
set osd_home_dist_pos = 2104
set osd_altitude_pos = 2497
set osd_warnings_pos = 14729
set osd_avg_cell_voltage_pos = 2519
set osd_disarmed_pos = 2315
set osd_stat_tim_2 = OFF
set osd_stat_max_dist = ON
set osd_stat_min_batt = OFF
set osd_stat_min_rssi = OFF
set osd_stat_bbox = OFF
set osd_stat_bb_no = OFF
set osd_stat_flight_dist = ON
set osd_stat_total_time = ON
set debug_mode = CYCLETIME
set vtx_band = 1
set vtx_channel = 1
set vtx_power = 1
set vtx_low_power_disarm = ON
set vtx_freq = 5865
set vcd_video_system = PAL
set camera_control_mode = DAC
set gyro_1_sensor_align = CW270
set gyro_1_align_yaw = 2700
set name = WSK

profile 0

# profile 0
set vbat_sag_compensation = 100
set ff_interpolate_sp = AVERAGED_4

profile 1

profile 2

# restore original profile selection
profile 0

rateprofile 0

rateprofile 1

rateprofile 2

rateprofile 3

rateprofile 4

rateprofile 5

# restore original rateprofile selection
rateprofile 0

# save configuration
save
```

```
# resource show all
Currently active IO resource assignments:
(reboot to update)
--------------------
A00: FREE
A01: FREE
A02: FREE
A03: SERIAL_RX 2
A04: FREE
A05: SPI_SCK 1
A06: SPI_MISO 1
A07: SPI_MOSI 1
A08: FREE
A09: SERIAL_TX 1
A10: SERIAL_RX 1
A11: USB
A12: USB
A13: SWD
A14: LED 2
A15: FREE
B00: FREE
B01: ADC_RSSI
B02: FREE
B03: SPI_SCK 3
B04: SPI_MISO 3
B05: SPI_MOSI 3
B06: I2C_SCL 1
B07: I2C_SDA 1
B08: FREE
B09: LED 1
B10: OSD_CS
B11: FREE
B12: FREE
B13: SPI_SCK 2
B14: SPI_MISO 2
B15: SPI_MOSI 2
C00: FREE
C01: SDCARD_CS
C02: GYRO_CS 1
C03: GYRO_EXTI
C04: ADC_CURR
C05: ADC_BATT
C06: MOTOR 1
C07: MOTOR 2
C08: MOTOR 3
C09: MOTOR 4
C10: SERIAL_TX 3
C11: FREE
C12: SERIAL_TX 5
C13: BEEPER
C14: FREE
C15: FREE
D00: FREE
D01: FREE
D02: FREE
D03: FREE
D04: FREE
D05: FREE
D06: FREE
D07: FREE
D08: FREE
D09: FREE
D10: FREE
D11: FREE
D12: FREE
D13: FREE
D14: FREE
D15: FREE
E00: FREE
E01: FREE
E02: FREE
E03: FREE
E04: FREE
E05: FREE
E06: FREE
E07: FREE
E08: FREE
E09: FREE
E10: FREE
E11: FREE
E12: FREE
E13: FREE
E14: FREE
E15: FREE
F00: FREE
F01: FREE
F02: FREE
F03: FREE
F04: FREE
F05: FREE
F06: FREE
F07: FREE
F08: FREE
F09: FREE
F10: FREE
F11: FREE
F12: FREE
F13: FREE
F14: FREE
F15: FREE

Currently active Timers:
-----------------------
TIM1: FREE
TIM2: FREE
TIM3:
    CH1: MOTOR 1
TIM4: FREE
TIM5: FREE
TIM6: FREE
TIM7: FREE
TIM8:
    CH2: MOTOR 2
    CH3: MOTOR 3
    CH4: MOTOR 4
TIM9: FREE
TIM10: FREE
TIM11: FREE
TIM12: FREE
TIM13: FREE
TIM14: FREE

Currently active DMA:
--------------------
DMA1 Stream 0: FREE
DMA1 Stream 1: FREE
DMA1 Stream 2: TIMUP 3
DMA1 Stream 3: FREE
DMA1 Stream 4: FREE
DMA1 Stream 5: FREE
DMA1 Stream 6: FREE
DMA1 Stream 7: FREE
DMA2 Stream 0: ADC 1
DMA2 Stream 1: TIMUP 8
DMA2 Stream 2: FREE
DMA2 Stream 3: FREE
DMA2 Stream 4: FREE
DMA2 Stream 5: FREE
DMA2 Stream 6: FREE
DMA2 Stream 7: FREE
```

**Setup / Versions**
betaflight 4.2
FC from eachine tyro 129 (also sold with tyro 119)

**Additional context**
Scope [screenshots](https://imgur.com/a/0HRzs39).
Yellow Jumper TX
Blue Jumper RX

mikeller, 2020-07-06:
@luqasz: This _looks_ like it should be working. Have you checked that it's not something in the TX side, like not having re-detected the sensors in OpenTX?

luqasz, 2020-07-06:
I have deleted and redetected all sensors in telementry openTX. No luck.

haslinghuis, 2020-07-06:
> resource MOTOR 5 NONE
> resource SERIAL_TX 11 A15

@luqasz You have resourced motor 5 pin to softserial, if you enable SOFTSERIAL feature you could test using s5 pad for smartport.
Assuming the receiver survived tampering, a dump would be helpful.

luqasz, 2020-07-06:
> you could test using s5 pad for smartport.

@haslinghuis  RX and TX on receiver are separate pads. Do you suggest to join them together on S5 and use softserial just to see if receiver survuved ?

haslinghuis, 2020-07-06:
No leave SBUS connected as is, but connect the receiver smartport RX pin to motor 5 pin on the FC and enable SOFTSERIAL feature. Save and reboot. On Ports tab enable smartport on the emulated UART.

Maybe have to swap values for tlm_inverted / tlm_halfduplex (I have to look it up but dinner is ready ;-)

luqasz, 2020-07-06:
S5 will be then bidirectional. This will not work as receiver has SEPARATE RX, TX SPort pads.

haslinghuis, 2020-07-06:
I meant connecting SPORT from jumper R1 to motor pad 5 on the FC. 
Enable SOFTSERIAL feature and change `set tlm_halfduplex = ON`

Just to confirm: SBUS is currently configured on RX2 on the FC.

http://www.mateksys.com/?portfolio=f405-std#tab-id-3

luqasz, 2020-07-06:
Since I've removed inverters from jumper receiver, RX and TX are now separate pins. Single wire will only work in one direction.

haslinghuis, 2020-07-06:
In my understanding on the receiver side SBUS = TX and SPort = RX ? I am missing something?

luqasz, 2020-07-07:
[this](https://imgur.com/a/A91wCSN) is how it looks like

haslinghuis, 2020-07-07:
Exactly ! Nice diagram. Yes indeed this is just to test if you receiver survived. Did you already hook up your analyzer to see something comes out?

luqasz, 2020-07-07:
Not yet. I wanted to fly this today (I have another one unmodified with SPort in softserial) I need to unsolder it first and setup my test place so will take some time.

etracer65, 2020-07-07:
> Since I've removed inverters from jumper receiver, RX and TX are now separate pins. Single wire will only work in one direction.

Smartport telemetry is bidirectional. Can't only connect the rx or tx from the receiver. I don't see any way you can make this work.

luqasz, 2020-07-07:
[Here](https://drive.google.com/file/d/1cz_RFXHD1umyTXSLW7eDClPAEQhxxLdM/view?usp=sharing) is sigrok/PulseView dump.

haslinghuis, 2020-07-07:
The file is private. Please attach the files by dragging & dropping, selecting or pasting them. Or click on the same message below in your response window.

luqasz, 2020-07-07:
Sorry. I've fixed file permission. Now you can download

luqasz, 2020-07-09:
@haslinghuis have you found anything ?

etracer65, 2020-07-09:
@luqasz As I said above, the Betaflight implementation is bidirectional half-duplex serial on a single wire. There is no facility to communicate using separate rx/tx connections.

luqasz, 2020-07-09:
Ok. I will test it with joined RX and TX from jumper R1+ and solder them to RX pad on FC.

@etracer65  So betaflight switchtches hardware UART pin in RX mode then TX mode ?

etracer65, 2020-07-09:
No. On STM32 MCU's the TX line of the UART can be configured to run in half-duplex bidirectional mode. That's what Betaflight uses. You're not going to be able to just connect the rx/tx on the receiver together. That's why there was a buffer/inverter chip on the receiver. It's not going to work.

luqasz, 2020-07-09:
Ok. After reading [this](https://quadmeup.com/smartport-inverter-for-f4-flight-controllers/) I am confused. Does that blog post apply only for older versions of BF ?

haslinghuis, 2020-07-09:
This is interesting. You have to use `set sport_halfduplex = ON` according to that. Not sure how to wire this up? Receiver RX+TX -> FC Smartport pad? 

> I havent' mentioned SBUS at all.

How about SBUS, it is configured so this is connected and working?

luqasz, 2020-07-09:
> This is interesting. You have to use set sport_halfduplex = ON according to that.

```
# get sport_halfduplex
###ERROR: get: INVALID NAME###
```
> Not sure how to wire this up? Receiver RX+TX -> FC Smartport pad? This would be TX pad not RX pad. 

I will figure something out.

> How about SBUS, it is configured so this is connected and working?

SBUS is working since board has a inverter so no problem there.

etracer65, 2020-07-09:
I really don't understand why you removed components from the receiver. If anything you just needed to bypass the inverter, not split the rx/tx. I haven't looked deeply into the code by there's a chance that if you do `set tlm_halfduplex =OFF` and then connect the receiver rx to the FC tx and receiver tx to fc rx it _might_ work. But I don't have high hopes as it will no longer be a half-duplex communication.

haslinghuis, 2020-07-09:
Smartport using softserial would be simplest solution. You also can turn off the hardware inversion circuit on F4 FC if it uses a GPIO pin to trigger the inversion: https://intofpv.com/t-turning-off-inversion-circuit-on-f4-flight-controller

luqasz, 2020-07-09:
> I really don't understand why you removed components from the receiver. If anything you just needed to bypass the inverter, not split the rx/tx. I haven't looked deeply into the code by there's a [...]

That is exactly what I did. Split up RX and TX lines.

> Without the hack Smartport using softserial would be simplest solution. You also can turn off the hardware inversion circuit on F4 FC if it uses a GPIO pin to trigger the inversion: https://intofpv.com/t-turning-off-inversion-circuit-on-f4-flight-controller

No `INVERTER` in https://github.com/betaflight/betaflight/blob/master/src/main/target/MATEKF405/target.h

etracer65, 2020-07-10:
> That is exactly what I did. Split up RX and TX lines. I've even proved (with pluse view dumps) that BF gets packets and sends data back to jumper.

The point is that you shouldn't have split the signal lines as Betaflight is designed to do smartport over a single line using half-duplex.

If you had used softserial for smartport as suggested you could have avoided making any changes. Softserial can handle a bidirectional inverted signal. No reason to modify the receiver.

luqasz, 2020-07-10:
> If you had used softserial for smartport as suggested you could have avoided making any changes. Softserial can handle a bidirectional inverted signal. No reason to modify the receiver.

With softserial all works (with unmodified receiver). With softserial, just after I power on quad, I get "sensor lost" in my radio for about 5 minutes. Then it settles.

My point is to use hardware serial as it will use DMA etc. without wasting CPU cycles. 

> The point is that you shouldn't have split the signal lines as Betaflight is designed to do smartport over a single line using half-duplex.

Ok so probably on older versions of BF, full duplex worked as mentioned in blog [post](https://quadmeup.com/smartport-inverter-for-f4-flight-controllers/).

haslinghuis, 2020-07-10:
As pointed out by @etracer65, Betaflight is not functionally designed to support such use case. If it worked on previous versions it is no recipe for success in the future. Although it is an interesting approach and still am curious. Some receivers just have an non-inverted pad to bypass the inverter. In light of this I think there is no ground for supporting this.

luqasz, 2020-07-11:
I've figured it out. Reason for why direct connection (RX-TX) didn't work is because jumper didn't see it's own TX packet. Once I've connected jumper TX to RX and set `tlm_halfduplex = off` telemetry works.

[image]

Concluding. Receiver must see it's own packets in order to work. I will test with FPort and report here.

etracer65, 2020-07-11:
A better solution would be to not remove components from your receiver in the first place. None of this is really helpful and not a use-case that anybody should be attempting.

mikeller, 2020-07-12:
@luqasz: Interesting, thanks for documenting your findings. This is the exact opposite of the use case in Betaflight when SmartPort / FPort is done with separate TX / RX pins on the MCU, in the case of F4 boards with an external inverter. In this use case, special code is necessary to prevent the receiving side from throwing an error when 'receiving' the output that is sent on the TX pin.