betaflight/betaflight #9978: FRSKY OSD FREEZES

JCFPV, 2020-07-04:
Matek F722px
Bn 880 gps
Frsky osd built in 
Betaflight 4.2

Dear Betaflight I recently upgraded to a matek F722px with graphical Frsky osd built in once I set everything up and started flying the osd started freezing up, the home arrow is freezing in location and then it unfreezes and updates info only to freeze again I was just running a mamba f405 mk2 flight controller and esc and all was great all I changed was the FC to give me more uarts once I disabled the frsky osd in the ports tab I have no osd at all please help me thanks for your time.

ModdQuad, 2020-07-04:
Try updating the osd software (if you can).  I tested an early unit in January and my scope did not pick it up as a valid PAL or NTSC signal, so I lost interest because I could not measure its jitter easily, could not lock on to a specific scan line.

JCFPV, 2020-07-05:
There is no way to update software on the built in osd chip and by disabling it there is no osd I feel like this is a bigger problem between betaflight and frsky I wish they would work out there differences so I could enjoy the new matek f722px FC it has given me more problems then ever before and I am pretty much done with it what a  $50 waste of money

ModdQuad, 2020-07-05:
When I looked at FrSky OSD, I found its interface quite flexible, but too complicated, too big a surface for bugs to appear.  It looks like its up to the betaflight team to find and work around these bugs.  I am not part of the BF team, I created a different software OSD that mimicked the interface of a MAX7456 with addition registers.  I was going to open source it, after a few thousand units were sold (to insure I get the credit for my work).  But I failed to generate enough interest.  I am just hear to request a link (2 issues below) in hope someone will better luck than me will continue.  Nothing useful to contribute here, signing off.

JCFPV, 2020-07-05:
I truly appreciate all the help and the funny thing is I would have no problems paying for Inav or Betaflight so that those who contribute so much are rewarded for their time as long as it worked Thanks for checking into the matter I bought this FC solely for inav thinking it would work better all in do time I guess,  the weird part is everything worked flawlessly before switching to this FC not with inav but on betaflight

ModdQuad, 2020-07-05:
If the frsky osd works in inav, then it should work eventually in betaflight, be patient.   btw there are moneyed interests in the form of sponsors.  I was locked out of unified targets and wondered if it was because of this (I needed pitch and roll written to registers in a AD7456 backwards compatible way).  I am also curious if they will link to my project as it has a schematic for a [working](https://youtu.be/MtMT04ecjL4) flight controller with a software osd.  My open sourced approach would have been better in the long run, but most of the FC are designed and made in China.

mikeller, 2020-07-05:
@JCFPV: Maybe if you added the information you were asked to provide when opening this issue, and not just expected that others help you for free based on guesswork there would be a better chance of somebody taking the time to look into your problem.

JCFPV, 2020-07-06:
I understand mikeller I truly appreciate your time the only reason i didn't attach additional info was i could not get github to open issue tab from my computer so i tried to put as much as I could from my phone I am still new to all of this and I couldn't get a dump or any videos uploaded from my phone. I dont appreciate the attitude though I wasn't trying to upset you just looking for help.

JCFPV, 2020-07-06:
Hey moddquad I truly appreciate your time and help with this matter I didn't realize I was caught in the middle of business I figured betaflight was compensated if not by frsky then by matek I just bought a new flight controller thinking it would be great considering it says betaflight compatible I figured it would work on betaflight It's like buying a laptop only to find out its blank and wont work with said program. Thanks for your time hopefully in time everyone will resolve there issues to create amazing software which I wouldn't mind buying,

ModdQuad, 2020-07-06:
I don't want to start any conspiracy theory here.   I know nothing of about betaflight finances, other than there is a sponsor button with a pink heart.

JCFPV, 2020-07-06:
I definitely don't want to start anything or have any problems either I just want to be able to flash my fc and go fly with BF4.2 before the fc switch I was having a blast with the software and love the easy of use I have been in fpv for 6 months and never had this problem before. I was more or less looking for a little help to see if the new FC was bad 👍

ModdQuad, 2020-07-06:
Unfortunately its more of an early adopter issue.   Give it a week or two.  A reproducible bug description or video would help narrow down your particular issue.

JCFPV, 2020-07-06:
I agree as soon as I can get my computer to work I will upload all the stuff needed on github from now on I didn't realize I would cause any problems and I am definitely new to all this

ModdQuad, 2020-07-06:
Naw, its me.  I'm just bitter at the analytics of my website and videos.  When back to one of those silicon valley jobs, so no time anymore.

JCFPV, 2020-07-07:
After working with
 rdq we have determined the flight controller is possibly bad and are going to try replacing it first to see if the same issues happen I dont feel at this point it's a betaflight issue but I will keep everyone posted Thanks for all the help

closed 2020-07-07 by JCFPV