betaflight/betaflight #9979: Motor 2 and 4 not spinning in horizon mode nd angle mode

cdogg22, 2020-07-04:
I have the new kakute mini f7 stack version 2. Well i put betaflight 4.2 on it. Well the motors start nd work with no issues in acro mode but if I try to use horizon or angle mode. Motor 2 and 4. Won't start up. They stutter once in a while maybe make a full spin once or twice. But motors 1 nd 3. Spin fine.

maybenikhil, 2020-07-05:
could you post a diff + also a screenshot of your ESC firmware?

cdogg22, 2020-07-05:
diff

# version
# Betaflight / STM32F745 (S745) 4.2.0 Jun 14 2020 / 03:05:04 (8f2d21460)
MSP API: 1.43
# config: manufacturer_id: HBRO, board_name: KAKUTEF7MINI, version:
2c2f9c32, date: 2020-06-16T20:17:34Z

# start the command batch
batch start

board_name KAKUTEF7MINI
manufacturer_id HBRO

# name: MINI

# feature
feature -RX_PARALLEL_PWM
feature RX_SERIAL
feature TELEMETRY
feature ESC_SENSOR

# beacon
beacon RX_SET

# map
map TAER1234

# serial
serial 5 64 115200 57600 0 115200
serial 6 1024 115200 57600 0 115200

# aux
aux 0 0 2 1300 2100 0 0
aux 1 2 2 1700 2100 0 0
aux 2 13 6 1300 1700 0 0
aux 3 26 5 1300 1700 0 0
aux 4 31 5 1700 2100 0 0
aux 5 35 6 1700 2100 0 0
aux 6 36 7 1150 2100 0 0
aux 7 49 0 1525 2100 0 0

# rxrange
rxrange 0 988 2011
rxrange 1 988 2011
rxrange 2 988 2011
rxrange 3 989 2009

# master
set gyro_lowpass2_hz = 325
set dyn_notch_width_percent = 0
set dyn_notch_q = 250
set dyn_notch_min_hz = 100
set dyn_notch_max_hz = 350
set dyn_lpf_gyro_min_hz = 260
set dyn_lpf_gyro_max_hz = 650
set acc_calibration = -188,42,-155,1
set baro_hardware = NONE
set max_check = 1960
set serialrx_provider = CRSF
set dshot_bidir = ON
set motor_pwm_protocol = DSHOT600
set motor_poles = 12
set yaw_motors_reversed = ON
set small_angle = 180
set osd_units = IMPERIAL
set osd_vbat_pos = 14562
set osd_link_quality_pos = 2211
set osd_rssi_dbm_pos = 131
set osd_tim_2_pos = 14506
set osd_flymode_pos = 289
set osd_throttle_pos = 2375
set osd_crosshairs_pos = 2413
set osd_current_pos = 14644
set osd_mah_drawn_pos = 2325
set osd_gps_speed_pos = 2315
set osd_gps_sats_pos = 2147
set osd_warnings_pos = 2185
set osd_avg_cell_voltage_pos = 2306
set osd_disarmed_pos = 2256
set osd_esc_tmp_pos = 2125
set osd_core_temp_pos = 2120
set osd_log_status_pos = 2243
set debug_mode = GYRO_SCALED
set gyro_1_align_yaw = 2700
set name = MINI

profile 0

# profile 0
set dyn_lpf_dterm_min_hz = 91
set dyn_lpf_dterm_max_hz = 221
set dterm_lowpass2_hz = 195
set vbat_pid_gain = ON
set p_pitch = 55
set i_pitch = 108
set d_pitch = 46
set f_pitch = 137
set p_roll = 50
set i_roll = 102
set d_roll = 42
set f_roll = 130
set p_yaw = 54
set i_yaw = 108
set f_yaw = 130
set d_min_roll = 28
set d_min_pitch = 30
set idle_min_rpm = 2

rateprofile 0

# rateprofile 0
set rates_type = ACTUAL
set roll_rc_rate = 20
set pitch_rc_rate = 20
set yaw_rc_rate = 20
set roll_expo = 60
set pitch_expo = 60
set yaw_expo = 60
set roll_srate = 100
set pitch_srate = 100
set yaw_srate = 100

etracer65, 2020-07-09:
This is your problem:
```
set idle_min_rpm = 2
```

cdogg22, 2020-07-09:
Ok gonna try right now. I'll let you know. Thanks alot

cdogg22, 2020-07-09:
Ok. Think you nailed it. I set it to 20. Then armed motors all armed.
Unlike before if i rolled left or right. Motors 2 nd 4. Would stop. Now
they stay on and speed up. Thank you. Never. Had a issue like this on
3.5>4.1. Thanks again

etracer65, 2020-07-09:
Dynamic idle is a new feature in 4.2. Seems like you should turn it back off by setting to 0 (the default).

Otherwise you should read the documentation if you intend to use new features.

https://github.com/betaflight/betaflight/wiki/Tuning-Dynamic-Idle

cdogg22, 2020-07-09:
Actually reading about it now. Thanks for the info

closed 2020-07-09 by etracer65