betaflight/betaflight #9986: OSD Warning LQ (Crossfire v2) Alarm Threshold

N4V1G4T0R, 2020-07-06:
When LQ Warning (Link Quality) is activated in the OSD, the warning shows up too "early". It should not show up at 2:xx at all, as this is uncritical for Crossfire.

In the CLI LQ Alarm is set to 80 (default, "osd_link_quality_alarm = 80") which is fine. But only at RMFD below 2 (<1:80).

This appears on all tested hardware (racepit mini, airbot nano v6, v7, etc.)
Firmware is BF 4.2
Crossfire Micro TX Module (FW V3.28/HW V1.03) with Nano RX (FW 3.28/HW V1.32)

Request: OSD Warning LQ (Crossfire v2) including RMFD <1 as alarm threshold.

(It is a KISS ESC, in case you wonder about the name ;-))
[image]

etracer65, 2020-07-06:
> It should not show up at 2:xx at all, as this is uncritical for Crossfire.

Unless the user has their radio locked in RFMD 2 then it's extremely critical. Betaflight has no way of knowing if this is the case or not.

N4V1G4T0R, 2020-07-06:
Who would to such thing (locking at RFMD 2)?! 😱 If so, I agree, that is critical under this circumstance.

Well, it is possible to set the RFMD and LQ% value then? I assume most of the ppl will get this warning for no reason, because crossfire will switch to FRMD1 and everying is fine.

0crap, 2020-07-17:
Both of you have a valid point.

I do agree it would be nice if the end user can configure RFMD to be part of the alarming.
I don't lock my CRSF to RFMD 2 because I prefer NOT to have a failsave at the cost of a tiny bit more delay.
Seems to make most sense for almost all of us IMHO.

mikeller, 2020-07-19:
@0crap:

> I don't lock my CRSF to RFMD 2 because I prefer NOT to have a failsave at the cost of a tiny bit more delay.

There are a lot of reports from users who, when not locked to RFMD 2, experience a fall back to RFMD 1 on just about every flight. From a low update rate perspective this is not acceptable for most users for racing / acrobatics, so I think with Betaflight targeting these areas, treating 'locked to RFMD 2' as the default seems to be a reasonable assumption top make.

N4V1G4T0R, 2020-07-20:
@mikeller OK, I see the argument. It makes sense to make a firmware as foolproof as possible.

"There are a lot of reports from users who, when not locked to RFMD 2, experience a fall back to RFMD 1 on just about every flight." I actually believe that. Ppl forgot how to position an antenna correctly, as XF is working really well even with badly placed antennas. But let's not start that discussion. ;-)

Anyway, if LQ Alarms would be handled just a little better, I - and probably all XF Users - would be thankful (see #9839). So, I'll close that issue now.

closed 2020-07-20 by N4V1G4T0R