betaflight/betaflight #9994: no gyro / acc after betaflight firmware flash from 4.0 to 4.2

Stevie83dtl, 2020-07-07:
First... i'm from germany and my english isn't the best.

After i flashed my Airbot F4 v2 Flightcontroller successfully to fw 4.2 , the gyro and accellerotor are not detected anymore.
I'm was searching for troubleshooting for hours and hours, but still found nothing.

I request some help of an pro please!

diff
version Betaflight / STM32F405 (S405) 4.1.5 Mar 16 2020 / 05:19:58 (d4e74e3) MSP API: 1.42
manufacturer_id: AIRB board_name: AIRBOTF4 custom defaults: YES
start the command batch
batch start

board_name AIRBOTF4
manufacturer_id AIRB

profile 0

rateprofile 0

end the command batch
batch end

status
MCU F40X Clock=168MHz (PLLP-HSE), Vref=3.30V, Core temp=50degC
Stack size: 2048, Stack address: 0x1000fff0
Config size: 3792, Max available config: 16384

System Uptime: 240 seconds, Current Time: 2020-07-07T12:41:21.448+00:00
CPU:0%, cycle time: 0, GYRO rate: 0, RX rate: 33, System rate: 9
Voltage: 0 * 0.01V (0S battery - NOT PRESENT)
I2C Errors: 0
SD card: Not configured
Arming disable flags: NOGYRO CLI MSP

Asizon, 2020-07-07:
@Stevie83dtl im not sure of witch is AirbotF4 V2 flight controller, can you share more details?

mikeller, 2020-07-07:
@Stevie83dtl:

> version Betaflight / STM32F405 (S405) 4.1.5 Mar 16 2020 / 05:19:58 (d4e74e3) MSP API: 1.42

Flashing obviously failed, as this is not Betaflight 4.2 - use the latest version of Betaflight configurator to properly installing the firmware.

Stevie83dtl, 2020-07-07:
The FC is the Airbot F4 ,an older model from Airbot... not Onibus!

I flashed diffrent Versions of betaflight and diffrent targets. Nothing worked:(

etracer65, 2020-07-07:
Provide pictures of both sides of the board.

Stevie83dtl, 2020-07-08:
[image]
[image]

Stevie83dtl, 2020-07-08:
Before updating and crashing evrerything worked fine on BF 4.0

Asizon, 2020-07-08:
This Flight controller doesnt have dedicated Target, seems an F4 board that comes from Airbot F7 board. @x4FF3 what do you think?

Stevie83dtl, 2020-07-08:
it's the Airbot F4 but in version 2.
I guess this Bord has 2 Gyros.

Asizon, 2020-07-08:
Its a Airbot F7 with F405 processor, dual gyro board but with one only gyro. As I said, i think there is no Dedicated target for it, so your gyro resources are not configured. Sure that it comes with a custom build from manufacturer

Stevie83dtl, 2020-07-08:
Ok.
The Target for the F7 Bord i also tried, but then i just can find the bord with a special tool called ,,impulse rc driver fixer".
But just in DFU mode.

mikeller, 2020-07-08:
Looks like this guy: https://cn.amaxshop.com/index.php?route=product/product&product_id=162

I'd be surprised if Airbot has ever heard of this target. You'll have to ask your supplier to provide you with firmware for it.

etracer65, 2020-07-08:
If it's a "dual-gyro" version then it needs to use the `AIRBOTF4SD` target.

x4FF3, 2020-07-08:
ok, so this seems like a version which i was told wouldnt be sold. but it seems it is...ill try to acquire info about it, and if i get some. ill provide a target.

Stevie83dtl, 2020-07-08:
#x4FF3 Thanx a lot

Stevie83dtl, 2020-07-08:
@entracer65 
I tried this but also no success.

Ndarcoy, 2020-07-10:
Same issue with me, I use iflight succex F4 whoop, this afternoon I upgrade from 4.1.2 to 4.2 and this issue come in, I try to downgrade to 4.1.7 it's still same, I return to 4.1.2 still same, but cli said my gyro is ok, can anybody please help me to solve this issue, I hope it's not hardware problem, because the gyro is work before I upgrade the firmware

Ndarcoy, 2020-07-10:
Oh by the way, I try both stm32f411 and iff411pro, still same

Stevie83dtl, 2020-07-10:
@Ndarcoy
Do you know the exact label of your flightcontroller?
If possible make some pics please.

x4FF3, 2020-07-14:
hi,

so i made a target based on the info i got. waiting for my board.
@Stevie83dtl could you test the attached firmware?
[betaflight_4.2.0_AIRBOTF4V2_8f2d21460.zip](https://github.com/betaflight/betaflight/files/4921375/betaflight_4.2.0_AIRBOTF4V2_8f2d21460.zip)

Stevie83dtl, 2020-07-14:
Jepp... in progress.

Stevie83dtl, 2020-07-14:
# 

# diff

# version
# Betaflight / AIRBOTF4V2 (AF42) 4.2.0 Jul 14 2020 / 11:41:38 (8f2d21460) MSP API: 1.43

# start the command batch
batch start

board_name AIRBOTF4V2

# name: Stevie-Bee

# feature
feature -SOFTSERIAL
feature -TELEMETRY
feature LED_STRIP

# beeper
beeper -ON_USB
beeper -BLACKBOX_ERASE

# beacon
beacon RX_LOST
beacon RX_SET

# map
map TAER1234

# serial
serial 0 64 115200 57600 0 115200
serial 4 8192 115200 57600 0 115200
serial 5 0 115200 57600 0 115200

# led
led 0 0,0::C:9
led 1 1,0::C:9
led 2 0,1::A:0
led 3 1,1::A:0

# mode_color
mode_color 6 0 2
mode_color 6 1 6
mode_color 7 0 5

# aux
aux 0 0 0 1850 2100 0 0
aux 1 2 1 1300 1700 0 0
aux 2 27 2 1400 2100 1 0
aux 3 13 2 1700 2100 0 0
aux 4 35 4 1725 2075 0 0
aux 5 39 4 900 1300 0 0
aux 6 47 3 1700 2100 0 0

# master
set acc_calibration = 94,12,-56,1
set motor_pwm_protocol = DSHOT600
set align_board_yaw = 180
set beeper_dshot_beacon_tone = 3
set small_angle = 180
set deadband = 3
set yaw_deadband = 3
set name = Stevie-Bee

profile 0

# profile 0
set i_pitch = 70
set d_pitch = 37
set i_roll = 60
set p_yaw = 35
set i_yaw = 100
set horizon_level_strength = 65
set horizon_transition = 85

rateprofile 0

# rateprofile 0
set roll_rc_rate = 129
set pitch_rc_rate = 129
set roll_expo = 21
set pitch_expo = 21
set yaw_expo = 11
set roll_srate = 77
set pitch_srate = 77
set yaw_srate = 71

# end the command batch
batch end

# status
MCU F40X Clock=168MHz (PLLP-HSE), Vref=3.30V, Core temp=56degC
Stack size: 2048, Stack address: 0x1000fff0
Config size: 3714, Max available config: 16384
Gyros detected: gyro 1
GYRO=ICM20602, ACC=ICM20602
OSD: MAX7456
System Uptime: 400 seconds, Current Time: 2020-07-14T21:35:30.892+00:00
CPU:6%, cycle time: 125, GYRO rate: 8000, RX rate: 111, System rate: 9
Voltage: 1170 * 0.01V (3S battery - OK)
I2C Errors: 0
Arming disable flags: CLI

Stevie83dtl, 2020-07-14:
So far, so good...
But in Angel and Horizon Mode it's totally freaking out.

x4FF3, 2020-07-15:
> So far, so good...
> But in Angel and Horizon Mode it's totally freaking out.

what do you mean by freaking out?

Stevie83dtl, 2020-07-15:
After activation of one the both modes it flipt over and two motors are turning up higher and higher and there's no response anymore when  i turn back the Throttlestick.

etracer65, 2020-07-15:
> After activation of one the both modes it flipt over and two motors are turning up higher and higher and there's no response anymore when i turn back the Throttlestick.
> I have to disarm immediatelly or turn on the failsave switch

You need to calibrate the accelerometer.

Stevie83dtl, 2020-07-15:
believe me... done first!
used a levelmeter  to align the drone.

etracer65, 2020-07-15:
Either you're not calibrating the accelerometer or you have the sensor alignment set incorrectly. Make sure the 3D model on the Setup tab exactly matches the movement of the quad and that there are no strange overshoots or slow drifts in position.

Stevie83dtl, 2020-07-16:
Mission accomplished.
The  Flashdisk was full (16 of 16 MB).  -> deleted!
And maybe the crashflip mode was part of the problem, so i also deleted it from the mode-list. Now it works.
But i had just a few minutes for a very short testflight.

Stevie83dtl, 2020-07-16:
Thanks to all helping hands!!!

mikeller, 2020-07-19:
@x4FF3: Thanks for looking into this - will you be opening a pull request for a Unified Target for this?

x4FF3, 2020-07-19:
> @x4FF3: Thanks for looking into this - will you be opening a pull request for a Unified Target for this?

yes when i get my board and can test the target.

x4FF3, 2020-07-31:
i got my board and will test asap to provide the target.

closed 2020-10-12 by stale[bot]

knnadeera, 2022-12-02:
> hi,
> 
> so i made a target based on the info i got. waiting for my board. @Stevie83dtl could you test the attached firmware? [betaflight_4.2.0_AIRBOTF4V2_8f2d21460.zip](https://github.com/betaflight/betaflight/files/4921375/betaflight_4.2.0_AIRBOTF4V2_8f2d21460.zip)

I have same problem with my HGLRC XJB F428 FC. no GYRO after firmware updating.
I have two of them. one is defected and on is working.
these are the resources from defected one. I can give  you working one resource as well.
could you please help me

# version
# Betaflight / OMNIBUSF4 (OBF4) 4.1.1 Nov 15 2019 / 12:41:27 (1e5e3d369) MSP API: 1.42

# resource
resource BEEPER 1 B04
resource MOTOR 1 B00
resource MOTOR 2 B01
resource MOTOR 3 A03
resource MOTOR 4 A02
resource MOTOR 5 A01
resource MOTOR 6 A08
resource PPM 1 B14
resource PWM 1 B14
resource PWM 2 B15
resource PWM 3 C06
resource PWM 4 C07
resource PWM 5 C08
resource PWM 6 C09
resource SONAR_TRIGGER 1 A01
resource SONAR_ECHO 1 A08
resource LED_STRIP 1 A01
resource SERIAL_TX 1 A09
resource SERIAL_TX 3 B10
resource SERIAL_TX 6 C06
resource SERIAL_RX 1 A10
resource SERIAL_RX 3 B11
resource SERIAL_RX 6 C07
resource INVERTER 1 C00
resource LED 1 B05
resource SPI_SCK 1 A05
resource SPI_SCK 3 C10
resource SPI_MISO 1 A06
resource SPI_MISO 3 C11
resource SPI_MOSI 1 A07
resource SPI_MOSI 3 C12
resource ESCSERIAL 1 B14
resource ADC_BATT 1 C02
resource ADC_CURR 1 C01
resource FLASH_CS 1 B03
resource OSD_CS 1 A15
resource RX_SPI_CS 1 D02
resource RX_SPI_EXTI 1 A00
resource GYRO_EXTI 1 C04
resource GYRO_CS 1 A04
resource USB_DETECT 1 C05

and 

# resource show
Currently active IO resource assignments:
(reboot to update)
--------------------
A00: RX_SPI_EXTI
A01: FREE
A02: MOTOR 4
A03: MOTOR 3
A04: PREINIT 1
A05: SPI_SCK 1
A06: SPI_MISO 1
A07: SPI_MOSI 1
A08: FREE
A09: FREE
A10: FREE
A11: USB
A12: USB
A15: OSD_CS
B00: MOTOR 1
B01: MOTOR 2
B03: FLASH_CS
B04: BEEPER
B05: LED 1
B06: FREE
B07: FREE
B08: FREE
B09: FREE
B10: FREE
B11: FREE
B12: FREE
B13: FREE
B14: FREE
B15: FREE
C00: INVERTER 1
C01: ADC_CURR
C02: ADC_BATT
C03: FREE
C04: FREE
C05: USB_DETECT
C06: FREE
C07: FREE
C08: FREE
C09: FREE
C10: SPI_SCK 3
C11: SPI_MISO 3
C12: SPI_MOSI 3
D02: RX_SPI_CS

pavankmr853, 2023-02-03:
i have same gyro issue its not detecting showing no gyro pls help me guys my board is foxeer f7 v4