betaflight/betaflight #9997: Move digital idle value to profiles

butterSnake, 2020-07-08:
I set up a quad the other day for a friend that used the auto battery cell detection system and motor limit to let him run 2550kv motors with 4, 5 and 6s. Works really well and was pleased with the results. But the one thing I had to compromise on was the idle speed ended up at 4% which was still a bit to high for the motors on 6s and a little bit to low for them on 4s. If the digital idle value was profile dependant then each cell profile could have its own idle settings

haslinghuis, 2020-07-09:
Logical it seems a good idea to put Digital Idle Offset in (PID) profile settings.
For background #7516

butterSnake, 2020-07-10:
Yeah totally why I think it needs to be profile dependant it’s impossible to get the idle right to suit 4s and 6s currently

haslinghuis, 2020-07-10:
My point is let's make this feature update adjust the digital idle value accordingly based on `auto_profile_cell_count` without the need configuring this. Just calculate an offset from the global percentage based on cell count.

butterSnake, 2020-07-10:
I’d say that’s a bit more complex than it needs to be. There isn’t an infinite amount of space for new features and this isn’t one worth wasting loads of space on

haslinghuis, 2020-07-10:
True, but then it needs work on configurator too. A few extra lines for calculations won't tip the scale on space.

haslinghuis, 2020-07-11:
From slack I learned it is already being considered to move certain settings to profiles in 4.3. Thank you for the input. This is appreciated. For the mean time maybe, i don't if it helps or if you already know, `idle_min_rpm` could help here, not sure how this behaves between different cells. You will have to do some testing to write down the rpm's running different cells.
https://github.com/betaflight/betaflight/wiki/Tuning-Dynamic-Idle

butterSnake, 2020-07-11:
Am aware of this. I posted this feature request cause I’m part of that discussion 😉