betaflight/betaflight #10003: BF 4.2 MAMBA F722S UART PROBLEM

Ringo982, 2020-07-09:
Hello. I’m using BF 4.1.5 on a Quad with Mamba F722S stack and a Runcam Hybrid cam.
The cam is configured in Uart Control mode, so I’m able to configure a switch on my taranis to activate the recording directly from my radio.
With BF 4.1.5 all works perfectly.
I have installed BF 4.2 and the switch stopped to work. I can see on BF that the switch is correctly activated on mode tab, but the recording on the camera doesn’t start.
Both in BF4.1.5 and in BF4.2 the port and mode tabs are configured exactly in the same way, also in the taranis nothing is changed, so I think for some reason with BF4.2 the input for the recording to start is not passed correctly to the runcam hybrid.
I have spoken both with runcam and diatone support but they suppose this is a BF4.2 BUG.
Can you help please?

Thank you
Regards
Francesco

haslinghuis, 2020-07-09:
As requested when you opened the issue, please provide the follow information:

Flight controller configuration
Create a diff and post it here in a code block. Put ``` (three backticks) at the start and end of the diff block (instructions on how to do a diff: https://oscarliang.com/use-diff-not-dump-betaflight/)

Use resource show all to create a resource allocation list and post it here in a code block. Put ``` (three backticks) at the start and end of the output block.

Ringo982, 2020-07-09:
> As requested when you opened the issue, please provide the follow information:
>
> Flight controller configuration
> Create a diff and post it here in a code block. Put ``` (three backticks) at the start [...]

Sure 
see here:

'''
# 

# diff all

# version
# Betaflight / STM32F7X2 (S7X2) 4.2.0 Jun 14 2020 / 03:05:26 (8f2d21460) MSP API: 1.43
# config: manufacturer_id: DIAT, board_name: MAMBAF722, version: a5269145, date: 2020-04-09T22:32:24Z

# start the command batch
batch start

# reset configuration to default settings
defaults nosave

board_name MAMBAF722
manufacturer_id DIAT
mcu_id 003f002a3237510c31313935
signature 

# name: Cidora

# feature
feature -RX_PARALLEL_PWM
feature GPS

# beacon
beacon RX_LOST
beacon RX_SET

# map
map TAER1234

# serial
serial 1 16384 115200 57600 0 115200
serial 2 2048 115200 57600 0 115200
serial 4 2 115200 9600 0 115200

# aux
aux 0 0 0 1950 2050 0 0
aux 1 1 1 1450 1550 0 0
aux 2 46 6 1450 1550 0 0
aux 3 13 2 1925 2075 0 0
aux 4 33 4 1450 1550 0 0
aux 5 35 3 1450 1550 0 0

# vtxtable
vtxtable bands 6
vtxtable channels 8
vtxtable band 1 BOSCAM_A A FACTORY 5865 5845 5825 5805 5785 5765 5745 5725
vtxtable band 2 BOSCAM_B B FACTORY 5733 5752 5771 5790 5809 5828 5847 5866
vtxtable band 3 BOSCAM_E E FACTORY 5705 5685 5665    0 5885 5905    0    0
vtxtable band 4 FATSHARK F FACTORY 5740 5760 5780 5800 5820 5840 5860 5880
vtxtable band 5 RACEBAND R FACTORY 5658 5695 5732 5769 5806 5843 5880 5917
vtxtable band 6 IMD6     I CUSTOM  5732 5765 5828 5840 5866 5740    0    0
vtxtable powerlevels 4
vtxtable powervalues 0 1 2 3
vtxtable powerlabels 25 200 600 120

# master
set gyro_lowpass2_hz = 275
set dyn_lpf_gyro_min_hz = 220
set dyn_lpf_gyro_max_hz = 550
set acc_calibration = -77,-22,98,1
set min_check = 1010
set rssi_channel = 16
set rc_smoothing_derivative_type = PT1
set dshot_idle_value = 700
set dshot_bidir = ON
set small_angle = 180
set gps_provider = UBLOX
set gps_auto_baud = ON
set gps_ublox_use_galileo = ON
set vtx_band = 4
set vtx_channel = 1
set vtx_power = 3
set vtx_low_power_disarm = ON
set vtx_freq = 5740
set gyro_1_align_yaw = 1800
set gyro_rpm_notch_q = 800
set name = Cidora

profile 0

# profile 0
set dyn_lpf_dterm_min_hz = 77
set dyn_lpf_dterm_max_hz = 187
set dyn_lpf_dterm_curve_expo = 6
set dterm_lowpass2_hz = 165
set vbat_sag_compensation = 100
set feedforward_transition = 40
set iterm_relax_cutoff = 10
set yaw_lowpass_hz = 70
set throttle_boost_cutoff = 10
set p_pitch = 51
set i_pitch = 99
set d_pitch = 42
set f_pitch = 105
set p_roll = 46
set i_roll = 94
set d_roll = 39
set f_roll = 99
set p_yaw = 50
set i_yaw = 99
set f_yaw = 99
set d_min_roll = 25
set d_min_pitch = 28
set ff_interpolate_sp = AVERAGED_3
set ff_spike_limit = 55
set ff_smooth_factor = 40
set ff_boost = 0

profile 1

profile 2

# restore original profile selection
profile 0

rateprofile 0

# rateprofile 0
set roll_rc_rate = 75
set pitch_rc_rate = 75
set yaw_rc_rate = 75
set roll_expo = 30
set pitch_expo = 30
set yaw_expo = 30
set roll_srate = 86
set pitch_srate = 86
set yaw_srate = 86

rateprofile 1

rateprofile 2

rateprofile 3

rateprofile 4

rateprofile 5

# restore original rateprofile selection
rateprofile 0

# save configuration
save
# 

'''

and 

'''
# 

# resource show all
Currently active IO resource assignments:
(reboot to update)
--------------------
A00: SERIAL_TX 4
A01: SERIAL_RX 4
A02: SERIAL_TX 2
A03: SERIAL_RX 2
A04: GYRO_CS 1
A05: SPI_SCK 1
A06: SPI_MISO 1
A07: SPI_MOSI 1
A08: MOTOR 3
A09: MOTOR 4
A10: FREE
A11: USB
A12: USB
A13: SWD
A14: SWD
A15: FLASH_CS
B00: PINIO 1
B01: FREE
B02: BEEPER
B03: FREE
B04: FREE
B05: SPI_MOSI 3
B06: FREE
B07: SERIAL_RX 1
B08: CAMERA_CONTROL
B09: FREE
B10: SERIAL_TX 3
B11: FREE
B12: OSD_CS
B13: SPI_SCK 2
B14: SPI_MISO 2
B15: SPI_MOSI 2
C00: FREE
C01: ADC_BATT
C02: FREE
C03: ADC_CURR
C04: GYRO_EXTI
C05: FREE
C06: FREE
C07: FREE
C08: MOTOR 1
C09: MOTOR 2
C10: SPI_SCK 3
C11: SPI_MISO 3
C12: SERIAL_TX 5
C13: FREE
C14: LED 2
C15: LED 1
D00: FREE
D01: FREE
D02: SERIAL_RX 5
D03: FREE
D04: FREE
D05: FREE
D06: FREE
D07: FREE
D08: FREE
D09: FREE
D10: FREE
D11: FREE
D12: FREE
D13: FREE
D14: FREE
D15: FREE
E00: FREE
E01: FREE
E02: FREE
E03: FREE
E04: FREE
E05: FREE
E06: FREE
E07: FREE
E08: FREE
E09: FREE
E10: FREE
E11: FREE
E12: FREE
E13: FREE
E14: FREE
E15: FREE
F00: FREE
F01: FREE
F02: FREE
F03: FREE
F04: FREE
F05: FREE
F06: FREE
F07: FREE
F08: FREE
F09: FREE
F10: FREE
F11: FREE
F12: FREE
F13: FREE
F14: FREE
F15: FREE

Currently active Timers:
-----------------------
TIM1: FREE
TIM2: FREE
TIM3: FREE
TIM4:
    CH3: CAMERA_CONTROL
TIM5: FREE
TIM6: FREE
TIM7: FREE
TIM8:
    CH1: DSHOT_BITBANG 3
    CH2: DSHOT_BITBANG 1
TIM9: FREE
TIM10: FREE
TIM11: FREE
TIM12: FREE
TIM13: FREE
TIM14: FREE

Currently active DMA:
--------------------
DMA1 Stream 0: FREE
DMA1 Stream 1: FREE
DMA1 Stream 2: FREE
DMA1 Stream 3: FREE
DMA1 Stream 4: FREE
DMA1 Stream 5: FREE
DMA1 Stream 6: FREE
DMA1 Stream 7: FREE
DMA2 Stream 0: ADC
DMA2 Stream 1: FREE
DMA2 Stream 2: DSHOT_BITBANG 3
DMA2 Stream 3: DSHOT_BITBANG 1
DMA2 Stream 4: FREE
DMA2 Stream 5: FREE
DMA2 Stream 6: FREE
DMA2 Stream 7: FREE

# 

'''

Ringo982, 2020-07-09:
> ```

Sure
see here:

```
# 

# diff all

# version
# Betaflight / STM32F7X2 (S7X2) 4.2.0 Jun 14 2020 / 03:05:26 (8f2d21460) MSP API: 1.43
# config: manufacturer_id: DIAT, board_name: MAMBAF722, version: a5269145, date: 2020-04-09T22:32:24Z

# start the command batch
batch start

# reset configuration to default settings
defaults nosave

board_name MAMBAF722
manufacturer_id DIAT
mcu_id 003f002a3237510c31313935
signature 

# name: Cidora

# feature
feature -RX_PARALLEL_PWM
feature GPS

# beacon
beacon RX_LOST
beacon RX_SET

# map
map TAER1234

# serial
serial 1 16384 115200 57600 0 115200
serial 2 2048 115200 57600 0 115200
serial 4 2 115200 9600 0 115200

# aux
aux 0 0 0 1950 2050 0 0
aux 1 1 1 1450 1550 0 0
aux 2 46 6 1450 1550 0 0
aux 3 13 2 1925 2075 0 0
aux 4 33 4 1450 1550 0 0
aux 5 35 3 1450 1550 0 0

# vtxtable
vtxtable bands 6
vtxtable channels 8
vtxtable band 1 BOSCAM_A A FACTORY 5865 5845 5825 5805 5785 5765 5745 5725
vtxtable band 2 BOSCAM_B B FACTORY 5733 5752 5771 5790 5809 5828 5847 5866
vtxtable band 3 BOSCAM_E E FACTORY 5705 5685 5665    0 5885 5905    0    0
vtxtable band 4 FATSHARK F FACTORY 5740 5760 5780 5800 5820 5840 5860 5880
vtxtable band 5 RACEBAND R FACTORY 5658 5695 5732 5769 5806 5843 5880 5917
vtxtable band 6 IMD6     I CUSTOM  5732 5765 5828 5840 5866 5740    0    0
vtxtable powerlevels 4
vtxtable powervalues 0 1 2 3
vtxtable powerlabels 25 200 600 120

# master
set gyro_lowpass2_hz = 275
set dyn_lpf_gyro_min_hz = 220
set dyn_lpf_gyro_max_hz = 550
set acc_calibration = -77,-22,98,1
set min_check = 1010
set rssi_channel = 16
set rc_smoothing_derivative_type = PT1
set dshot_idle_value = 700
set dshot_bidir = ON
set small_angle = 180
set gps_provider = UBLOX
set gps_auto_baud = ON
set gps_ublox_use_galileo = ON
set vtx_band = 4
set vtx_channel = 1
set vtx_power = 3
set vtx_low_power_disarm = ON
set vtx_freq = 5740
set gyro_1_align_yaw = 1800
set gyro_rpm_notch_q = 800
set name = Cidora

profile 0

# profile 0
set dyn_lpf_dterm_min_hz = 77
set dyn_lpf_dterm_max_hz = 187
set dyn_lpf_dterm_curve_expo = 6
set dterm_lowpass2_hz = 165
set vbat_sag_compensation = 100
set feedforward_transition = 40
set iterm_relax_cutoff = 10
set yaw_lowpass_hz = 70
set throttle_boost_cutoff = 10
set p_pitch = 51
set i_pitch = 99
set d_pitch = 42
set f_pitch = 105
set p_roll = 46
set i_roll = 94
set d_roll = 39
set f_roll = 99
set p_yaw = 50
set i_yaw = 99
set f_yaw = 99
set d_min_roll = 25
set d_min_pitch = 28
set ff_interpolate_sp = AVERAGED_3
set ff_spike_limit = 55
set ff_smooth_factor = 40
set ff_boost = 0

profile 1

profile 2

# restore original profile selection
profile 0

rateprofile 0

# rateprofile 0
set roll_rc_rate = 75
set pitch_rc_rate = 75
set yaw_rc_rate = 75
set roll_expo = 30
set pitch_expo = 30
set yaw_expo = 30
set roll_srate = 86
set pitch_srate = 86
set yaw_srate = 86

rateprofile 1

rateprofile 2

rateprofile 3

rateprofile 4

rateprofile 5

# restore original rateprofile selection
rateprofile 0

# save configuration
save
# 
```

and 

```
# 

# resource show all
Currently active IO resource assignments:
(reboot to update)
--------------------
A00: SERIAL_TX 4
A01: SERIAL_RX 4
A02: SERIAL_TX 2
A03: SERIAL_RX 2
A04: GYRO_CS 1
A05: SPI_SCK 1
A06: SPI_MISO 1
A07: SPI_MOSI 1
A08: MOTOR 3
A09: MOTOR 4
A10: FREE
A11: USB
A12: USB
A13: SWD
A14: SWD
A15: FLASH_CS
B00: PINIO 1
B01: FREE
B02: BEEPER
B03: FREE
B04: FREE
B05: SPI_MOSI 3
B06: FREE
B07: SERIAL_RX 1
B08: CAMERA_CONTROL
B09: FREE
B10: SERIAL_TX 3
B11: FREE
B12: OSD_CS
B13: SPI_SCK 2
B14: SPI_MISO 2
B15: SPI_MOSI 2
C00: FREE
C01: ADC_BATT
C02: FREE
C03: ADC_CURR
C04: GYRO_EXTI
C05: FREE
C06: FREE
C07: FREE
C08: MOTOR 1
C09: MOTOR 2
C10: SPI_SCK 3
C11: SPI_MISO 3
C12: SERIAL_TX 5
C13: FREE
C14: LED 2
C15: LED 1
D00: FREE
D01: FREE
D02: SERIAL_RX 5
D03: FREE
D04: FREE
D05: FREE
D06: FREE
D07: FREE
D08: FREE
D09: FREE
D10: FREE
D11: FREE
D12: FREE
D13: FREE
D14: FREE
D15: FREE
E00: FREE
E01: FREE
E02: FREE
E03: FREE
E04: FREE
E05: FREE
E06: FREE
E07: FREE
E08: FREE
E09: FREE
E10: FREE
E11: FREE
E12: FREE
E13: FREE
E14: FREE
E15: FREE
F00: FREE
F01: FREE
F02: FREE
F03: FREE
F04: FREE
F05: FREE
F06: FREE
F07: FREE
F08: FREE
F09: FREE
F10: FREE
F11: FREE
F12: FREE
F13: FREE
F14: FREE
F15: FREE

Currently active Timers:
-----------------------
TIM1: FREE
TIM2: FREE
TIM3: FREE
TIM4:
    CH3: CAMERA_CONTROL
TIM5: FREE
TIM6: FREE
TIM7: FREE
TIM8:
    CH1: DSHOT_BITBANG 3
    CH2: DSHOT_BITBANG 1
TIM9: FREE
TIM10: FREE
TIM11: FREE
TIM12: FREE
TIM13: FREE
TIM14: FREE

Currently active DMA:
--------------------
DMA1 Stream 0: FREE
DMA1 Stream 1: FREE
DMA1 Stream 2: FREE
DMA1 Stream 3: FREE
DMA1 Stream 4: FREE
DMA1 Stream 5: FREE
DMA1 Stream 6: FREE
DMA1 Stream 7: FREE
DMA2 Stream 0: ADC
DMA2 Stream 1: FREE
DMA2 Stream 2: DSHOT_BITBANG 3
DMA2 Stream 3: DSHOT_BITBANG 1
DMA2 Stream 4: FREE
DMA2 Stream 5: FREE
DMA2 Stream 6: FREE
DMA2 Stream 7: FREE

# 
```

haslinghuis, 2020-07-09:
Do you start recording before or after arming?
Try switching the camera to analog and back to digital to see if this helps.
See also #8283

Ringo982, 2020-07-09:
> Do you start recording before or after arming?
> Try switching the camera to analog and back to digital to see if this helps.
> See also #8283

With BF4.1.5 I can star recording both before and after arming. With BF4.2 I cannot start recording using the switch at all, before and after.
How can I switch the camera to analog???

Ringo982, 2020-07-09:
> Do you start recording before or after arming?
> Try switching the camera to analog and back to digital to see if this helps.
> See also #8283

Perhaps for "analog mode" you mean "joystick control mode". If so i have tried to switch to joystick mode and the to turn back to uart ctrl mode but nothing is changed. I confirm that before with BF4.1.5 i was able to activate the recording both before and after arming, now with BF4.2 is not possible before and also after arming.

What can i do?
Thank you

haslinghuis, 2020-07-09:
Did you update camera to the latest firmware?
https://support.runcam.com/hc/en-us/articles/360037561534-Hybrid-firmware-updating-via-computer

Ringo982, 2020-07-09:
> Did you update camera to the latest firmware?
> https://support.runcam.com/hc/en-us/articles/360037561534-Hybrid-firmware-updating-via-computer

Yes sure I’m using the latest available fw on the camera

haslinghuis, 2020-07-09:
I would except finding feature `RCDEVICE_PROTOCOL_FEATURE_START_RECORDING` and `RCDEVICE_PROTOCOL_FEATURE_STOP_RECORDING` in the betaflight code.

https://support.runcam.com/hc/en-us/articles/360014537794-RunCam-Device-Protocol

There are some settings to play with:
```
set rcdevice_protocol_version = 1
set rcdevice_feature = 7
```
I haven't found yet conclusive documentation about this.

Ringo982, 2020-07-30:
Hello
any other update please?
The problem is not solved yet and the answer of runcam support is "use BF 4.1.X instead of 4.2.X".... and it is not an acceptable answer for me...
Thank you

Ringo982, 2020-07-30:
> I would expect finding feature `RCDEVICE_PROTOCOL_FEATURE_START_RECORDING` and `RCDEVICE_PROTOCOL_FEATURE_STOP_RECORDING` in the betaflight code.
> 
> https://support.runcam.com/hc/en-us/articles/360014537794-RunCam-Device-Protocol
> 
> There are some settings to play with:
> 
> ```
> set rcdevice_protocol_version = 1
> set rcdevice_feature = 7
> ```
> 
> I haven't found yet conclusive documentation about this.

Hello
any other update please?
The problem is not solved yet and the answer of runcam support is "use BF 4.1.X instead of 4.2.X".... and it is not an acceptable answer for me...
Thank you

closed 2020-09-05 by stale[bot]